cartographer-project/cartographer #942: How to deal with motion  distortion

xiangkui123, 2018-02-28:
hi, How to deal with radar motion distortion in cartographer ? Radar data of one frame of data which the scanning point is not in the same coordinate system the.

cschuet, 2018-02-28:
Cartographer uses odometry and IMU data to extrapolate the last known pose to the time of the incoming frame of data. If Cartographer is given relative times of the points within a data frame we even unwarp the backpack motion for each individual point.

xiangkui123, 2018-02-28:
Probability grid can reduce the impact of distortion?

xiangkui123, 2018-02-28:
Cartographer can use Pre-integral theory?

cschuet, 2018-02-28:
Can you be a little more elaborate? What are you trying to find out? Are you impacted by distortion and you are trying to find out how to improve the situation?

xiangkui123, 2018-02-28:
In the cartographer just see imu and odometry point cloud position correction, but did not see any correction of point cloud distortion，ｔhis is question one．I do not have a lot of measurement cartographer global map accuracy,Does it assume that tightly coupled imu information improves the accuracy of the global map ．The problem is naive

closed 2018-03-05 by xiangkui123