cartographer-project/cartographer #943: Can not build map

xiangkui123, 2018-02-28:
hello everyone!
      Test 3D drawing, but can not be a normal map,No matter how to move the backpack, the built map is rotated at the origin!

cschuet, 2018-02-28:
Hi, your report is a little vague. Can you tell us more about your setup, a screen shot or capture maybe of the problem. Configuration files and a recorded bag would also help of course.

xiangkui123, 2018-02-28:
[image]
I used velodyne 16 and stim300.

cschuet, 2018-02-28:
First suspicion is that your IMU to rangefinder transform is not correctly specified in your urdf. Can you check that?

xiangkui123, 2018-02-28:
I modified urdf file, What calibration tools are not used, just a simple measurement of translation vector, imu and radar coordinate system is roughly the same．What calibration tools are you using?（：

closed 2018-03-05 by xiangkui123