cartographer-project/cartographer #978: about visualization

deepmeng, 2018-03-11:
hello, I'm curious about how you load the triangle in rviz?
[image]

cschuet, 2018-03-11:
@SirVer Did you create the demo animation?

ojura, 2018-03-11:
AFAIK that was a visualization tool which predates the rviz plugin, which was not publically released.

SirVer, 2018-03-12:
@cschuet @damonkohler did. But @ojura is correct, this is our old Java based viz, predating cartographer_ros.

@deepmeng We drew the cursor as a separate layer in our Viz tool using a hardcoded set of points, transformed according to the current pose.

deepmeng, 2018-03-12:
Interesting! Thanks for your replies.

closed 2018-03-12 by deepmeng

ojura, 2018-03-12:
@deepmeng you can specify simple shapes in an urdf and display them in rviz (e.g. see the `<geometry>` tag). You can also specify in the urdf a model (in the collada format, IIRC) and attach it to a link - in this case, `base_link`.