cartographer-project/cartographer #993: If there is any one can help me with  this error

senses1, 2018-03-15:
I compiled cartographer-master 
$cmake CMakeLists.txt
when it comes out error infomation like:
`- Build type: Release
CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106600
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:31 (find_package)

-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   iostreams
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
CMake Error at /usr/local/lib/cmake/Ceres/CeresConfig.cmake:88 (message):
  Failed to find Ceres - Missing requested Ceres components: [SuiteSparse]
  (components requested: [SuiteSparse]).  Detected Ceres version: 1.14.0
  installed in: /usr/local with components: [EigenSparse,
  SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP, Multithreading].
Call Stack (most recent call first):
  /usr/local/lib/cmake/Ceres/CeresConfig.cmake:390 (ceres_report_not_found)
  CMakeLists.txt:32 (find_package)

CMake Error at CMakeLists.txt:32 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/Ceres/CeresConfig.cmake

  but it set Ceres_FOUND to FALSE so package "Ceres" is considered to be NOT
  FOUND.

-- Configuring incomplete, errors occurred!
See also "/home/tby/cartographer-master/CMakeFiles/CMakeOutput.log".
See also "/home/tby/cartographer-master/CMakeFiles/CMakeError.log".
`
and I searched for this ,I found the Suitesparse problem is fixed under Debian , but not under Ubuntu.
how could I solve this problem?

pifon2a, 2018-03-15:
Hi,

have you tried `sudo apt-get install libsuitesparse-dev` and then rebuild Ceres? It should work on Ubuntu as well. 

If you are using the version in Ubuntu that you installed with `apt-get`, then remove it with `sudo apt-get remove` and build it yourself instead.

Here is the full list of commands:
```
sudo apt-get update
sudo apt-get install -y \
    cmake \
    g++ \
    git \
    google-mock \
    libboost-all-dev \
    libcairo2-dev \
    libeigen3-dev \
    libgflags-dev \
    libgoogle-glog-dev \
    liblua5.2-dev \
    libsuitesparse-dev \
    ninja-build \
    python-sphinx \
    stow
```
and then
```
git clone https://ceres-solver.googlesource.com/ceres-solver
cd ceres-solver
mkdir build
cd build
cmake .. -G Ninja -DCXX11=ON  -DCMAKE_INSTALL_PREFIX=/usr/local/stow/ceres
ninja -j15
sudo ninja install
cd /usr/local/stow/ 
sudo stow ceres
```
Now you can rebuild Cartographer with `cmake .. -G Ninja && ninja -j15`.  Do not forget to clean-up you build directories before that.

You can actually build Cartographer without SuiteSparse. It will be slower though. To do that, just open `CMakeLists.txt` and replace the line `find_package(Ceres REQUIRED COMPONENTS SuiteSparse)` to `find_package(Ceres REQUIRED)`.

senses1, 2018-03-18:
Hi pifon2a ,
I must thank you for your concern the question at your first convinence.
 I followed  your way but the terminal seems to hang up when i input 
$ninja -15 
[1/2] Building CXX object examples/CMakeFiles/nist.dir/nist.cc.o
the system seems very slow
------------------------------------------------------------------------------------
and i start a new terminal and execute 
$sudo ninja install 
[1/3] Building CXX object examples/CMakeFiles/nist.dir/nist.cc.o

the system seems very slow

------------------------------------------------------------------------------------
then all other shell cmd would not work properly
------------------------------------------------------------------------------------
I replace the line in CMakeLists.txt like you said,and compiled with the shell line
$cmake ./CMakeLists.txt -G Ninja && ninja -j15
 but the error infomation comes like below:

./CMakeLists.txt: 行 15: 未预期的符号 `VERSION' 附近有语法错误
./CMakeLists.txt: 行 15: `cmake_minimum_required(VERSION 2.8.12)  # Ships with Ubuntu 14.04 (Trusty)'

I am not so familiar with linux system,i usually use windows.
thanks again!!!!

pifon2a, 2018-03-18:
Hi @senses1 ,

could you please run ` cmake --version ` and copy-paste here the output. It might have happened that your `cmake` should be updated.

Also, please, do not run the next command after the previous has failed: 
`ninja -j15` this command compiles stuff
`sudo ninja install` this one copies the output of the previous step into the specified directory (DCMAKE_INSTALL_PREFIX).

senses1, 2018-03-19:
the cmake version is 3.5.1 , and i uninstalledd it

senses1, 2018-03-19:
The version of cmake is 3.5.1 ,I uninstalled it.

i download cmake 3.10.1 .
execute shell below:
$bootstrap
$make
$make install
but cmake could not install properly, and cmake command seems not to be found.

senses1, 2018-03-19:
Hi @pifon2a,
I think ninja compile is a dead process 
[1/2] Building CXX object  examples/CMakeFiles/nist.dir/nist.cc.o
another dead file is something like 3d_map...
I restart VM directly, I guess this way would not get pass.
Is there another way to compile ceres-solver without ninja?

BTW, I'm using Virtual Box : Ubuntu 16.04 on 1 Core 1.5GB RAM 
Physical is Win7 SP1 on  i7 6500U ,4GB RAM Solid State Disk 120GB.

senses1, 2018-03-19:
I only use 2D indoor SLAM , it is enough for me

pifon2a, 2018-03-19:
You can do it just with cmake without ninja. Install these:
```
sudo apt-get install autoconf \
automake \
build-essential \
clang \
clang-format \
cmake \
cmake-curses-gui \
curl \
g++ \
git \
gitk  \
google-mock \
kcachegrind \
libboost-all-dev \
libc++-dev \
libcairo2-dev \
libeigen3-dev \
libgflags-dev \
libgtest-dev \
libgoogle-glog-dev \
liblua5.2-dev \
libsuitesparse-dev \
libtool \
make \
meld \
ninja-build \
python-rosdep \
python-sphinx
python-wstool \
stow \
unzip \
valgrind
```

Ok, now let's start from scratch.

Ceres part:
```
git clone https://github.com/ceres-solver/ceres-solver.git
cd ceres-solver
mkdir build && cd build
cmake .. -G Ninja -DCXX11=ON -DCMAKE_INSTALL_PREFIX=/usr/local/stow/ceres
cd /usr/local/stow/
sudo stow ceres
```

Cartographer part:
```
git@github.com:googlecartographer/cartographer.git
cd cartographer
mkdir build
cd build
cmake ..
make -j10
```

senses1, 2018-03-20:
Hi @pifon2a,
compile seems went well,but I ran
$sudo stow ceres 
the terminal shows as below: 
$ cmake .. -G Ninja -DCXX11=ON -DCMAKE_INSTALL_PREFIX=/usr/local/stow/ceres
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detected Ceres version: 1.14.0 from /home/tby/ceres-solver/include/ceres/version.h
-- No preference for use of exported Eigen CMake configuration set, and no hints for include directory provided. Defaulting to preferring an installed/exported Eigen CMake configuration if available.
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found Eigen: /usr/include/eigen3 (found version "3.2.92") 
-- Found Eigen version 3.2.92: /usr/include/eigen3
-- Enabling use of Eigen as a sparse linear algebra library.
-- Looking for dgemm_
-- Looking for dgemm_ - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- A library with BLAS API found.
-- Found LAPACK library: /usr/lib/liblapack.so;/usr/lib/libf77blas.so;/usr/lib/libatlas.so
-- A library with BLAS API found.
-- Found AMD headers in: /usr/local/include
-- Found AMD library: /usr/local/lib/libamd.a
-- Found CAMD headers in: /usr/local/include
-- Found CAMD library: /usr/local/lib/libcamd.a
-- Found COLAMD headers in: /usr/local/include
-- Found COLAMD library: /usr/local/lib/libcolamd.a
-- Found CCOLAMD headers in: /usr/local/include
-- Found CCOLAMD library: /usr/local/lib/libccolamd.a
-- Found CHOLMOD headers in: /usr/local/include
-- Found CHOLMOD library: /usr/local/lib/libcholmod.a
-- Found SUITESPARSEQR headers in: /usr/local/include
-- Found SUITESPARSEQR library: /usr/local/lib/libspqr.a
-- Did not find Intel TBB library, assuming SuiteSparseQR was not compiled with TBB.
-- Found SUITESPARSE_CONFIG headers in: /usr/local/include
-- Found SUITESPARSE_CONFIG library: /usr/local/lib/libsuitesparseconfig.a
-- Found LIBRT library: /usr/lib/x86_64-linux-gnu/librt.so
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "4.0.2") 
-- Found SuiteSparse 4.0.2, building with SuiteSparse.
-- Found CXSparse: /usr/local/include (found version "2.3.0") 
-- Found CXSparse version: 2.3.0, building with CXSparse.
-- No preference for use of exported gflags CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported gflags CMake configuration if available.
-- Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for gflags, will perform search for installed gflags components.
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE - Success
-- Found Gflags: /usr/include  
-- Found Google Flags header in: /usr/include, in namespace: google
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
[... 11 lines cut ...]
-- Looking for C++ include memory
-- Looking for C++ include memory - found
-- Performing Test HAVE_SHARED_PTR_IN_STD_NAMESPACE
-- Performing Test HAVE_SHARED_PTR_IN_STD_NAMESPACE - Success
-- Found shared_ptr in std namespace using <memory> header.
-- Performing Test CXX11_MATH_FUNCTIONS_FOUND
-- Performing Test CXX11_MATH_FUNCTIONS_FOUND - Success
   ==============================================================
   Compiling Ceres using C++11.  This will result in a version 
   of Ceres that will require the use of C++11 in client code.
   ==============================================================
-- Failed to find Google benchmark library, disabling build of benchmarks.
-- Building Ceres as a static library.
-- No build type specified; defaulting to CMAKE_BUILD_TYPE=Release.
-- Enabling CERES_USE_EIGEN_SPARSE in Ceres config.h
-- Enabling CERES_USE_CXX11 in Ceres config.h
-- Enabling CERES_USE_OPENMP in Ceres config.h
-- Enabling CERES_HAVE_PTHREAD in Ceres config.h
-- Enabling CERES_HAVE_RWLOCK in Ceres config.h
-- Enabling CERES_STD_UNORDERED_MAP in Ceres config.h
-- Build the examples.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tby/ceres-solver/build
tby@tby-VirtualBox:~/ceres-solver/build$ cd /usr/local/stow/tby@tby-VirtualBox:/usr/local/stow$ sudo stow ceresstow: ERROR: The stow directory stow does not contain package ceres

pifon2a, 2018-03-20:
Oh, I am so sorry: you need two more commands for Ceres:
```
git clone https://github.com/ceres-solver/ceres-solver.git
cd ceres-solver
mkdir build && cd build
cmake .. -G Ninja -DCXX11=ON -DCMAKE_INSTALL_PREFIX=/usr/local/stow/ceres
make -j10
sudo make install
cd /usr/local/stow/
sudo stow ceres
```

senses1, 2018-03-21:
I changed the cmake command like:

cmake .. -G "Unix Makefiles" -DCXX11=ON -DCMAKE_INSTALL_PREFIX=/usr/local/stow/ceres
make -j2
sudo make install
cd /usr/local/stow/
sudo stow ceres
and Trying.
But I have tried yesterday morning in ceres-solver-master , the compile process still not respond when it goto nist.cc.o

senses1, 2018-03-21:
[100%] Linking CXX executable ../../../bin/ba_sparseschur_cxsparse_user_threads_test
[100%] Built target ba_sparseschur_cxsparse_user_threads_test

The terminal window show the last line like this and do not move on.
I think  make -j2 exhausted my laptop's resources.

The code compile should be under a very high performance PC?

pifon2a, 2018-03-21:
@senses1, I do not know, we have a quite powerful hardware here. Try to remove `-j*` at all.

senses1, 2018-03-22:
finally I compiled the cartographer,I found 2 Error , but still can not find which file to modify the situation  of not compile TEST funstions:

/usr/local/lib/libgmock_main.a(gmock_main.cc.o)：在函数‘RUN_ALL_TESTS()’中：
gmock_main.cc:(.text._Z13RUN_ALL_TESTSv[_Z13RUN_ALL_TESTSv]+0x5)：对‘testing::UnitTest::GetInstance()’未定义的引用
gmock_main.cc:(.text._Z13RUN_ALL_TESTSv[_Z13RUN_ALL_TESTSv]+0xd)：对‘testing::UnitTest::Run()’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/cartographer.transform.transform_test.dir/build.make:103: recipe for target 'cartographer.transform.transform_test' failed
make[2]: *** [cartographer.transform.transform_test] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/cartographer.transform.transform_test.dir/all' failed
make[1]: *** [CMakeFiles/cartographer.transform.transform_test.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

senses1, 2018-03-22:
在函数 ****中       means in function
未定义的引用       means undefined reference.

senses1, 2018-03-22:
$make -k 
but I wonder if this command would lead to a successful compile to key module?

senses1, 2018-03-23:
Hi pifon2a,
Well I am disturbing you again for the 3 commented question listed up above.
Thanks again

pifon2a, 2018-03-23:
Try to do 'make clean' and recompile. Gmock should link.

senses1, 2018-03-24:
I've commented the gtest code compile command in CMakelist.txt and it passed.
And I am going to try cartographer.
Thank you you your help very much!!!

senses1, 2018-03-24:
@pifon2a

closed 2018-03-26 by senses1

touchmii, 2020-08-11:
> I compiled cartographer-master
> $cmake CMakeLists.txt
> when it comes out error infomation like:
> `- Build type: Release
> CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message): [...]

build_isolated/ceres-solver/install/CMakeCache.txt 里找到EIGENSPARSE:BOOL=OFF ， 把OFF 改成ON