cartographer-project/cartographer #1001: Eigen + STL containers needs a special allocator

ojura, 2018-03-16:
I have just noticed that `TimedPointCloud` is actually a vector of `Eigen::Vector4f`:
https://github.com/googlecartographer/cartographer/blob/1f9c78a82bc2c732dd3e371895b4c5e8f208ec6a/cartographer/sensor/point_cloud.h#L39

There are some special considerations to take into account when using fixed-size vectorizable Eigen types such as `Vector4f` with STL containers:

https://eigen.tuxfamily.org/dox/group__TopicStlContainers.html

ojura, 2018-03-16:
Also, should classes such as Rigid2 [have an aligned new operator](https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html)?

gaschler, 2018-03-16:
I remember a similar discussion last year. The bottom line was that we only support 64 bits and have no plan to add eigen align macros everywhere.
https://google-cartographer.readthedocs.io/en/latest/#known-issues

ojura, 2018-03-16:
Ah, I have a faint memory of that as well. So, alignment is not an issue on 64 bit arch.

closed 2018-03-16 by ojura

ojura, 2018-03-16:
Linking this issue where this discussion was already held.
https://github.com/googlecartographer/cartographer/issues/34

ojura, 2019-01-02:
Is this the definitive answer - Cartographer will not support AVX alignment, ever?

Even if we don't need AVX in Cartographer (although I guess it couldn't hurt), suppose that the client code wants to use AVX with Eigen for some other stuff, therefore needs to compiled with e.g. -march=native or -march=corei7-avx or whatever. The moment we pull in Cartographer's headers, it can be game over because of alignment issues.

libcartographer users would be forced to separate interfacing with Cartographer into a separate library which has AVX disabled privately, and which would include Cartographer headers privately, while the rest of the codebase could use AVX. Perhaps this could work, but I can see it causing quite a bit of pain.

wohe, 2019-01-07:
@ojura No, this was not the answer. The answer was that http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1409 is a much better approach to add proper alignment compared to expecting every user (including transitive uses) of structs containing Eigen types to adhere to https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html.

ojura, 2019-01-07:
Thanks for reminding me about that. How do you use this? Just enable C++17, and Eigen structures will automagically be aligned? Or would I need to add `alignas(32)` to e.g. `Rigid3`?

ojura, 2019-06-09:
Here is a small update on this: code for closing the issue 1409 was merged in February. The Eigen documentation now mentions enabling C++17 should be enough: https://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html
Eigen 3.4. has not been yet released, though.

Meanwhile, I think a good interim solution is to add `-DEIGEN_MAX_ALIGN_BYTES=16`. Even if AVX is enabled, this will prevent Cartographer from crashing.

Ellon, 2019-06-28:
Hi @ojura , I believe I'm having alignment related issues with cartographer + Eigen 3.3.7 running on Fedora 29, and adding `-DEIGEN_MAX_ALIGN_BYTES=16` to the compilation command does not seem to fix the problem... Would you have any other insights on this problem ?

Ellon, 2019-06-28:
Okay nevermind. I confirm that the problem was related to alignments, and that adding `-DEIGEN_MAX_ALIGN_BYTES=16` solved the problem.

Ellon, 2019-07-03:
Hi again @ojura, regarding the workaround using `-DEIGEN_MAX_ALIGN_BYTES=16`, do you think it needs to be added to cartographer and its dependencies (notably CERES) or just to cartographer and the packages depending on it ?

I'm asking this question to assess if adding something like 
```
if(${EIGEN3_VERSION} VERSION_GREATER_EQUAL "3.3" AND ${EIGEN3_VERSION} VERSION_LESS "3.4")
  target_compile_definitions(${PROJECT_NAME} PUBLIC EIGEN_MAX_ALIGN_BYTES=16)
endif()
```
to cartographer's `CMakeList.txt` would be a solution.

ojura, 2019-07-03:
Your CMake code looks reasonable.

I think that Ceres uses Eigen only internally, and does not expose any eigen structures in it’s api, so it should not be necessary for Ceres.

Ellon, 2019-07-05:
All right, TYVM for your input!

ojura, 2019-07-06:
Btw, you should also check for C++17. Only that combo (Eigen 3.4 + C++17) will automagically align everything.