cartographer-project/cartographer #1005: Build Error - Unable to catkin_make_isolated

BakhtawarRehman, 2018-03-19:
Greetings,

I am stuck on installation of cartographer from last couple of days.
Now, the error is about the use of older version of protobuf.
 
BUT; here's the output of protoc --version:
libprotoc 3.4.0

I am not sure what to do now and i am extremely short in time for completing my project. 
Any Help please!!

Below is the part of the make errors:

`/usr/include/gazebo-7/gazebo/msgs/wireless_node.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:118:0,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Entity.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:34,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/include/gazebo_plugins/gazebo_ros_imu.h:31,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/src/gazebo_ros_imu.cpp:23:
/usr/include/gazebo-7/gazebo/msgs/wireless_nodes.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/wireless_nodes.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/wireless_nodes.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:122:0,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Entity.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:34,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/include/gazebo_plugins/gazebo_ros_imu.h:31,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/src/gazebo_ros_imu.cpp:23:
/usr/include/gazebo-7/gazebo/msgs/world_modify.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/world_modify.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/world_modify.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:124:0,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Entity.hh:30,
                 from /usr/include/gazebo-7/gazebo/physics/Model.hh:34,
                 from /usr/include/gazebo-7/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-7/gazebo/physics/physics.hh:2,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/include/gazebo_plugins/gazebo_ros_imu.h:31,
                 from /home/bakhtawar/catkin_ws/src/ar_601/gazebo_plugins/src/gazebo_ros_imu.cpp:23:
/usr/include/gazebo-7/gazebo/msgs/wrench_stamped.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-7/gazebo/msgs/wrench_stamped.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/wrench_stamped.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
ninja: build stopped: subcommand failed.
<== Failed to process package 'gazebo_plugins': 
  Command '['/home/bakhtawar/catkin_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/bakhtawar/catkin_ws/build_isolated/gazebo_plugins && /home/bakhtawar/catkin_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.
`

gaschler, 2018-03-19:
You have a conflict with a system-installed package (Gazebo), which requires a different version of the protobuf library.
To run Gazebo, you need to normal protobuf package your system provides.
To compile and run Cartographer, you to have Protobuf 3 (compiler and library) in a place where only Cartographer finds it.
In the past, users have installed Protobuf 3 only in `catkin_ws/install_isolated`.
You may have to manually set cmake flags so Protobuf 3 is found.

BakhtawarRehman, 2018-03-21:
well.  ihave tried it but no luck:(
It's taking alot of my precious time. I have tried many diffrerent ways to cope with but no luck.
I think i better quit.

BakhtawarRehman, 2018-03-21:
@gaschler Hello,

I have tried it using this https://gist.github.com/kdaun/51b7d19bbcd5e0798c8415cd093078fb 

However; after installing many packages; it is now giving me this error:

-----------------------------------------------------------------

-- roscpp: 1 messages, 3 services
-- Looking for include file ifaddrs.h
-- Looking for include file ifaddrs.h - found
-- Looking for trunc
-- Looking for trunc - not found
-- Looking for epoll_wait
-- Looking for epoll_wait - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bakhtawar/cartographer_ws/build_isolated/roscpp

==> ninja -j8 -l8 in '/home/bakhtawar/cartographer_ws/build_isolated/roscpp'
[8/49] Building CXX object CMakeFiles/roscpp.dir/src/libros/common.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/common.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/common.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/common.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/common.cpp
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/common.cpp:35:24: fatal error: ros/common.h: No such file or directory
compilation terminated.

[8/49] Building CXX object CMakeFiles/...pp.dir/src/libros/publisher_link.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/publisher_link.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/publisher_link.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/publisher_link.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/publisher_link.cpp
In file included from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/publisher_link.cpp:35:0:
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/ros/publisher_link.h:31:24: fatal error: ros/common.h: No such file or directory
compilation terminated.

[8/49] Building CXX object CMakeFiles/...r/src/libros/service_publication.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/service_publication.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/service_publication.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/service_publication.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/service_publication.cpp
In file included from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/ros/service_publication.h:31:0,
                 from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/service_publication.cpp:35:
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/ros/service_callback_helper.h:32:24: fatal error: ros/common.h: No such file or directory
compilation terminated.

[8/49] Building CXX object CMakeFiles/roscpp.dir/src/libros/subscriber.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/subscriber.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/subscriber.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/subscriber.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/subscriber.cpp
In file included from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/ros/node_handle.h:32:0,
                 from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/subscriber.cpp:29:
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/ros/publisher.h:32:24: fatal error: ros/common.h: No such file or directory
compilation terminated.

[8/49] Building CXX object CMakeFiles/...bros/single_subscriber_publisher.cpp.o
[... cut ...]
compilation terminated.

[8/49] Building CXX object CMakeFiles/roscpp.dir/src/libros/connection.cpp.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'roscpp': 
  Command '['/home/bakhtawar/cartographer_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/bakhtawar/cartographer_ws/build_isolated/roscpp && /home/bakhtawar/cartographer_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

BakhtawarRehman, 2018-03-22:
Update:

For some reasons; make keep asking for new files.
I have installed manually all the files (as much as i could find) in the directory. 

But now; it's not possible. No more progress possible. Here are the error; i am getting now:

--------------------------------------------------------------------------------------------------

<== Finished processing package [53 of 76]: 'xmlrpcpp'

==> Processing catkin package: 'roscpp'
==> Building with env: '/home/bakhtawar/cartographer_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/bakhtawar/cartographer_ws/build_isolated/roscpp'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/bakhtawar/cartographer_ws/build_isolated/roscpp'
[4/5] Building CXX object CMakeFiles/roscpp.dir/src/libros/statistics.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/statistics.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/statistics.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/statistics.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/statistics.cpp
In file included from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/statistics.cpp:30:0:
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/rosgraph_msgs/TopicStatistics.h:7:21: fatal error: ros/msg.h: No such file or directory
compilation terminated.
[4/5] Building CXX object CMakeFiles/roscpp.dir/src/libros/publication.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/publication.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/publication.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/publication.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/publication.cpp
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/publication.cpp:34:29: fatal error: std_msgs/Header.h: No such file or directory
compilation terminated.
[4/5] Building CXX object CMakeFiles/roscpp.dir/src/libros/rosout_appender.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/rosout_appender.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/rosout_appender.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/rosout_appender.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/rosout_appender.cpp
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/rosout_appender.cpp:42:31: fatal error: rosgraph_msgs/Log.h: No such file or directory
compilation terminated.
[4/5] Building CXX object CMakeFiles/roscpp.dir/src/libros/init.cpp.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include -I/home/bakhtawar/cartographer_ws/devel_isolated/roscpp/include/ros -I/home/bakhtawar/cartographer_ws/install_isolated/include -I/home/bakhtawar/cartographer_ws/install_isolated/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I. -fPIC   -std=c++11 -Wall -Wextra -MMD -MT CMakeFiles/roscpp.dir/src/libros/init.cpp.o -MF CMakeFiles/roscpp.dir/src/libros/init.cpp.o.d -o CMakeFiles/roscpp.dir/src/libros/init.cpp.o -c /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/init.cpp
In file included from /home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/init.cpp:53:0:
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/include/roscpp/GetLoggers.h:1:1: error: stray ‘\302’ in program
 <html class="" lang="en"><head prefix="og: http://ogp.me/ns#"><meta charset="utf-8"><meta content="IE=edge" http-equiv="X-UA-Compatible"><meta content="object" property="og:type"><meta content="GitLab" property="og:site_name"><meta content="firmware/libraries/ros_lib/roscpp/GetLoggers.h · b19bef6feeb98a6a61592f32cdecab2f5e81401e · Paulo Rezeck / hero_driver" property="og:title"><meta content="GitLab Community Edition" property="og:description"><meta content="https://www.verlab.dcc.ufmg.br/gitlab/assets/gitlab_logo-7ae504fe4f68fdebb3c2034e36621930cd36ea87924c11ff65dbcb8ed50dca58.png" property="og:image"><meta content="64" property [... cut ...]
/home/bakhtawar/cartographer_ws/src/ros_comm-lunar-devel/clients/roscpp/src/libros/init.cpp:54:35: fatal error: roscpp/SetLoggerLevel.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
<== Failed to process package 'roscpp': 
  Command '['/home/bakhtawar/cartographer_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/bakhtawar/cartographer_ws/build_isolated/roscpp && /home/bakhtawar/cartographer_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

closed 2018-03-28 by BakhtawarRehman