cartographer-project/cartographer #1012: GitHub API rate limit error during automated builds

MichaelGrupp, 2018-03-21:
```
Step 1/8 : FROM cartographer_ros:indigo
 ---> 3d926327d6f7
Step 2/8 : ADD https://api.github.com/repos/googlecartographer/cartographer_ros/git/refs/heads/master cartographer_magazino/cartographer_ros_version.json
ADD failed: failed to GET https://api.github.com/repos/googlecartographer/cartographer_ros/git/refs/heads/master with status 403 Forbidden: {"message":"API rate limit exceeded for 35.188.1.99. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://developer.github.com/v3/#rate-limiting"}
```
Full log: https://api.travis-ci.org/v3/job/355840081/log.txt

Looks like the build agents need an authentication token to allow more frequent access. How about adding a simple token with no permissions like these guys did? https://github.com/Microsoft/vscode/issues/3708#issuecomment-192213741

MichaelGrupp, 2018-03-27:
Failed again today: https://api.travis-ci.org/v3/job/358876755/log.txt

ojura, 2018-03-27:
Maybe https://api.github.com/repos/googlecartographer/cartographer_ros/git/refs/heads/master could be mirrored somewhere on a Google server, and CI changed to use the mirror?

MichaelGrupp, 2018-03-27:
I first thought that this might be a general issue, but it seems to happen only with our repository. Probably I'm just missing an access token in the API calls of our Dockerfiles.

See here: https://github.com/magazino/cartographer_magazino/issues/9

Closing this one for now.

ojura, 2018-03-27:
It happens on Google's CI as well (take a look at master):
https://travis-ci.org/googlecartographer/cartographer_ros/branches

MichaelGrupp, 2018-03-27:
Thanks, didn't check all logs because there were many other CI issues the last days.

MichaelGrupp, 2018-03-27:
It looks like my approach explained from https://github.com/magazino/cartographer_magazino/issues/9 works: 
```
docker build ${TRAVIS_BUILD_DIR} -t cartographer_magazino:${ROS_RELEASE} -f Dockerfile.${ROS_RELEASE} --build-arg github_token=${GITHUB_TOKEN}
Sending build context to Docker daemon  288.3kB

Step 1/9 : ARG github_token
 ---> 
Step 2/9 : FROM cartographer_ros:indigo
 ---> 615994f418e2
Step 3/9 : ADD https://api.github.com/repos/googlecartographer/cartographer_ros/git/refs/heads/master?access_token=$github_token cartographer_magazino/cartographer_ros_version.json

 ---> Using cache
 ---> 040732a61e04
```
https://api.travis-ci.org/v3/job/358953660/log.txt

I will prepare follow-up PRs for the googlecartographer repos and request the maintainers to add the tokens to their Travis.

MichaelGrupp, 2018-03-27:
5000 requests per hour for authenticated requests are probably enough to not require mirroring ;)

closed 2018-04-05 by cschuet