cartographer-project/cartographer #1015: failed build with the master branch on Ubuntu 16.04 gcc-5

manifolds, 2018-03-23:
[  27%] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc.o
In file included from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers.h:87:0,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/landmark_cost_function_2d.h:23,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:30:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers_impl.h: In instantiation of ‘std::array<FloatType, 3ul> cartographer::mapping::pose_graph::ScaleError(std::array<FloatType, 3ul>, double, double) [with T = ceres::Jet<double, 6>]’:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/spa_cost_function_2d.h:51:46:   required from ‘bool cartographer::mapping::pose_graph::SpaCostFunction2D::operator()(const T*, const T*, T*) const [with T = ceres::Jet<double, 6>]’
/usr/include/ceres/internal/variadic_evaluate.h:166:19:   required from ‘static bool ceres::internal::VariadicEvaluate<Functor, T, N0, N1, 0, 0, 0, 0, 0, 0, 0, 0>::Call(const Functor&, const T* const*, T*) [with Functor = cartographer::mapping::pose_graph::SpaCostFunction2D; T = ceres::Jet<double, 6>; int N0 = 3; int N1 = 3]’
/usr/include/ceres/internal/autodiff.h:282:72:   required from ‘static bool ceres::internal::AutoDiff<Functor, T, N0, N1, N2, N3, N4, N5, N6, N7, N8, N9>::Differentiate(const Functor&, const T* const*, int, T*, T**) [with Functor = cartographer::mapping::pose_graph::SpaCostFunction2D; T = double; int N0 = 3; int N1 = 3; int N2 = 0; int N3 = 0; int N4 = 0; int N5 = 0; int N6 = 0; int N7 = 0; int N8 = 0; int N9 = 0]’
/usr/include/ceres/autodiff_cost_function.h:211:66:   required from ‘bool ceres::AutoDiffCostFunction<CostFunctor, kNumResiduals, N0, N1, N2, N3, N4, N5, N6, N7, N8, N9>::Evaluate(const double* const*, double*, double**) const [with CostFunctor = cartographer::mapping::pose_graph::SpaCostFunction2D; int kNumResiduals = 3; int N0 = 3; int N1 = 3; int N2 = 0; int N3 = 0; int N4 = 0; int N5 = 0; int N6 = 0; int N7 = 0; int N8 = 0; int N9 = 0]’
/home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:349:1:   required from here
/home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers_impl.h:45:19: error: no match for ‘operator*=’ (operand types are ‘std::array<ceres::Jet<double, 6>, 3ul>::value_type {aka ceres::Jet<double, 6>}’ and ‘double’)
   scaled_error[0] *= translation_weight;
                   ^
In file included from /usr/include/ceres/internal/autodiff.h:145:0,
                 from /usr/include/ceres/autodiff_cost_function.h:132,
                 from /usr/include/ceres/ceres.h:37,
                 from /home/tong/Desktop/cartographer/cartographer/common/math.h:25,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/submaps.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/pose_graph_interface.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/optimization_problem_interface.h:28,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.h:31,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:17:
/usr/include/ceres/jet.h:217:14: note: candidate: ceres::Jet<T, N>& ceres::Jet<T, N>::operator*=(const ceres::Jet<T, N>&) [with T = double; int N = 6]
   Jet<T, N>& operator*=(const Jet<T, N> &y) {
              ^
/usr/include/ceres/jet.h:217:14: note:   no known conversion for argument 1 from ‘double’ to ‘const ceres::Jet<double, 6>&’
In file included from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers.h:87:0,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/landmark_cost_function_2d.h:23,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:30:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers_impl.h:46:19: error: no match for ‘operator*=’ (operand types are ‘std::array<ceres::Jet<double, 6>, 3ul>::value_type {aka ceres::Jet<double, 6>}’ and ‘double’)
   scaled_error[1] *= translation_weight;
                   ^
In file included from /usr/include/ceres/internal/autodiff.h:145:0,
                 from /usr/include/ceres/autodiff_cost_function.h:132,
                 from /usr/include/ceres/ceres.h:37,
                 from /home/tong/Desktop/cartographer/cartographer/common/math.h:25,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/submaps.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/pose_graph_interface.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/optimization_problem_interface.h:28,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.h:31,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:17:
/usr/include/ceres/jet.h:217:14: note: candidate: ceres::Jet<T, N>& ceres::Jet<T, N>::operator*=(const ceres::Jet<T, N>&) [with T = double; int N = 6]
   Jet<T, N>& operator*=(const Jet<T, N> &y) {
              ^
/usr/include/ceres/jet.h:217:14: note:   no known conversion for argument 1 from ‘double’ to ‘const ceres::Jet<double, 6>&’
In file included from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers.h:87:0,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/landmark_cost_function_2d.h:23,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:30:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers_impl.h:47:19: error: no match for ‘operator*=’ (operand types are ‘std::array<ceres::Jet<double, 6>, 3ul>::value_type {aka ceres::Jet<double, 6>}’ and ‘double’)
   scaled_error[2] *= rotation_weight;
                   ^
In file included from /usr/include/ceres/internal/autodiff.h:145:0,
                 from /usr/include/ceres/autodiff_cost_function.h:132,
                 from /usr/include/ceres/ceres.h:37,
                 from /home/tong/Desktop/cartographer/cartographer/common/math.h:25,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/submaps.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/pose_graph_interface.h:24,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/optimization_problem_interface.h:28,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.h:31,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:17:
/usr/include/ceres/jet.h:217:14: note: candidate: ceres::Jet<T, N>& ceres::Jet<T, N>::operator*=(const ceres::Jet<T, N>&) [with T = double; int N = 6]
   Jet<T, N>& operator*=(const Jet<T, N> &y) {
              ^
/usr/include/ceres/jet.h:217:14: note:   no known conversion for argument 1 from ‘double’ to ‘const ceres::Jet<double, 6>&’
In file included from /home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers.h:87:0,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/landmark_cost_function_2d.h:23,
                 from /home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc:30:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/pose_graph/cost_helpers_impl.h: In instantiation of ‘std::array<T, 6ul> cartographer::mapping::pose_graph::ScaleError(std::array<T, 6ul>, double, double) [with T = ceres::Jet<double, 13>]’:
/home/tong/Desktop/cartographer/cartographer/mapping/internal/2d/pose_graph/landmark_cost_function_2d.h:61:46:   required from ‘bool cartographer::mapping::pose_graph::LandmarkCostFunction2D::operator()(const T*, const T*, const T*, const T*, T*) const [with T [... cut ...]
CMakeFiles/cartographer.dir/build.make:1517: recipe for target 'CMakeFiles/cartographer.dir/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc.o' failed
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/internal/2d/pose_graph/optimization_problem_2d.cc.o] Error 1
CMakeFiles/Makefile2:1435: recipe for target 'CMakeFiles/cartographer.dir/all' failed
make[1]: *** [CMakeFiles/cartographer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

manifolds, 2018-03-23:
With the cschuet-patch-1 branch, build successfully

cschuet, 2018-03-23:
Seems to be related to Ceres Jets. @pifon2a, Alex, can you take a look?

pifon2a, 2018-03-23:
@manifolds, how are you building your Ceres lib? Do you follow [our readme](https://google-cartographer.readthedocs.io/en/latest/)? Do you use `cmake .. -G Ninja -DCXX11=ON` and also install all the libs as described?

cschuet, 2018-03-23:
@pifon2a our CQ pipeline is also failing with the same error. I will take a look.

pifon2a, 2018-03-23:
It is interesting, because when the changes to Jets casts were submitted, all tests passed.

cschuet, 2018-03-23:
Jenkins is using a different Docker file though, i.e. jenkins/Dockerfile.kinetic

cschuet, 2018-03-23:
The failure can be reproduced using

```
docker build -f jenkins/Dockerfile.kinetic -t kinetic-jenkins-slave --build-arg CACHEBUST=$(date +%s)  .
```

manifolds, 2018-03-23:
No I use the system ceres.

manifolds, 2018-03-23:
Build Cartographer in Archlinux with system ceres, with the following error,

`/usr/bin/ld: CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o: undefined reference to symbol '_ZN7testing8internal24FormatMatcherDescriptionEbPKcRKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS9_EE'
/usr/lib/libgmock.so.0: error adding symbols: DSO missing from command line
collect2: 错误：ld 返回 1
make[2]: *** [CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/build.make:124：cartographer.transform.transform_interpolation_buffer_test] 错误 1
make[1]: *** [CMakeFiles/Makefile2:68：CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2`

glonns, 2018-03-26:
I had the same problem when compiling against Ceres 1.12.0. Using 1.13.0 instead (as mentioned in the readme) solved it.

closed 2018-03-26 by pifon2a