cartographer-project/cartographer #1019: Empty range data prohibits new range data from being added

glonns, 2018-03-26:
When `RangeDataCollator::CropAndMerge` encounters empty range data, it doesn't seem to remove it from `id_to_pending_data_`. This causes `RangeDataCollator::AddRangeData` to enter the conditional on lines 33-41 during its next execution:

```
  if (id_to_pending_data_.count(sensor_id) != 0) {
    current_start_ = current_end_;
    // When we have two messages of the same sensor, move forward the older of
    // the two (do not send out current).
    current_end_ = id_to_pending_data_.at(sensor_id).time;
    auto result = CropAndMerge();
    id_to_pending_data_.emplace(sensor_id, timed_point_cloud_data);
    return result;
  }
```

Here, `id_to_pending_data_[sensor_id]` still contains the empty range data. Running `emplace` does not overwrite it with new data, i.e. `timed_point_cloud_data` is basically ignored. This means that `LocalTrajectoryBuilder3D::AddRangeData` won't receive new data anymore after encountering empty range data.

gaschler, 2018-03-26:
Thanks for reporting.

gaschler, 2018-03-26:
For input coming from ROS, this is fixed by https://github.com/googlecartographer/cartographer_ros/pull/767.

gaschler, 2018-04-04:
@glonns, thanks again for giving such precise error description.
Can you please verify that #1036 solves it?

glonns, 2018-04-04:
Yes, this seems to fix the problem! Thanks!

closed 2018-04-04 by glonns