cartographer-project/cartographer #1021: PoseGraphInterface can mostly be const

gaschler, 2018-03-27:
Currently, all methods in `PoseGraphInterface` are not const, even though most of them can be.
I think the reason was that they require a read mutex, but semantically, they are const.

SirVer, 2018-04-24:
That is a philosophical decision. We decided way back that in Cartographer, const should mean the same for the compiler and us, so we never used mutable ever. Mutable is also a slippery slope, diluting the possibility to correctly reason about class state from the outside. I think the Chromium project landed on the other side, using mutable all over the place and trying to express with `const` semantic constness.

Further reading, though a bit outdated:
- https://www.amazon.de/More-Effective-Addison-Wesley-Professional-Computing-ebook/dp/B004VSMDNY

ojura, 2018-04-24:
I stumbled into this talk, which I found interesting:
https://channel9.msdn.com/posts/C-and-Beyond-2012-Herb-Sutter-You-dont-know-blank-and-blank

sebastianklose, 2018-05-15:
@ojura - this is a nice talk, thanks for sharing.

For those who can't spare the time to watch it, the summary is that in C+11 the meaning of `const` and `mutable` has change away from "logically const" to "thread-safe" (bitwise const or internally synchronized). Meaning, declaring a synchronized object (like a mutex) `mutable` is concise and actually desired.

That being said, I will move forward with changing `Cartographer` classes to reflect this, as it also makes other classes using these object more sane in terms of `const`-ness.

PR #1156 (2018-05-17): More const PoseGraph interface
This PR makes getters in `PoseGraph*` `const`. 
Before most getters where marked non-const because of the need to acquire the mutex for thread safety. This forces all code using the PoseGraph to pass it around as a non-const object even if the consumer is not altering the object. 
By making the mutexes `mutable` (for details see [this talk by Herb Sutter](https://channel9.msdn.com/posts/C-and-Beyond-2012-Herb-Sutter-You-dont-know-blank-and-blank)) we can make getters and methods that do not change the PoseGraph const and thus enable consumer APIs to express how t [... cut]
commit d419fe8f (2018-05-17): More const PoseGraph interface (#1156)

Making getters and non-modifying methods  in PoseGraph* const.

Before most getters where marked non-const because of the need to acquire the mutex for thread safety. This forces all code using the PoseGraph to pass it around as a non-const object even if the consumer is not altering the object.
By making the mutexes mutable we can make getters and methods that do not change the PoseGraph const and thus enable consumer APIs to express how they are going to use the PoseGraph as well.

fixes #1021
closed 2018-05-17 by sebastianklose