cartographer-project/cartographer #1026: Check fails for non-sorted range data added to queue

gaschler, 2018-03-29:
How to reproduce:

Download https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b0-2014-10-07-12-43-25.bag
```
roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=$HOME/Downloads/b0-2014-10-07-12-43-25.bag no_rviz:=true
```

After 20-30 seconds, check fails:
```
F0329 10:00:02.341856 210410 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (635482827007268794 vs. 635482827007264928) Non-sorted data added to queue: '(0, echoes)'
[FATAL] [1522310402.342357850, 1412685900.726752800]: F0329 10:00:02.000000 210410 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (635482827007268794 vs. 635482827007264928) Non-sorted data added to queue: '(0, echoes)'
*** Check failure stack trace: ***
    @     0x7f461b6f6aed  google::LogMessage::Fail()
    @     0x7f461b6f8953  google::LogMessage::SendToLog()
    @     0x7f461b6f6648  google::LogMessage::Flush()
    @     0x7f461b6f93b9  google::LogMessageFatal::~LogMessageFatal()
    @     0x55f6a2ac0e66  (unknown)
```

Same thing happens for https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b0-2014-10-07-12-34-42.bag. Of 35 2D bag files I tested, 12 did not finish, I didn't check the logs, though.

gaschler, 2018-04-05:
Fixed by https://github.com/googlecartographer/cartographer_ros/pull/806.

closed 2018-04-05 by gaschler

MichaelGrupp, 2018-08-03:
What are your issues with the IMU?

zhengshunkai, 2018-08-06:
**Hi,I am using the version around December 2017.SLAM(2d_Laser_R2000+IMU_LPMS) boot from the start.
The following problems were found during use:**

[33m[ WARN] [1531721257.999882894]: Detected jump back in time of 1.3227s. Clearing TF buffer.[0m
F0716 14:07:39.336696  2118 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636673180593344190 vs. 636673180580382243) Non-sorted data added to queue: '(1, scan)'
[31m[FATAL] [1531721259.336893549]: F0716 14:07:39.000000  2118 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636673180593344190 vs. 636673180580382243) Non-sorted data added to queue: '(1, scan)'[0m
*** Check failure stack trace: ***
    @     0x7f557b5575cd  google::LogMessage::Fail()
    @     0x7f557b559433  google::LogMessage::SendToLog()
    @     0x7f557b55715b  google::LogMessage::Flush()
    @     0x7f557b559e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x5ffddc  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x600cb4  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x6057c5  cartographer::sensor::Collator::AddSensorData()
    @           0x6738a8  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @           0x59cf21  cartographer_ros::SensorBridge::HandleImuMessage()
    @           0x56eda7  cartographer_ros::Node::HandleImuMessage()
   @           0x57e2fe  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5875f9  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f55791c444d  ros::SubscriptionQueue::call()
    @     0x7f557916a210  ros::CallbackQueue::callOneCB()
    @     0x7f557916b683  ros::CallbackQueue::callAvailable()
    @     0x7f55791c8511  ros::SingleThreadedSpinner::spin()
    @     0x7f55791ad4cb  ros::spin()
    @           0x568d00  cartographer_ros::(anonymous namespace)::Run()
    @           0x566274  main
    @     0x7f5577bba830  __libc_start_main
    @           0x5688b9  _start
    @              (nil)  (unknown)

**Then I modify as described here,but  a new problem appears,as following:**

[33m[ WARN] [1533277621.467822841]: Detected jump back in time of 0.410554s. Clearing TF buffer.[0m
odom.pose.pose.orientationx: 0
[33m[ WARN] [1533277617.850859332]: W0803 14:26:57.000000  2114 pose_extrapolator.cc:161] Queue too short for velocity estimation. Queue duration: 0.0004993 ms[0m
[0m[ INFO] [1533277618.731820190]: I0803 14:26:58.000000  2114 collated_trajectory_builder.cc:74] imu rate: 101.26 Hz 9.88e-03 s +/- 9.54e-03 s (pulsed at 102.39% real time)[0m
[0m[ INFO] [1533277618.731881239]: I0803 14:26:58.000000  2114 collated_trajectory_builder.cc:74] scan rate: 34.96 Hz 2.86e-02 s +/- 6.33e-03 s (pulsed at 105.26% real time)[0m
[33m[ WARN] [1533277621.467851617]: Detected jump back in time of 0.410525s. Clearing TF buffer.[0m
[33m[ WARN] [1533277621.472593238]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744214769568[0m
[33m[ WARN] [1533277621.514552894]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744214980982[0m
[33m[ WARN] [1533277621.523875008]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744215360790[0m
[33m[ WARN] [1533277621.566511449]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744215550886[0m
[33m[ WARN] [1533277621.610513282]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744215931681[0m
[33m[ WARN] [1533277621.610688940]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744216121756[0m
[33m[ WARN] [1533277621.654525430]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744216501530[0m
[33m[ WARN] [1533277621.665506976]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744216777063[0m
[33m[ WARN] [1533277621.706530724]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current subdivision time 636688744217073689[0m
[33m[ WARN] [1533277621.750478753]: W0803 14:27:01.000000  2114 sensor_bridge.cc:165] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636688744218616403 is not before current su [... cut ...]
F0803 14:27:06.426424  2114 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636688744263895302 vs. 636688744263891087) Non-sorted data added to queue: '(1, imu)'
[31m[FATAL] [1533277626.426657824]: F0803 14:27:06.000000  2114 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636688744263895302 vs. 636688744263891087) Non-sorted data added to queue: '(1, imu)'

**Is it because only the time stamp of scan is processed, but the time stamp of IMU is not ?
thx**

MichaelGrupp, 2018-08-07:
First a general tip: use markdown code blocks for logs, makes it much easier to read. https://help.github.com/articles/creating-and-highlighting-code-blocks/

It looks like your IMU data has a time jump. I don't know about your IMU model or driver unfortunately.

mkhansen-intel, 2019-01-30:
When running on the turtlebot3, using cartographer 1.0.0 and cartographer_ros 1.0.0, I see the same problem as @zhengshunkai - 
```
F0130 09:12:12.036525  9197 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636844651318901501 vs. 636844651318861515) Non-sorted data added to queue: '(0, imu)'
F0130 09:12:12.000000  9197 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636844651318901501 vs. 636844651318861515) Non-sorted data added to queue: '(0, imu)'
```
Is there a work-around or fix for this?

mkhansen-intel, 2019-01-30:
I was actually able to ignore the failure with this patch:
```
cartographer_ros$ git diff
diff --git a/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc b/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc
index 0401587..bec47fe 100644
--- a/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc
+++ b/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc
@@ -138,6 +138,13 @@ void SensorBridge::HandleImuMessage(const std::string& sensor_id,
                                     const sensor_msgs::msg::Imu::ConstSharedPtr& msg) {
   std::unique_ptr<carto::sensor::ImuData> imu_data = ToImuData(msg);
   if (imu_data != nullptr) {
+    if (imu_data->time <= previous_imu_time_) {
+      LOG(WARNING) << "Ignored imu message from sensor "
+                   << sensor_id << " because previous imu data time "
+                   << previous_imu_time_ << " is not before current imu data time "
+                   << imu_data->time;
+    }
+    previous_imu_time_ = imu_data->time;
     trajectory_builder_->AddSensorData(
         sensor_id,
         carto::sensor::ImuData{imu_data->time, imu_data->linear_acceleration,
diff --git a/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h b/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h
index f0b91b0..653b4b2 100644
--- a/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h
+++ b/ros2_cartographer/cartographer_ros/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h
@@ -87,6 +87,7 @@ class SensorBridge {
                          const ::cartographer::sensor::TimedPointCloud& ranges);

   const int num_subdivisions_per_laser_scan_;
+  cartographer::common::Time previous_imu_time_;
   std::map<std::string, cartographer::common::Time> sensor_to_previous_subdivision_time_;
   const TfBridge tf_bridge_;
   ::cartographer::mapping::TrajectoryBuilderInterface* const trajectory_builder_;
```
If this is something the cartographer team would like to integrate, I can submit this as a PR.

ojura, 2019-01-30:
The issue with unsorted sensor timestamps pops up occasionally (not just for the IMU, for all sensors), and the stance of Cartographer team (disclaimer: I am a third party contributor) was always that Cartographer expects well-formed sensor data, i.e. strictly increasing sensor stamps, and crashes hard to alert you that your sensor data is malformed.

MichaelGrupp, 2019-01-31:
The last time I ran into this I thought that the check in the library makes sense, but that there should be a protection against wrong input one layer above in the ROS node. Just like @mkhansen-intel has done in his patch, which I think makes absolutely sense. 

My opinion is that invariants in low-level code should be handled strictly, but this doesn't contradict with a protection against wrong input from outside in executables that wrap such code. For the same reason, I tried to refactor the ROS services such that they return an error status for wrong input instead of crashing on the lower level because of a user typo (which was pretty frustrating). In this case here, an "attacker" currently just needs to `rostopic pub` a garbage message to kill Cartographer, which is unnecessary brittle IMO.

ojura, 2019-01-31:
Well, the same can be said for the rest of ROS1. Publish 10000 to `/cmd_vel` and chaos ensues :) ROS2 solves that with protecting the communication channel itself, I don't think that Cartographer should be trying to solve that on its own.

But on the other hand, I agree that a global, strictly opt-in "ignore non-increasing stamps" flag in cartographer_ros might make life easier in situations like this. Maybe open an RFC to revisit this?