cartographer-project/cartographer #1056: map_filename error

robotsoft, 2018-04-11:
Hello,

When I tried to run demo_backpack_3d_localization.launch according to the document at https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#pure-localization

# Run pure localization:
roslaunch cartographer_ros demo_backpack_3d_localization.launch \
   load_state_filename:=${HOME}/Downloads/b3-2016-04-05-13-54-42.bag.pbstream \
   bag_filename:=${HOME}/Downloads/b3-2016-04-05-15-52-20.bag

I have an error message, "requires the 'map_filename' arg to be set".

Thus, I changed "load_state_filename" to "map_filename." However, I have the other error message "[cartographer_node-2] process has died". 

Could you let me know what I missed?

I tested it on Ubuntu 16.04 and ROS Kinetic. There is no problem when I tried to generate the map using offline_backpack_3d.launch.

gaschler, 2018-04-12:
Can you please reproduce this with the offline node (which is usually easier to set up reproducibly)?

```
<launch>
  <param name="/use_sim_time" value="true" />
    <node name="rviz" pkg="rviz" type="rviz" required="true"
        args="-d $(find cartographer_ros)/configuration_files/demo_3d.rviz" />
  <node name="cartographer_offline_node" pkg="cartographer_ros"
      required="true"
      type="cartographer_offline_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basenames backpack_3d.lua
          -urdf_filenames $(find cartographer_ros)/urdf/backpack_3d.urdf
          -load_state_filename ABSOLUTE_PBSTREAM_PATH
          -bag_filenames ABSOLUTE_BAG_PATH"
      output="screen">
    <remap from="points2_1" to="horizontal_laser_3d" />
    <remap from="points2_2" to="vertical_laser_3d" />
  </node>
</launch>
```

Then, please test again with a debugger by adding `launch-prefix="gdb --args"` to the node attributes.

Please post the log messages and also stack traces from the debugger. This could be a segfault. Thank you.

robotsoft, 2018-04-12:
The new launch file works. Thank you.

closed 2018-04-16 by cschuet