cartographer-project/cartographer #1062: lidar frame is not moving

robotsoft, 2018-04-12:
Hello,

I'm trying to run 3D SLAM with one 3D lidar and IMU. When I tried to build a map, the lidar frame is not moving and fixed in the map frame. 

I modified urdf/backpack_3d.urdf and configuration_files/backpack_3d.lua and configuration_files/demo_3d.rviz.

In urdf/backpack_3d.urdf, I delete  `vertical_vlp16_link` and changed `horizontal_vlp16_link` to the frame_id of my point cloud data. 

In configuration_files/backpack_3d.lua, I changed `num_point_clouds` to 1. 

In configuration_files/demo_3d.rviz, I set `vertical_vlp16_link` as false and changed `horizontal_vlp16_link` to the frame id of my point cloud data.

Could you let me know what I missed or did wrong?

gaschler, 2018-04-13:
What are the log messages?
Do you see the accumulated point cloud in rviz (not the input messages)?
Please give more details following the issue template https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/.github/ISSUE_TEMPLATE

robotsoft, 2018-04-13:
Thank you for letting me know the issue template. The output of the validation tool are as follows.

E0413 11:38:57.695830 22866 rosbag_validate_main.cc:350] frame_id "odom" on topic odometry/filtered has serialization time 1523485211.273044274 but sensor time 1523485181.300542593 differing by -29.9725 s.
E0413 11:38:57.696030 22866 rosbag_validate_main.cc:350] frame_id "base_link" on topic /imu/data has serialization time 1523485211.278231944 but sensor time 1523485181.345422302 differing by -29.9328 s.
E0413 11:38:57.696043 22866 rosbag_validate_main.cc:350] frame_id "imu_link" on topic /imu_um7/data has serialization time 1523485211.278242416 but sensor time 1523485181.345422302 differing by -29.9328 s.
E0413 11:38:59.495668 22866 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.5157 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0413 11:38:59.495726 22866 rosbag_validate_main.cc:389] IMU data (frame_id: "base_link") has a large gap, largest is 0.065963 s, recommended is [0.0005, 0.005] s with no jitter.
I0413 11:38:59.495906 22866 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu/data" (frame_id: "base_link"):
Count: 3895  Min: 0.039001  Max: 0.065963  Mean: 0.052505
[0.039001, 0.041698)	                    	Count: 3 (0.077022%)	Total: 3 (0.077022%)
[0.041698, 0.044394)	                    	Count: 20 (0.513479%)	Total: 23 (0.590501%)
[0.044394, 0.047090)	                  ##	Count: 332 (8.523748%)	Total: 355 (9.114249%)
[0.047090, 0.049786)	                  ##	Count: 340 (8.729140%)	Total: 695 (17.843390%)
[0.049786, 0.052482)	               #####	Count: 911 (23.388960%)	Total: 1606 (41.232349%)
[0.052482, 0.055178)	           #########	Count: 1691 (43.414635%)	Total: 3297 (84.646980%)
[0.055178, 0.057875)	                   #	Count: 284 (7.291399%)	Total: 3581 (91.938385%)
[0.057875, 0.060571)	                   #	Count: 292 (7.496791%)	Total: 3873 (99.435173%)
[0.060571, 0.063267)	                    	Count: 20 (0.513479%)	Total: 3893 (99.948654%)
[0.063267, 0.065963]	                    	Count: 2 (0.051348%)	Total: 3895 (100.000000%)
E0413 11:38:59.495944 22866 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.065963 s, recommended is [0.0005, 0.005] s with no jitter.
I0413 11:38:59.496101 22866 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu_um7/data" (frame_id: "imu_link"):
Count: 3895  Min: 0.039001  Max: 0.065963  Mean: 0.052505
[0.039001, 0.041698)	                    	Count: 3 (0.077022%)	Total: 3 (0.077022%)
[0.041698, 0.044394)	                    	Count: 20 (0.513479%)	Total: 23 (0.590501%)
[0.044394, 0.047090)	                  ##	Count: 332 (8.523748%)	Total: 355 (9.114249%)
[0.047090, 0.049786)	                  ##	Count: 340 (8.729140%)	Total: 695 (17.843390%)
[0.049786, 0.052482)	               #####	Count: 911 (23.388960%)	Total: 1606 (41.232349%)
[0.052482, 0.055178)	           #########	Count: 1691 (43.414635%)	Total: 3297 (84.646980%)
[0.055178, 0.057875)	                   #	Count: 284 (7.291399%)	Total: 3581 (91.938385%)
[0.057875, 0.060571)	                   #	Count: 292 (7.496791%)	Total: 3873 (99.435173%)
[0.060571, 0.063267)	                    	Count: 20 (0.513479%)	Total: 3893 (99.948654%)
[0.063267, 0.065963]	                    	Count: 2 (0.051348%)	Total: 3895 (100.000000%)
I0413 11:38:59.496523 22866 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "odometry/filtered" (frame_id: "odom"):
Count: 10226  Min: 0.000000  Max: 0.058992  Mean: 0.020003
[0.000000, 0.005899)	             #######	Count: 3705 (36.231174%)	Total: 3705 (36.231174%)
[0.005899, 0.011798)	                   #	Count: 505 (4.938392%)	Total: 4210 (41.169567%)
[0.011798, 0.017698)	                  ##	Count: 1047 (10.238607%)	Total: 5257 (51.408176%)
[0.017698, 0.023597)	                  ##	Count: 783 (7.656953%)	Total: 6040 (59.065128%)
[0.023597, 0.029496)	                   #	Count: 567 (5.544690%)	Total: 6607 (64.609818%)
[0.029496, 0.035395)	                   #	Count: 463 (4.527675%)	Total: 7070 (69.137489%)
[0.035395, 0.041294)	                ####	Count: 1839 (17.983572%)	Total: 8909 (87.121063%)
[0.041294, 0.047193)	                   #	Count: 372 (3.637786%)	Total: 9281 (90.758850%)
[0.047193, 0.053093)	                   #	Count: 751 (7.344025%)	Total: 10032 (98.102875%)
[0.053093, 0.058992]	                    	Count: 194 (1.897125%)	Total: 10226 (100.000000%)
E0413 11:38:59.496564 22866 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.102153 s, recommended is [0.0005, 0.05] s with no jitter.
I0413 11:38:59.496670 22866 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 2004  Min: 0.101844  Max: 0.102153  Mean: 0.102001
[0.101844, 0.101875)	                    	Count: 8 (0.399202%)	Total: 8 (0.399202%)
[0.101875, 0.101906)	                    	Count: 47 (2.345309%)	Total: 55 (2.744511%)
[0.101906, 0.101937)	                   #	Count: 128 (6.387226%)	Total: 183 (9.131737%)
[0.101937, 0.101968)	                  ##	Count: 208 (10.379242%)	Total: 391 (19.510979%)
[0.101968, 0.101999)	              ######	Count: 569 (28.393213%)	Total: 960 (47.904190%)
[0.101999, 0.102029)	              ######	Count: 578 (28.842316%)	Total: 1538 (76.746506%)
[0.102029, 0.102060)	                  ##	Count: 250 (12.475050%)	Total: 1788 (89.221558%)
[0.102060, 0.102091)	                  ##	Count: 152 (7.584830%)	Total: 1940 (96.806389%)
[0.102091, 0.102122)	                   #	Count: 54 (2.694611%)	Total: 1994 (99.500999%)
[0.102122, 0.102153]	                    	Count: 10 (0.499002%)	Total: 2004 (100.000000%)

robotsoft, 2018-04-13:
Yes, I saw some point clouds are accumulated in rviz.

gaschler, 2018-04-16:
> E0413 11:38:57.696030 22866 rosbag_validate_main.cc:350] frame_id "base_link" on topic /imu/data has serialization time 1523485211.278231944 but sensor time 1523485181.345422302 differing by [...]

Your serialized bag files have conflicting message (`ros::Time`) and header (sensor) time stamps.
Before pose is published by cartographer_ros, it extrapolates to `ros::Time:now`, in your case 30 s, so all published transforms will be useless.

Easiest fix would be to record another bag file on the machine where the sensors are.

gaschler, 2018-04-16:
Also, please post the log messages.

robotsoft, 2018-04-20:
Hi I collected IMU and Lidar data in the same machine. The output of the validation tool are as follows.

```
E0420 11:22:53.282181 21162 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1524001898.105818058 but sensor time 1524001866.291658640 differing by -31.8142 s.
E0420 11:22:53.283152 21162 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1524001898.125624690 but sensor time 1524001866.331605911 differing by -31.794 s.
E0420 11:22:53.291318 21162 rosbag_validate_main.cc:350] frame_id "odom" on topic /odometry/filtered has serialization time 1524001898.145776731 but sensor time 1524001866.331605911 differing by -31.8142 s.
E0420 11:23:08.845710 21162 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.5165 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0420 11:23:08.845849 21162 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.0650453 s, recommended is [0.0005, 0.005] s with no jitter.
I0420 11:23:08.846256 21162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu_um7_zotac/data" (frame_id: "imu_link"):
Count: 3862  Min: 0.036949  Max: 0.065045  Mean: 0.052473
[0.036949, 0.039759)	                    	Count: 12 (0.310720%)	Total: 12 (0.310720%)
[0.039759, 0.042569)	                    	Count: 3 (0.077680%)	Total: 15 (0.388400%)
[0.042569, 0.045378)	                    	Count: 5 (0.129467%)	Total: 20 (0.517866%)
[0.045378, 0.048188)	                    	Count: 50 (1.294666%)	Total: 70 (1.812532%)
[0.048188, 0.050997)	                 ###	Count: 594 (15.380631%)	Total: 664 (17.193165%)
[0.050997, 0.053807)	        ############	Count: 2387 (61.807354%)	Total: 3051 (79.000519%)
[0.053807, 0.056616)	                ####	Count: 775 (20.067322%)	Total: 3826 (99.067841%)
[0.056616, 0.059426)	                    	Count: 17 (0.440186%)	Total: 3843 (99.508026%)
[0.059426, 0.062236)	                    	Count: 4 (0.103573%)	Total: 3847 (99.611603%)
[0.062236, 0.065045]	                    	Count: 15 (0.388400%)	Total: 3862 (100.000000%)
I0420 11:23:08.847483 21162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/odometry/filtered" (frame_id: "odom"):
Count: 10133  Min: 0.000000  Max: 0.058820  Mean: 0.020000
[0.000000, 0.005882)	             #######	Count: 3664 (36.159084%)	Total: 3664 (36.159084%)
[0.005882, 0.011764)	                    	Count: 3 (0.029606%)	Total: 3667 (36.188690%)
[0.011764, 0.017646)	                    	Count: 0 (0.000000%)	Total: 3667 (36.188690%)
[0.017646, 0.023528)	               #####	Count: 2691 (26.556795%)	Total: 6358 (62.745483%)
[0.023528, 0.029410)	                    	Count: 1 (0.009869%)	Total: 6359 (62.755352%)
[0.029410, 0.035292)	                    	Count: 116 (1.144775%)	Total: 6475 (63.900127%)
[0.035292, 0.041174)	             #######	Count: 3585 (35.379452%)	Total: 10060 (99.279579%)
[0.041174, 0.047056)	                    	Count: 0 (0.000000%)	Total: 10060 (99.279579%)
[0.047056, 0.052938)	                    	Count: 64 (0.631600%)	Total: 10124 (99.911179%)
[0.052938, 0.058820]	                    	Count: 9 (0.088819%)	Total: 10133 (100.000000%)
E0420 11:23:08.847551 21162 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.104106 s, recommended is [0.0005, 0.05] s with no jitter.
I0420 11:23:08.847785 21162 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 1987  Min: 0.099879  Max: 0.104106  Mean: 0.101998
[0.099879, 0.100302)	                    	Count: 1 (0.050327%)	Total: 1 (0.050327%)
[0.100302, 0.100724)	                    	Count: 0 (0.000000%)	Total: 1 (0.050327%)
[0.100724, 0.101147)	                    	Count: 0 (0.000000%)	Total: 1 (0.050327%)
[0.101147, 0.101570)	                    	Count: 1 (0.050327%)	Total: 2 (0.100654%)
[0.101570, 0.101992)	            ########	Count: 799 (40.211372%)	Total: 801 (40.312027%)
[0.101992, 0.102415)	        ############	Count: 1184 (59.587318%)	Total: 1985 (99.899345%)
[0.102415, 0.102838)	                    	Count: 1 (0.050327%)	Total: 1986 (99.949677%)
[0.102838, 0.103261)	                    	Count: 0 (0.000000%)	Total: 1986 (99.949677%)
[0.103261, 0.103683)	                    	Count: 0 (0.000000%)	Total: 1986 (99.949677%)
[0.103683, 0.104106]	                    	Count: 1 (0.050327%)	Total: 1987 (100.000000%)
```
However, only some of point clouds are accumulated at the same position as shown in the figure below.
[image]

The log messages as follows.
```
[ INFO] [1524248941.966825229]: I0420 11:29:01.000000 21373 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/jdx_husky_3d.lua' for 'jdx_husky_3d.lua'.
[ INFO] [1524248941.967034980]: I0420 11:29:01.000000 21373 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1524248941.967059630]: I0420 11:29:01.000000 21373 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[... cut ...]
[ INFO] [1524248960.310527361, 1524002099.779971976]: I0420 11:29:20.000000 21446 constraint_builder_3d.cc:309] Score histogram:
Count: 0
[ INFO] [1524248960.310549796, 1524002099.779971976]: I0420 11:29:20.000000 21446 constraint_builder_3d.cc:310] Rotational score histogram:
Count: 0
[ INFO] [1524248960.310565247, 1524002099.779971976]: I0420 11:29:20.000000 21446 constraint_builder_3d.cc:312] Low resolution score histogram:
Count: 0
[cartographer_offline_node-2] process has finished cleanly
```

cschuet, 2018-04-24:
As @gaschler already mentioned and now the validation tool: The messages in your serialized bag have a conflicting message and header time. That needs to be fixed.

We run a pose extrapolator in cartographer_ros that uses IMU and odometry messages to extrapolate to ros::Time:now which is 30secs ahead of your serialized data. Extrapolating over 30 secs is not going to end well :)

robotsoft, 2018-04-24:
Hi @cschuet,

Thank you for comments. Do you mean the time stamp of odometry should also be synchronized with the time stamp of Lidar and IMU? I thought Catographer uses only IMU and Lidar for building a map.  Dose Catographer use the odometry for building a map?

robotsoft, 2018-04-27:
Hi @cschuet,

Hi I collected IMU and Lidar data only. The output of the validation tool are as follows.

```
E0427 16:01:23.210783  9986 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.545 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0427 16:01:23.211024  9986 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.0650007 s, recommended is [0.0005, 0.005] s with no jitter.
I0427 16:01:23.211069  9986 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/imu_um7_zotac/data" (frame_id: "imu_link"):
Count: 1091  Min: 0.037984  Max: 0.065001  Mean: 0.052599
[0.037984, 0.040686)	                    	Count: 6 (0.549954%)	Total: 6 (0.549954%)
[0.040686, 0.043388)	                    	Count: 0 (0.000000%)	Total: 6 (0.549954%)
[0.043388, 0.046089)	                    	Count: 0 (0.000000%)	Total: 6 (0.549954%)
[0.046089, 0.048791)	                    	Count: 5 (0.458295%)	Total: 11 (1.008249%)
[0.048791, 0.051492)	                   #	Count: 59 (5.407883%)	Total: 70 (6.416132%)
[0.051492, 0.054194)	  ##################	Count: 992 (90.925758%)	Total: 1062 (97.341888%)
[0.054194, 0.056896)	                    	Count: 21 (1.924840%)	Total: 1083 (99.266731%)
[0.056896, 0.059597)	                    	Count: 1 (0.091659%)	Total: 1084 (99.358383%)
[0.059597, 0.062299)	                    	Count: 2 (0.183318%)	Total: 1086 (99.541702%)
[0.062299, 0.065001]	                    	Count: 5 (0.458295%)	Total: 1091 (100.000000%)
E0427 16:01:23.211088  9986 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.102071 s, recommended is [0.0005, 0.05] s with no jitter.
I0427 16:01:23.211119  9986 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 562  Min: 0.101940  Max: 0.102071  Mean: 0.102000
[0.101940, 0.101953)	                    	Count: 1 (0.177936%)	Total: 1 (0.177936%)
[0.101953, 0.101966)	                    	Count: 0 (0.000000%)	Total: 1 (0.177936%)
[0.101966, 0.101979)	                    	Count: 4 (0.711744%)	Total: 5 (0.889680%)
[0.101979, 0.101992)	              ######	Count: 182 (32.384342%)	Total: 187 (33.274021%)
[0.101992, 0.102005)	              ######	Count: 181 (32.206406%)	Total: 368 (65.480431%)
[0.102005, 0.102019)	             #######	Count: 186 (33.096085%)	Total: 554 (98.576515%)
[0.102019, 0.102032)	                    	Count: 6 (1.067616%)	Total: 560 (99.644127%)
[0.102032, 0.102045)	                    	Count: 1 (0.177936%)	Total: 561 (99.822067%)
[0.102045, 0.102058)	                    	Count: 0 (0.000000%)	Total: 561 (99.822067%)
[0.102058, 0.102071]	                    	Count: 1 (0.177936%)	Total: 562 (100.000000%)
```

However, only some of point clouds are accumulated at the same position as shown in the figure below.

[image]

The log messages as follows.

```
[ INFO] [1524870342.921244030]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/jdx_husky_3d.lua' for 'jdx_husky_3d.lua'.
[ INFO] [1524870342.921631456]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1524870342.921686537]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1524870342.921783345]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1524870342.921843515]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1524870342.922010498]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1524870342.922053029]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1524870342.922119736]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1524870342.922159922]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1524870342.922272737]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1524870342.922333467]: I0427 16:05:42.000000 10148 configuration_file_resolver.cc:41] Found '/home/soonhac/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3 [... cut ...]
[ INFO] [1524870345.712144490, 1524782138.871966053]: I0427 16:05:45.000000 10218 constraint_builder_3d.cc:307] 0 computations resulted in 0 additional constraints.
[ INFO] [1524870345.712169054, 1524782138.871966053]: I0427 16:05:45.000000 10218 constraint_builder_3d.cc:309] Score histogram:
Count: 0
[ INFO] [1524870345.712182265, 1524782138.871966053]: I0427 16:05:45.000000 10218 constraint_builder_3d.cc:310] Rotational score histogram:
Count: 0
[ INFO] [1524870345.712197427, 1524782138.871966053]: I0427 16:05:45.000000 10218 constraint_builder_3d.cc:312] Low resolution score histogram:
Count: 0
```

gaschler, 2018-05-02:
Your newest rosbag_validate results look fine.

> [ INFO] [1524870342.941880563]: I0427 16:05:42.000000 10148 map_builder_bridge.cc:130] Added trajectory with ID '0'.
> [ INFO] [1524870342.941917404]: I0427 16:05:42.000000 10148 offline_node.cc:270] [...]

Not a single submap is created, there must be something wrong with your configuration. (Unless you are sending only empty point clouds.)
Please share your config in a fork, as described in the issue template.

robotsoft, 2018-05-03:
Hi @cschuet,

Thank you for your comments. You may download configure files at  https://drive.google.com/open?id=12RblY1O876poGwjyavPXQwngxpfoWH83 and point cloud data at https://drive.google.com/open?id=1t-cRFnmrswQjayyQqly7OxLvgRSxpiey.

gaschler, 2018-07-27:
Closing inactive question.

closed 2018-07-27 by gaschler