cartographer-project/cartographer #1065: Position uncertainty from pose_extrapolator

SyedAzizEnam, 2018-04-13:
Hello, 

I'm trying to use the pose maintained in cartographer for odometry. Only problem is that I'm having trouble finding any kind of uncertainty about the position. 
Is there anything in cartographer that may correspond to a covariance about the position and orientation? Any help would be greatly appreciated!

gaschler, 2018-04-16:
The gradient or curvature of the local scan matcher cost function do not give good insight about what the uncertainty is in practice, not even the score is very helpful. So we do not report anything.

For the `PoseExtrapolator` alone, you can be sure that uncertainty grows with the time since the last scan match.
You could easily measure and publish this "extrapolator drift time" as `FromRos(ros::Time::now()) - local_slam_data->time` near 
https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/node.cc#L187 if you find it helpful for your purposes.

SyedAzizEnam, 2018-04-16:
Thanks for the answers @gaschler :)