cartographer-project/cartographer #1086: install error

jn-c, 2018-04-19:
I'm trying to install cartographer_ros in ubuntu16.04 and ROS kinetic.

But I got this output like this:

> CMake Error at /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:383 (add_library):
>   add_library cannot create imported target "gmock_main" because another
>   target with the same name already exists.
> Call Stack (most recent call first):
>   /opt/ros/kinetic/share/catkin/cmake/all.cmake:147 (include)
>   /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
>   CMakeLists.txt:43 (find_package)
> 
> 
> -- Using Python nosetests: /usr/bin/nosetests-2.7
> -- catkin 0.7.11
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
> -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
> -- Checking for one of the modules 'yaml-cpp>=0.5.1'
> -- Found Lua: /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.2.4") 
> -- Checking for module 'eigen3'
> --   Found eigen3, version 3.2.92
> -- Found eigen: /usr/include/eigen3  
> -- Boost version: 1.58.0
> -- Found the following Boost libraries:
> --   system
> --   filesystem
> --   thread
> --   date_time
> --   iostreams
> --   serialization
> --   chrono
> --   atomic
> --   regex
> -- Checking for module 'libopenni'
> --   Found libopenni, version 1.5.4.0
> -- Found openni: /usr/lib/libOpenNI.so  
> -- Checking for module 'libopenni2'
> --   Found libopenni2, version 2.2.0.3
> -- Found OpenNI2: /usr/lib/libOpenNI2.so  
> ** WARNING ** io features related to pcap will be disabled
> ** WARNING ** io features related to png will be disabled
> -- The imported target "vtkRenderingPythonTkWidgets" references the file
>    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
> but not all the files it references.
> 
> -- The imported target "vtk" references the file
>    "/usr/bin/vtk"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
> but not all the files it references.
> 
> -- Found libusb-1.0: /usr/include  
> -- looking for PCL_COMMON
> -- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
> -- looking for PCL_OCTREE
> -- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
> -- looking for PCL_IO
> -- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
> -- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_common.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_common.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_octree.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_octree.so;/usr/lib/libOpenNI.so;/usr/lib/libOpenNI2.so;vtkImagingStencil;vtkCommonComputationalGeometry;vtkCommonDataModel;vtkCommonMath;vtkCommonCore;vtksys;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkImagingCore;vtkCommonExecutionModel;vtkFiltersAMR;vtkFiltersGeneral;vtkFiltersCore;vtkParallelCore;vtkIOLegacy;vtkIOCore;/usr/lib/x86_64-linux-gnu/libz.so;vtkInteractionWidgets;vtkFiltersHybrid;vtkImagingSources;vtkRenderingCore;vtkCommonColor;vtkFiltersExtraction;vtkFiltersStatistics;vtkImagingFourier;vtkalglib;vtkFiltersGeometry;vtkFiltersSources;vtkFiltersModeling;vtkImagingGeneral;vtkImagingHybrid;vtkIOImage;vtkDICOMParser;vtkmetaio;/usr/lib/x86_64-linux-gnu/libjpeg.so;/usr/lib/x86_64-linux-gnu/libpng.so;/usr/lib/x86_64-linux-gnu/libtiff.so;vtkInteractionStyle;vtkRenderingAnnotation;vtkImagingColor;vtkRenderingFreeType;/usr/lib/x86_64-linux-gnu/libfreetype.so;vtkftgl;vtkRenderingVolume;vtkIOParallelNetCDF;vtkParallelMPI;/usr/lib/x86_64-linux-gnu/libnetcdf_c++.so;/usr/lib/x86_64-linux-gnu/libnetcdf.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5.so;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libdl.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5_hl.so;vtkRenderingOpenGL;vtkIOLSDyna;vtkIOXML;vtkIOGeometry;/usr/lib/x86_64-linux-gnu/libjsoncpp.so;vtkIOXMLParser;/usr/lib/x86_64-linux-gnu/libexpat.so;vtkLocalExample;vtkInfovisCore;vtkGeovisCore;vtkInfovisLayout;vtkViewsCore;vtkproj4;/usr/lib/x86_64-linux-gnu/libpython2.7.so;vtkTestingGenericBridge;/usr/lib/libgl2ps.so;verdict;vtkIOMovie;/usr/lib/x86_64-linux-gnu/libtheoraenc.so;/usr/lib/x86_64-linux-gnu/libtheoradec.so;/usr/lib/x86_64-linux-gnu/libogg.so;vtkFiltersImaging;vtkIOMINC;vtkRenderingLOD;vtkViewsQt;vtkGUISupportQt;vtkViewsInfovis;vtkChartsCore;vtkRenderingContext2D;vtkRenderingLabel;vtkRenderingImage;vtkFiltersFlowPaths;vtkxdmf2;/usr/lib/x86_64-linux-gnu/libxml2.so;vtkFiltersReebGraph;vtkViewsContext2D;vtkIOXdmf2;vtkIOAMR;vtkRenderingContextOpenGL;vtkImagingStatistics;vtkIOParallel;vtkFiltersParallel;vtkIONetCDF;vtkexoIIc;vtkGUISupportQtOpenGL;vtkIOParallelLSDyna;vtkFiltersParallelGeometry;vtkGUISupportQtWebkit;vtkIOPLY;vtkWrappingTools;vtkFiltersHyperTree;vtkRenderingVolumeOpenGL;vtkIOExodus;vtkIOPostgreSQL;vtkIOSQL;sqlite3;vtkWrappingJava;vtkFiltersParallelFlowPaths;vtkFiltersParallelStatistics;vtkFiltersProgrammable;vtkFiltersParallelImaging;vtkRenderingParallelLIC;vtkRenderingLIC;vtkInteractionImage;vtkFiltersPython;vtkWrappingPythonCore;vtkIOParallelExodus;vtkFiltersGeneric;vtkIOVideo;vtkRenderingQt;vtkFiltersTexture;vtkIOInfovis;vtkGUISupportQtSQL;vtkRenderingFreeTypeOpenGL;vtkInfovisBoostGraphAlgorithms;vtkRenderingGL2PS;vtkIOGeoJSON;vtkFiltersVerdict;vtkViewsGeovis;vtkIOImport;vtkTestingIOSQL;vtkPythonInterpreter;vtkIOODBC;vtkIOEnSight;vtkIOMySQL;vtkRenderingMatplotlib;vtkDomainsChemistry;vtkIOExport;vtkFiltersParallelMPI;vtkIOParallelXML;vtkTestingRendering;vtkIOMPIParallel;vtkParallelMPI4Py;vtkFiltersSMP;vtkFiltersSelection;vtkIOVPIC;VPIC;vtkImagingMath;vtkImagingMorphological;vtkRenderingParallel;vtkRenderingFreeTypeFontConfig;vtkIOFFMPEG;vtkIOMPIImage;vtkIOGDAL;optimized;/usr/lib/x86_64-linux-gnu/libpcl_io.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_io.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libOpenNI.so;/usr/lib/libOpenNI2.so;vtkImagingStencil;vtkCommonComputationalGeometry;vtkCommonDataModel;vtkCommonMath;vtkCommonCore;vtksys;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkImagingCore;vtkCommonExecutionModel;vtkFiltersAMR;vtkFiltersGeneral;vtkFiltersCore;vtkParallelCore;vtkIOLegacy;vtkIOCore;/usr/lib/x86_64-linux-gnu/libz.so;vtkInteractionWidgets;vtkFiltersHybrid;vtkImagingSources;vtkRenderingCore;vtkCommonColor;vtkFiltersExtraction;vtkFiltersStatistics;vtkImagingFourier;vtkalglib;vtkFiltersGeometry;vtkFiltersSources;vtkFiltersModeling;vtkImagingGeneral;vtkImagingHybrid;vtkIOImage;vtkDICOMParser;vtkmetaio;/usr/lib/x86_64-linux-gnu/libjpeg.so;/usr/lib/x86_64-linux-gnu/libpng.so;/usr/lib/x86_64-linux-gnu/libtiff.so;vtkInteractionStyle;vtkRenderingAnnotation;vtkImagingColor;vtkRenderingFreeType;/usr/lib/x86_64-linux-gnu/libfreetype.so;vtkftgl;vtkRenderingVolume;vtkIOParallelNetCDF;vtkParallelMPI;/usr/lib/x86_64-linux-gnu/libnetcdf_c++.so;/usr/lib/x86_64-linux-gnu/libnetcdf.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libdl.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/x86_64-linux-gnu/hdf5/serial/lib/libhdf5_hl.so;vtkRenderingOpenGL;vtkIOLSDyna;vtkIOXML;vtkIOGeo [... cut ...]
>   Command '['/home/dauker/ros_library/env.sh', 'cmake', '/home/dauker/cambot_ws/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/dauker/cambot_ws/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/dauker/ros_library', '-G', 'Ninja']' returned non-zero exit status 1
> 
> Reproduce this error by running:
> ==> cd /home/dauker/cambot_ws/build_isolated/cartographer_ros && /home/dauker/ros_library/env.sh cmake /home/dauker/cambot_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/dauker/cambot_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/dauker/ros_library -G Ninja
> 
> Command failed, exiting.

Now what am I supposed to do?

jn-c, 2018-04-19:
@gaschler  I am a muggle in ros, would you mind explaining the first work-around clearly to me? In addition, I tried the second work-around, but still got the same output described in my issue.

gaschler, 2018-04-20:
I don't have enough context to help much here. For instance, I don't know why your logs show that you are linking openni and vtk? One of the many cmake scripts of other libraries perhaps sets some "gmock" target or variables?
Things you can try:

* Try the standard `catkin_make_isolated --use-ninja --pkg cartographer cartographer_ros` first and fix this independently first.
* Hard code all gmock related variables manually in cartographer's FindGmock.cmake (which is also used by cartographer_ros), then go from there.

jn-c, 2018-04-24:
This is the output of `catkin_make_isolated --use-ninja --pkg cartographer cartographer_ros`:

> Base path: /home/dauker/cambot_ws
> Source space: /home/dauker/cambot_ws/src
> Build space: /home/dauker/cambot_ws/build_isolated
> Devel space: /home/dauker/cambot_ws/devel_isolated
> Install space: /home/dauker/ros_library
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> ~~  traversing 15 packages in topological order:
> ~~  - cambot
> ~~  - cambot_description
> ~~  - cambot_gazebo
> ~~  - cambot_pantilt_controller
> ~~  - cambot_purepursuit_controller
> ~~  - cambot_ros
> ~~  - cambot_teleop
> ~~  - cartographer_ros_msgs
> ~~  - ceres-solver (plain cmake)
> ~~  - cartographer (plain cmake)
> ~~  - cartographer_ros
> ~~  - cartographer_cambot
> ~~  - cartographer_rviz
> ~~  - imu_hi219m
> ~~  - rplidar_ros
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> The packages or cmake arguments have changed, forcing cmake invocation
> Skipping package: 'cambot'
> Skipping package: 'cambot_description'
> Skipping package: 'cambot_gazebo'
> Skipping package: 'cambot_pantilt_controller'
> Skipping package: 'cambot_purepursuit_controller'
> Skipping package: 'cambot_ros'
> Skipping package: 'cambot_teleop'
> Skipping package: 'cartographer_ros_msgs'
> Skipping package: 'ceres-solver'
> 
> ==> Processing plain cmake package: 'cartographer'
> ==> Creating build directory: 'build_isolated/cartographer/install'
> ==> Building with env: '/home/dauker/ros_library/env.sh'
> ==> cmake /home/dauker/cambot_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/dauker/ros_library -G Ninja in '/home/dauker/cambot_ws/build_isolated/cartographer/install'
> Unhandled exception of type 'OSError':
> Traceback (most recent call last):
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 981, in build_workspace_isolated
>     number=index + 1, of=len(ordered_packages)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 691, in build_package
>     destdir=destdir, use_ninja=use_ninja
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 513, in build_cmake_package
>     run_command_colorized(cmake_cmd, build_dir, quiet)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 187, in run_command_colorized
>     run_command(cmd, cwd, quiet=quiet, colorize=True, add_env=add_env)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 205, in run_command
>     raise OSError("Failed command '%s': %s" % (cmd, e))
> OSError: Failed command '['/home/dauker/ros_library/env.sh', 'cmake', '/home/dauker/cambot_ws/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/dauker/ros_library', '-G', 'Ninja']': [Errno 2] No such file or directory
> <== Failed to process package 'cartographer': 
>   Failed command '['/home/dauker/ros_library/env.sh', 'cmake', '/home/dauker/cambot_ws/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/dauker/ros_library', '-G', 'Ninja']': [Errno 2] No such file or directory
> Command failed, exiting.

jn-c, 2018-04-25:
I followed the instruction on this [site](https://google-cartographer-ros.readthedocs.io/en/latest/), and I post the output on this [gist](https://gist.github.com/jn-c/5a9f9c6b0abeb97710b96388ea138b1b) after executing command line `catkin_make_isolated --install --use-ninja`.

gaschler, 2018-07-27:
Closing inactive issues.

closed 2018-07-27 by gaschler

whitbrunn, 2025-03-18:
> Interpreting your logs, your project depends on more libraries than just cartographer(_ros).
> 
> In cartographer_ros/CMakeLists.txt, `google_enable_testing()` calls cartographer's FindGMock.cmake, which builds a target "gmock_main". This seems to conflict with other libraries' cmake scripts you are also including.
> 
> (Unfortunately, some version of ROS cmake scripts also creates an imported target with the same name, which lead to similar issues in the past [cartographer-project/cartographer_ros#776](https://github.com/cartographer-project/cartographer_ros/issues/776).)
> 
> There are a couple of work-arounds:
> 
> 1. Make sure GMock is found and cmake variable `GMOCK_LIBRARIES` is set before cartographer's CMake scripts are run.
> 2. If you don't care about tests, comment out `google_enable_testing()` and the section `if (CATKIN_ENABLE_TESTING)`

Comment out `google_enable_testing()` works great for me, with the same error related to GMOCK_LIBRARY!

bufeng-12, 2025-03-18:
您好！我已收到你的邮件！谢谢！          ——曾君