cartographer-project/cartographer #1089: Robot base drifts drastically away after sometime with warning of range_data_collator

hz658832, 2018-04-19:
Hey cartographer community, 

I got a terrible error that my robot base drifts quickly away from the right position, this problem can be seen in the following pic. I am just wondering what is wrong because this problem doesnot happen every time, if I relaunch the whole system, it could word as wished. I noticed this problem twice, last time I just reinstalled the OS (ubuntu 16) after updating, because the problem came out after updating the OS. After reinstalling, it works again. 
The problem happens with the warning of  range_data_collator.cc: Dropped 796 earlier points. And I guess, perhaps there is something wrong with the laser scanner data or other libs. 

Could someone help me or is someone suffering from the same problem?
I am so appreciate for the cartographer team.
[image]

Thx.
Best regards

gaschler, 2018-04-19:
Please follow the default issue template instructions, in particular please post the results of cartographer_rosbag_validate.

hz658832, 2018-05-04:
Hey there,
I found out the problem, nothing to do with the algorithm, my computer doesnot work somehow.
thx.

closed 2018-05-04 by hz658832

bigpigsu, 2018-06-21:
@hz658832 Hey,there,
I have the same problem with you, could you please help me or let me know what's wrong with it?
I am so appreciate for you