cartographer-project/cartographer #1120: Check failed: time >= imu_tracker->time()

nprof50, 2018-04-25:
I've been using cartographer for a few weeks now in conjunction with the turtlebot3 platform. Everything was working fine, but as of about 2 weeks ago, I've been getting this crash regarding the check between the system time and the IMU's reported timestamp (I believe?). I'll paste my logs below with the exact error message. The rosbag file I'm using to test this can be located here: [demo.bag](https://drive.google.com/open?id=1L1gERGyjeO6qm-mJrQNLeNDPQ4cPQlW_). The configuration file I'm using can be located here: [turtlebot_cartographer.lua](https://drive.google.com/open?id=1Q_2DUgYvMU7N54bir82NF9LBhf7LM-uT). Below is the output I get when running a cartographer node with that configuration:

```
started roslaunch server http://SERVER:37475/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /use_sim_time: False

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)

ROS_MASTER_URI=http://SERVER:11311/

process[cartographer_node-1]: started with pid [20873]
process[cartographer_occupancy_grid_node-2]: started with pid [20874]
[ INFO] [1524680591.626482425]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/catkin_ws/src/turtlebot3/turtlebot3_slam/config/turtlebot_cartographer.lua' for 'turtlebot_cartographer.lua'.
[ INFO] [1524680591.626755820]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1524680591.626807225]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1524680591.626872266]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1524680591.626904878]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1524680591.627022391]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1524680591.627050001]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1524680591.627092009]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1524680591.627115575]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1524680591.627190867]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1524680591.627216805]: I0425 14:23:11.000000 20873 configuration_file_resolver.cc:41] Found '/home/username/cart_catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1524680591.634919502]: I0425 14:23:11.000000 20873 submap_2d.cc:203] Added submap 1
[ INFO] [1524680591.634979615]: I0425 14:23:11.000000 20873 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1524680591.900286096]: I0425 14:23:11.000000 20873 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636602747890017584'.
[ INFO] [1524680591.912585707]: I0425 14:23:11.000000 20873 local_trajectory_builder_2d.cc:292] Extrapolator not yet initialized.
[ INFO] [1524680591.912976673]: I0425 14:23:11.000000 20873 local_trajectory_builder_2d.cc:135] Extrapolator is still initializing.
F0425 14:23:25.749990 20873 pose_extrapolator.cc:229] Check failed: time >= imu_tracker->time() (636602748028477307 vs. 636602748028527237) 
[FATAL] [1524680605.750711045]: F0425 14:23:25.000000 20873 pose_extrapolator.cc:229] Check failed: time >= imu_tracker->time() (636602748028477307 vs. 636602748028527237) 
*** Check failure stack trace: ***
    @     0x7fd9e68925cd  google::LogMessage::Fail()
    @     0x7fd9e6894433  google::LogMessage::SendToLog()
    @     0x7fd9e689215b  google::LogMessage::Flush()
    @     0x7fd9e6894e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x5e97ce  cartographer::mapping::PoseExtrapolator::ExtrapolateRotation()
    @           0x5e9d2d  cartographer::mapping::PoseExtrapolator::AddOdometryData()
    @           0x590419  cartographer_ros::Node::HandleOdometryMessage()
    @           0x5a318e  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5ae899  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7fd9e41eae2d  ros::SubscriptionQueue::call()
    @     0x7fd9e41906f8  ros::CallbackQueue::callOneCB()
    @     0x7fd9e41920fb  ros::CallbackQueue::callAvailable()
    @     0x7fd9e41eeef9  ros::SingleThreadedSpinner::spin()
    @     0x7fd9e41d3edb  ros::spin()
    @           0x589d14  cartographer_ros::(anonymous namespace)::Run()
    @           0x586f74  main
    @     0x7fd9e2bdf830  __libc_start_main
    @           0x589839  _start
    @              (nil)  (unknown)
[cartographer_node-1] process has died [pid 20873, exit code -6, cmd /home/username/cart_catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/username/catkin_ws/src/turtlebot3/turtlebot3_slam/config -configuration_basename turtlebot_no_imu_lidar_2d.lua scan:=/scan __name:=cartographer_node __log:=/home/username/.ros/log/43ef0738-48b0-11e8-9f33-00e18c8e5c42/cartographer_node-1.log].
log file: /home/username/.ros/log/43ef0738-48b0-11e8-9f33-00e18c8e5c42/cartographer_node-1*.log
```

This seems to happen randomly when I'm either playing a rosbag, or using the turtlebot physically, but the amount of time before it crashes varies widely. There's a pretty good chance I have an incorrect configuration, so if someone could point me in the right direction as to how to remedy the situation, I'd really appreciate it.

Thanks!

gaschler, 2018-04-26:
If you can reproduce this with a bag file, this should be easy to fix.

The failed check indicates that an older IMU packet arrives in `LocalTrajectoryBuilder`. This is unexpected because IMU packets are sorted by the `Collator`.

Can you please post the text output of `cartographer_rosbag_validate` on a bag file that causes a failed check? Thank you.

nprof50, 2018-04-26:
Thanks for the quick response! Here's the output:

```
W0426 10:04:47.878545 28885 rosbag_validate_main.cc:344] frame_id "odom" on topic /odom has serialization time 1524677987.637164025 but sensor time 1524677987.392919693 differing by -0.244244 s.
W0426 10:04:47.881412 28885 rosbag_validate_main.cc:161] Sensor with frame_id "laser_frame" measurements overlap in time Previous range message, ending at time stamp 1524677988.285722200, must finish before current range message, which ranges from 1524677988.284971829 to 1524677988.429217200
W0426 10:04:47.882550 28885 rosbag_validate_main.cc:161] Sensor with frame_id "laser_frame" measurements overlap in time Previous range message, ending at time stamp 1524677988.860269200, must finish before current range message, which ranges from 1524677988.858589103 to 1524677989.002608200
E0426 10:04:47.883342 28885 rosbag_validate_main.cc:307] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
W0426 10:04:47.884642 28885 rosbag_validate_main.cc:161] Sensor with frame_id "laser_frame" measurements overlap in time Previous range message, ending at time stamp 1524677991.582571200, must finish before current range message, which ranges from 1524677991.581390122 to 1524677991.724024200
E0426 10:04:47.887147 28885 rosbag_validate_main.cc:307] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0426 10:04:47.892658 28885 rosbag_validate_main.cc:307] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0426 10:04:47.972021 28885 rosbag_validate_main.cc:342] frame_id "laser_frame" on topic /scan has serialization time 1524678108.780746378 but sensor time 1524678108.266799200 differing by -0.513947 s.
W0426 10:04:47.980988 28885 rosbag_validate_main.cc:198] Sensor with frame_id "laser_frame" range measurements have longest overlap of 0.0168329 s
E0426 10:04:47.981017 28885 rosbag_validate_main.cc:357] Average IMU linear acceleration is 8.70059 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
E0426 10:04:47.981055 28885 rosbag_validate_main.cc:381] IMU data (frame_id: "imu_link") has a large gap, largest is 0.146868 s, recommended is [0.0005, 0.005] s with no jitter.
I0426 10:04:47.981226 28885 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "imu_link"):
Count: 9430  Min: -0.044247  Max: 0.146868  Mean: 0.013512
[-0.044247, -0.025136)	                    	Count: 5 (0.053022%)	Total: 5 (0.053022%)
[-0.025136, -0.006024)	                    	Count: 3 (0.031813%)	Total: 8 (0.084836%)
[-0.006024, 0.013087)	       #############	Count: 6183 (65.567337%)	Total: 6191 (65.652176%)
[0.013087, 0.032199)	             #######	Count: 3114 (33.022270%)	Total: 9305 (98.674446%)
[0.032199, 0.051310)	                    	Count: 91 (0.965005%)	Total: 9396 (99.639450%)
[0.051310, 0.070422)	                    	Count: 25 (0.265111%)	Total: 9421 (99.904556%)
[0.070422, 0.089533)	                    	Count: 4 (0.042418%)	Total: 9425 (99.946976%)
[0.089533, 0.108645)	                    	Count: 2 (0.021209%)	Total: 9427 (99.968185%)
[0.108645, 0.127756)	                    	Count: 1 (0.010604%)	Total: 9428 (99.978790%)
[0.127756, 0.146868]	                    	Count: 2 (0.021209%)	Total: 9430 (100.000000%)
E0426 10:04:47.981242 28885 rosbag_validate_main.cc:374] Point data (frame_id: "laser_frame") has a large gap, largest is 0.158806 s, recommended is [0.0005, 0.05] s with no jitter.
I0426 10:04:47.981282 28885 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser_frame"):
Count: 889  Min: 0.127232  Max: 0.158806  Mean: 0.143517
[0.127232, 0.130389)	                    	Count: 1 (0.112486%)	Total: 1 (0.112486%)
[0.130389, 0.133547)	                    	Count: 0 (0.000000%)	Total: 1 (0.112486%)
[0.133547, 0.136704)	                    	Count: 0 (0.000000%)	Total: 1 (0.112486%)
[0.136704, 0.139862)	                    	Count: 0 (0.000000%)	Total: 1 (0.112486%)
[0.139862, 0.143019)	               #####	Count: 222 (24.971878%)	Total: 223 (25.084364%)
[0.143019, 0.146176)	     ###############	Count: 664 (74.690666%)	Total: 887 (99.775024%)
[0.146176, 0.149334)	                    	Count: 1 (0.112486%)	Total: 888 (99.887512%)
[0.149334, 0.152491)	                    	Count: 0 (0.000000%)	Total: 888 (99.887512%)
[0.152491, 0.155649)	                    	Count: 0 (0.000000%)	Total: 888 (99.887512%)
[0.155649, 0.158806]	                    	Count: 1 (0.112486%)	Total: 889 (100.000000%)
I0426 10:04:47.981416 28885 rosbag_validate_main.cc:390] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
Count: 3239  Min: -0.017247  Max: 0.147000  Mean: 0.039415
[-0.017247, -0.000822)	                    	Count: 5 (0.154369%)	Total: 5 (0.154369%)
[-0.000822, 0.015602)	                    	Count: 4 (0.123495%)	Total: 9 (0.277864%)
[0.015602, 0.032027)	                    	Count: 9 (0.277864%)	Total: 18 (0.555727%)
[... cut ...]
[0.048452, 0.064877)	                    	Count: 39 (1.204075%)	Total: 3213 (99.197281%)
[0.064877, 0.081301)	                    	Count: 20 (0.617475%)	Total: 3233 (99.814758%)
[0.081301, 0.097726)	                    	Count: 2 (0.061747%)	Total: 3235 (99.876503%)
[0.097726, 0.114151)	                    	Count: 1 (0.030874%)	Total: 3236 (99.907379%)
[0.114151, 0.130575)	                    	Count: 1 (0.030874%)	Total: 3237 (99.938255%)
[0.130575, 0.147000]	                    	Count: 2 (0.061747%)	Total: 3239 (100.000000%)
```

gaschler, 2018-04-26:
> E0426 10:04:47.883342 28885 rosbag_validate_main.cc:307] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.

This really should not happen. Is IMU data modified (or rearranged) after it comes out of the driver?
Does it come from a different device that sends UDP packets?

nprof50, 2018-04-26:
This is the default IMU supplied with the Turtlebot3 embedded in the OpenCR board. Presumably it is a low cost IMU, but it seemed to be working fine with cartographer for a while now. It wasn't until recently that I started experiencing problems.

To go into a bit more detail, the IMU is being read by a raspberry pi 3B+ connected to the OpenCR board. It then publishes to the IMU topic, and the separate host computer running the core receives it. The host computer is the one I'm running cartographer on. My guess is this could create the potential for receiving the IMU data out of order. Aside from that, I don't see any other place where IMU data would be modified.

nprof50, 2018-04-30:
Definitely sounds like a hardware issue then. Thank you, I appreciate the advice!

closed 2018-04-30 by nprof50

znstj, 2024-08-16:
Is it useful to reduce imu_sample_rate?