cartographer-project/cartographer #1123: Multistory SLAM

smbryan, 2018-04-26:
I'm working on a project with a strong need for consistency between floors. Specifically, we need the orientation and position of elevator shafts should be as similar as possible. 

The team is using an older version of Cartographer to produce 2D maps of some large facilities. As you would expect, features can drift between floors which I'd like to minimize. 

Loosely following some of the recent improvements led us to believe that we can achieve something similar to ["Consistent Mapping of Multistory Buildings by Introducing Global Constraints to Graph-based SLAM"](https://pdfs.semanticscholar.org/48a7/9267ff7e2ec150aecdfa7f5eabaab3ac6c6a.pdf) by Karg et al. The paper uses SLAM techniques to generate constraints between floors, improving the maps of each.

How should I go about doing something similar with Cartographer?

ojura, 2018-04-26:
In 3D slam, I had the problem that bad constraints were being generated between floors, because I had a large Z window for searching loop closures. This flattened my trajectory (instead of it being a climbing spiral). 

Your post had me thinking: if we detected that we have a cross-floor loop closure (with a pronounced Z of the residual), perhaps the [constraint cost function](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/optimization/cost_functions/spa_cost_function_3d.h) could be modified to set the Z residual to zero. This way, we would use the x, y and orientation from the constraint pose, while ignoring the height difference. I think this would result in better alignment of the floors.

ojura, 2018-04-27:
Cool :) About 2D slam, I believe the team at some point used a barometer for measuring height. In the GIF animation on the front page, you can see the submaps for the ground level disappearing as the backpack climbs the stairs. I believe this got removed, I don't remember seeing any stuff related to height in 2D slam. @wohe might know more

smbryan, 2018-04-27:
Tuning a test case right now to see how two similar floors might work. I've been getting a lot of erroneous matches, which create some "unique" results. 

I'm thinking we can generate constraints between floors, optimize that solution, and then preserve the output as distinct maps. Might be a useful feature to add to the ROS offline node. Being able to isolate and filter inter-trajectory would be a big help too. 

Finally, I see your parallelepiped and raise [you](http://exploreworldby.blogspot.com/2010/12/leaning-tower-of-pisa.html).

gaschler, 2018-07-27:
Closing inactive issues.

closed 2018-07-27 by gaschler

wohe, 2018-07-27:
Just FYI: 2D Cartographer never had support for multiple floors.

12w2, 2022-08-05:
> 我正在做一个强烈需要楼层之间一致性的项目。具体来说，我们需要电梯井的方向和位置应该尽可能相似。
> 
> 该团队正在使用旧版本的 Cartographer 生成一些大型设施的 2D 地图。如您所料，功能可能会在我希望最小化的楼层之间漂移。
> 
> 松散地遵循最近的一些改进使我们相信我们可以实现类似于Karg 等人[的“通过将全局约束引入基于图形的 SLAM 来实现多层建筑的一致性映射”的东西。](https://pdfs.semanticscholar.org/48a7/9267ff7e2ec150aecdfa7f5eabaab3ac6c6a.pdf)该论文使用 SLAM 技术在楼层之间生成约束，改进每个楼层的地图。
> 
> 我应该如何使用 Cartographer 做类似的事情？

> I'm working on a project with a strong need for consistency between floors. Specifically, we need the orientation and position of elevator shafts should be as similar as possible.
>
> The team is using [...]

Hey,

did you find a solution for producing 2D maps of some large facilities?