cartographer-project/cartographer #1127: Global constraint search not triggered

AlexReimann, 2018-04-27:
Depending on the settings it seems the global constraint search does not get triggered at all for me.

From [the code](https://github.com/googlecartographer/cartographer/blob/c46fe073b48dd1e33d38c285163fbda8fca4fefa/cartographer/mapping/internal/3d/pose_graph_3d.cc#L219) the global constraint search gets triggered only if the local constraint search does not get triggered.
The comment in the code says that local search gets triggered if:

```c++
    // If the node and the submap belong to the same trajectory or if there has
    // been a recent global constraint that ties that node's trajectory to the
    // submap's trajectory, it suffices to do a match constrained to a local
    // search window.
```

Farther looking into it, the `last_connection_time` (= check for `recent global constraint`) seems to get set by [`PoseGraph3D::UpdateTrajectoryConnectivity`](https://github.com/googlecartographer/cartographer/blob/c46fe073b48dd1e33d38c285163fbda8fca4fefa/cartographer/mapping/internal/3d/pose_graph_3d.cc#L335) which gets called mostly from the [`PoseGraph3D::HandleWorkQueue()`](https://github.com/googlecartographer/cartographer/blob/c46fe073b48dd1e33d38c285163fbda8fca4fefa/cartographer/mapping/internal/3d/pose_graph_3d.cc#L354).
This doesn't distinguish between global and local constraints. So the `last_connection_time` seems to include also `local constraints`.

It seems like the global constraint search is not triggered because the local constraint search is triggered more frequently than `global_constraint_search_after_n_seconds`.

For a bigger map the global constraint search would be triggered eventually when the node / trajectory moves out of range for local optimization (I assume), but current nodes / trajectory will not optimized globally?

Is this intended? If yes, what's the reasoning behind it?

cschuet, 2018-04-27:
The code in UpdateTrajectoryConnectivity() explicitly checks for the globalness (or we call them INTER_CONSTRAINTS) of the constraints. So that function can't be called with local constraints.

AlexReimann, 2018-04-27:
Okay thanks, I'll do some more digging then and re-open if I find something.

AlexReimann, 2018-04-30:
[`UpdateTrajectoryConnectivity()`](https://github.com/googlecartographer/cartographer/blob/a3b746ff67011869aaaa1767ee15b468d5db9950/cartographer/mapping/internal/3d/pose_graph_3d.cc#L336) checks for `PoseGraphInterface::Constraint::INTER_SUBMAP` but when the constraint is made, [the `tag` is the same no matter if it is global or not](https://github.com/googlecartographer/cartographer/blob/a3b746ff67011869aaaa1767ee15b468d5db9950/cartographer/mapping/internal/constraints/constraint_builder_3d.cc#L270):

```C++
  constraint->reset(new Constraint{
      submap_id,
      node_id,
      {constraint_transform, options_.loop_closure_translation_weight(),
       options_.loop_closure_rotation_weight()},
       Constraint::INTER_SUBMAP});
```

Also on a side note, from the logic `UpdateTrajectoryConnectivity` is (only executed when there is a constraint)[https://github.com/googlecartographer/cartographer/blob/a3b746ff67011869aaaa1767ee15b468d5db9950/cartographer/mapping/internal/3d/pose_graph_3d.cc#L354], which means it would repeatedly try to make a global constraint in the failure case without caring about `POSE_GRAPH.global_constraint_search_after_n_seconds`?

AlexReimann, 2018-04-30:
Only now got the logic.. I guess the global constraint search is only supposed to be triggered if it doesn't find a constraint between trajectory a and b for `POSE_GRAPH.global_constraint_search_after_n_seconds`.

Also was not clear that the "normal" optimization is also in between multiple trajectories, I thought it was only for the current trajectory.

closed 2018-04-30 by AlexReimann