cartographer-project/cartographer #1134: 3D SLAM with rotating laser on a drone

andrecmilano, 2018-05-01:
Hello,

I already wrote a [note](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/google-cartographer/27Z16PuhrlQ/noC7U7KEBAAJ)  on Cartographer group, where I explained the project that I am working on.

I was curious if in the meanwhile there has been any progress on this topic.

I looked at the demo: **Taurob Tracker**; it uses a rotating laser on a ground robot for localization and mapping. However, when I run the rosbag captured with the drone, using the same configuration of Taurob Tracker, the results are not as good as in the demo. 

The graphical results are reported in this [video](https://youtu.be/YVfvPYZ8d-0). The test has been executed in a University Lab indoor. What is weird to me is that the map is moving with the robot, while it should be static. 

There is anything that I am forgetting in the configuration file? 
Is anyone already working on 3D-SLAM with a drone using rotating 2D-LRF? 

Thank you.

andrecmilano, 2018-05-03:
I have created a drive link with the bag file and all config files (including the results from the test results from the bag file): https://drive.google.com/open?id=1x5Cejk2-YWjuIoW-zdYNhxxS9p5iba88 

Also this is the output from the

gaschler, 2018-05-03:
> The graphical results are reported in this video. The test has been executed in a University Lab indoor. What is weird to me is that the map is moving with the robot, while it should be static.

I think the moving thing is the point cloud, not the map. Toggle visualization of "Submap" to see if there is a map at all.

> I have created a drive link

I can only see the launch file. Generally, it is easier for the community if you fork, add your config, post a link to the diff view that contains exactly what you changed wrt to master.

> 3D-SLAM with a drone using rotating 2D-LRF

A flat point cloud is not sufficient for 3D scan matching. You at least need to accumulate multiple measurements.

Here is an example of 3D SLAM with a tilting 2D lidar on a mobile robot: https://github.com/tu-darmstadt-ros-pkg/cartographer_hector_tracker

andrecmilano, 2018-05-03:
My bad, here is the link again:

https://drive.google.com/open?id=1Y9BS0Px2-4R13FeH3Ro205PcaRJNMm_v

andrecmilano, 2018-05-03:
Here is the result from our bag

> E0502 15:49:07.929915  1217 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525101528.183640971 but sensor time 1525105152.852662991 differing by 3624.67 s.
E0502 15:49:07.930198  1217 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525101528.183654521 but sensor time 1525105152.864614944 differing by 3624.68 s.
E0502 15:49:07.930233  1217 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525101528.183696778 but sensor time 1525105152.873204777 differing by 3624.69 s.
E0502 15:49:07.930538  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019531085656, current one at time 636607019528610490
E0502 15:49:07.930554  1217 rosbag_validate_main.cc:315] Sensor with frame_id "laser" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0502 15:49:07.930565  1217 rosbag_validate_main.cc:326] frame_id "laser" is send on multiple topics. It was seen at least on /most_intense and /last
E0502 15:49:07.930639  1217 rosbag_validate_main.cc:326] frame_id "laser" is send on multiple topics. It was seen at least on /most_intense and /last
E0502 15:49:07.930706  1217 rosbag_validate_main.cc:326] frame_id "laser" is send on multiple topics. It was seen at least on /most_intense and /last
E0502 15:49:07.931183  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019531868081, current one at time 636607019529753212
E0502 15:49:07.931200  1217 rosbag_validate_main.cc:315] Sensor with frame_id "laser" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0502 15:49:07.931670  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019532472242, current one at time 636607019530018622
E0502 15:49:07.931689  1217 rosbag_validate_main.cc:315] Sensor with frame_id "laser" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0502 15:49:07.933331  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019534045173, current one at time 636607019533024191
E0502 15:49:07.934007  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019533784783, current one at time 636607019533784783
E0502 15:49:07.934268  1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019534045173, current one at time 636607019534045173
E0502 15:49:07.935930  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.515371390 equals preceding data with 1081 points.
E0502 15:49:07.936120  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.515371390 equals preceding data with 1081 points.
E0502 15:49:07.936300  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.541097739 equals preceding data with 1081 points.
E0502 15:49:07.936486  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.541097739 equals preceding data with 1081 points.
E0502 15:49:07.937108  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.566795402 equals preceding data with 1081 points.
E0502 15:49:07.944720  1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105154.090548413 equals preceding data with 1081 points.
W0502 15:49:08.736094  1217 rosbag_validate_main.cc:203] Sensor with frame_id "laser" range measurements have longest overlap of 0.837526 s
I0502 15:49:08.736446  1217 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/dynamixel/odometry" (frame_id: "base_link"):
Count: 7039  Min: 0.008089  Max: 0.015576  Mean: 0.010000
[0.008089, 0.008838)	                    	Count: 90 (1.278591%)	Total: 90 (1.278591%)
[0.008838, 0.009587)	                    	Count: 153 (2.173604%)	Total: 243 (3.452195%)
[0.009587, 0.010335)	  ##################	Count: 6508 (92.456314%)	Total: 6751 (95.908508%)
[0.010335, 0.011084)	                   #	Count: 218 (3.097031%)	Total: 6969 (99.005539%)
[0.011084, 0.011832)	                    	Count: 25 (0.355164%)	Total: 6994 (99.360703%)
[0.011832, 0.012581)	                    	Count: 18 (0.255718%)	Total: 7012 (99.616425%)
[0.012581, 0.013330)	                    	Count: 19 (0.269925%)	Total: 7031 (99.886345%)
[... cut ...]
[0.019407, 0.020117)	          ##########	Count: 1819 (52.165184%)	Total: 2489 (71.379410%)
[0.020117, 0.020827)	                  ##	Count: 340 (9.750502%)	Total: 2829 (81.129913%)
[0.020827, 0.021537)	                ####	Count: 647 (18.554632%)	Total: 3476 (99.684540%)
[0.021537, 0.022247)	                    	Count: 8 (0.229424%)	Total: 3484 (99.913963%)
[0.022247, 0.022957)	                    	Count: 2 (0.057356%)	Total: 3486 (99.971321%)
[0.022957, 0.023668]	                    	Count: 1 (0.028678%)	Total: 3487 (100.000000%)

>

gaschler, 2018-05-07:
> E0502 15:49:07.930538 1217 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636607019531085656, current one at time 636607019528610490

Cartographer makes a strong assumption that input header timestamps are sequential (to be precise, sequential for a single given sensor). Fix this first, otherwise nothing will work.

> E0502 15:49:07.937108 1217 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525105153.566795402 equals preceding data with 1081 points.

Shouldn't happen either. How do you connect to the sensor? Avoid any filtering nodes that mess up the sequence or duplicate messages.

> E0502 15:49:07.929915 1217 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525101528.183640971 but sensor time 1525105152.852662991 differing by 3624.67 s.

cartographer_ros' PoseExtrapolator will extrapolate for that time difference and all output will be useless.

Hope this helps.

andrecmilano, 2018-05-07:
We are going to collect a new data-set today and we will see if these issues are solved.

Also to us, the first and second error seems weird.

We will post here the results after the test.

andrecmilano, 2018-05-14:
we collected new datasets, however, some errors remain:

> E0514 11:52:27.603804  7170 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525769467.173695618 but sensor time 1525769466.635043664 differing by -0.538652 s.
E0514 11:52:27.604573  7170 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525769467.173740805 but sensor time 1525769466.643222214 differing by -0.530519 s.
E0514 11:52:27.604615  7170 rosbag_validate_main.cc:350] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525769467.173905129 but sensor time 1525769466.653209733 differing by -0.520695 s.
W0514 11:52:27.604667  7170 rosbag_validate_main.cc:352] frame_id "base_link" on topic /dynamixel/odometry has serialization time 1525769467.174078623 but sensor time 1525769466.683208810 differing by -0.49087 s.
E0514 11:52:27.605494  7170 rosbag_validate_main.cc:315] Sensor with frame_id "base_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0514 11:52:27.605514  7170 rosbag_validate_main.cc:326] frame_id "base_link" is send on multiple topics. It was seen at least on /dynamixel/odometry and /pointcloud
E0514 11:52:27.605598  7170 rosbag_validate_main.cc:315] Sensor with frame_id "base_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0514 11:52:27.605610  7170 rosbag_validate_main.cc:326] frame_id "base_link" is send on multiple topics. It was seen at least on /dynamixel/odometry and /pointcloud
E0514 11:52:27.605664  7170 rosbag_validate_main.cc:326] frame_id "base_link" is send on multiple topics. It was seen at least on /dynamixel/odometry and /pointcloud
E0514 11:52:27.606106  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662669784513, current one at time 636613662669784513
E0514 11:52:27.606529  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662670304000, current one at time 636613662670043719
E0514 11:52:27.606590  7170 rosbag_validate_main.cc:315] Sensor with frame_id "laser" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0514 11:52:27.606993  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662670658064, current one at time 636613662669526541
E0514 11:52:27.607527  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662671197830, current one at time 636613662670658064
E0514 11:52:27.607659  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662670658064, current one at time 636613662670043719
E0514 11:52:27.608258  7170 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636613662670928506, current one at time 636613662670658064
E0514 11:52:27.611510  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769467.260382061 equals preceding data with 1081 points.
E0514 11:52:27.611678  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769467.260382061 equals preceding data with 1081 points.
E0514 11:52:27.611928  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769467.286345685 equals preceding data with 1081 points.
E0514 11:52:27.614277  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769467.426568592 equals preceding data with 1081 points.
E0514 11:52:27.627147  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769468.373731732 equals preceding data with 1080 points.
E0514 11:52:27.628096  7170 rosbag_validate_main.cc:190] Sensor with frame_id "laser" sends exactly the same range measurements multiple times. Range data at time 1525769468.452972664 equals preceding data with 1081 points.
W0514 11:52:29.270400  7170 rosbag_validate_main.cc:203] Sensor with frame_id "laser" range measurements have longest overlap of 0.131902 s

We updated the data-processing of the IMU data and now the localization works better thanks to a better estimation of the movement of the robot performed with the IMU. 

The issue that remains is how to get a 3D representation of the map. I noticed that each sub-map is an x-ray representation of the pointcloud generated during the time interval of two consecutive trajectories. There is any functionality that provides the 3D map?

gaschler, 2018-05-14:
> The issue that remains is how to get a 3D representation of the map. I noticed that each sub-map is an x-ray representation of the pointcloud generated during the time interval of two consecutive trajectories. There is any functionality that provides the 3D map?

If you want a single point cloud of the entire map, you can use the assets_writer (please see cartographer_ros documentation).

andrecmilano, 2018-05-14:
I tried assets_writer in two ways:
1) As described in the documentation (create pbstream file etc.) and the output is a set of images of the 3D point cloud.
[image]
2) using the launch file visualize_pbstream. Where, as illustrated in the picture, only the x-rays layers are visible and the point cloud is not present:
[image]

My ultimate goal is to subscribe to two topics that provide the 3D point cloud and the 6 DOF pose in real time of the drone. 

When I execute Cartographer these are the topics available:

- constraint_list (visualization_msgs/MarkerArray)
- initialpose (geometry_msgs/PoseWithCovarianceStamped)
- landmark_poses_list (visualization_msgs/MarkerArray)
- map (nav_msgs/OccupancyGrid)
- move_base_simple/goal (geometry_msgs/PoseStamped)
- pointcloud (sensor_msgs/PointCloud2)
- scan_matched_points2 (sensor_msgs/PointCloud2)
- submap_list (cartographer_ros_msgs/SubmapList)
- trajectory_node_list (visualization_msgs/MarkerArray)

I do not see how to get the 3D map, I do not find a point cloud topic during the execution.

Thank you for your invaluable help!

gaschler, 2018-05-14:
Ok, exporting a single point cloud is not documented. Documentation says "The individual PointsProcessors are all in the cartographer/io sub-directory and documented in their individual header files." :-))

Please take a look at https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/assets_writer_backpack_2d.lua
```
    {
      action = "write_ply",
      filename = "points.ply",
    },
```

and the `PcdWritingPointsProcessor` and `PlyWritingPointsProcessor` and other processors in `cartographer/io/`

andrecmilano, 2018-06-12:
Hello!

I am still working on the same problem mentioned above. 

I looked into PcdWritingPointsProcessor and PlyWritingPointsProcessor;
but they are just tools for saving the point cloud in an external file that can be used offline (I think). Indeed these tools do not solve my problem of 3D SLAM online. 

I had the possibility to capture additional datasets outdoor in large areas. When I run Cartographer with these datasets the outcome is worse than the simple reprojection of the points using the odometry of the drone (the outputs are illustrated in the figures), why does this happen?

point cloud projected in the odom frame without scanmatching or point filtering:
[image]

map output of cartographer:
[image]

I was wondering if there is any loop-closure for 3D point cloud, and also how does the scan matching work for 3D? Because it seems to not be available in this version of Cartographer. 

My main question then, is it possible to perform 3D global mapping using Google Cartographer?

After an analysis, I would not state that Cartographer can be applied for 3D global mapping. It seems to only generate 2D projection of the points, both in the submaps and global map. 
This is insufficient with what we are trying to achieve using a MAV. 

Thank you for your help and amazing work.

gaschler, 2018-06-12:
> I was wondering if there is any loop-closure for 3D point cloud

Yes, cartographer does 3D loop closure.
As an example, please run the 3D demo https://google-cartographer-ros.readthedocs.io/en/latest/index.html#running-the-demos with one of the larger 3D Deutsches Museum bag files.

> My main question then, is it possible to perform 3D global mapping using Google Cartographer?

Absolutely yes. Many users do 3D global mapping.

> It seems to only generate 2D projection of the points, both in the submaps and global map.

No, the projection is only done for visualization purposes. What you see in rviz is not the state of cartographer.

> I looked into PcdWritingPointsProcessor and PlyWritingPointsProcessor;
but they are just tools for saving the point cloud in an external file that can be used offline (I think).  Indeed these tools do not solve my problem of 3D SLAM online.

Right, assets writer is a batch tool.

> map output of cartographer:

Unfortunately, it is hard to tell from the image. This feels like a tuning issue.
Is the config and bag file you posted current? (I cannot find config changes in your fork https://github.com/andrecmilano/cartographer_ros) Should I take a look?

andrecmilano, 2018-06-12:
Thank you for your quick response.
I will come back to you with an updated version of bags and configuration files. :)

andrecmilano, 2018-06-12:
I uploaded my bags and config files [here](https://drive.google.com/drive/folders/1_rly5XcME6no_VdBMFVNHXNOfvkzIJq0?usp=sharing)

If you can have a quick look at the .lua file would be really nice. I have the feeling that I am missing something.

Thank you.

gaschler, 2018-06-12:
I had a quick look at the launch and lua files.

> use_odometry = true,

So this is quadrotor. 
Odometry, especially velocities, will be noisy, don't use it. (By odometry input, we usually think of wheel encoders.)
IMU should be enough to estimate rotation.

> TRAJECTORY_BUILDER_3D.submaps.high_resolution = 0.05

This is pretty small for outdoors.

> TRAJECTORY_BUILDER_3D.voxel_filter_size = 0.15

This should normally be the same as above.

> TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160

Is this the number of messages you receive in a physical lidar rotation?

Again, in a 3D setup it is super important that the IMU to lidar rotation (and also translation) is correct. 
Yours seems to be read from /tf in the bag file. Sure it is correct?

Please turn off global SLAM with `POSE_GRAPH.optimize_every_n_nodes = 9999999` and tune local SLAM first.

andrecmilano, 2018-06-19:
Thank you for your response and the suggestions provided with the .lua file.

> So this is quadrotor.
> Odometry, especially velocities, will be noisy, don't use it. (By odometry input, we usually think of wheel encoders.)

We still use odometry because the point cloud considered (topic: /scan_matched_points2) without odometry does not correctly match the 3D real world.

> Again, in a 3D setup, it is super important that the IMU to lidar rotation (and also translation) is correct.
> Yours seems to be read from /tf in the bag file. Sure it is correct?

Yes, we read from /tf and we assume it is correct because the point cloud seems to match the 3D world.

---------------------------------

Thanks to your hints now the loop-closure seems correctly executed and I can see in the 2d map that the walls of the building are aligned once the loop closure optimization is concluded. Thank you.

Now I was wondering where can I find the 3D point cloud containing all the points. 
I was thinking to call the function GetAllSubmapData() in PhoseGraph3D and iterate through all the submaps and collect the points. Is it a good approach? Is the point cloud after the optimization stored in the submap vector?

I also tried to have a look into the PlyWritingPointsProcessor class where the method Proces receive in input std::unique_ptr<PointsBatch> which contains all the 3D Points. But it is difficult to follow and understand where to Get the  PointsBatch object.

So far I only looked into this two possible "point cloud providers". My ultimate goal is to check the 3D point cloud live during the optimization process.

gaschler, 2018-06-20:
> Now I was wondering where can I find the 3D point cloud containing all the points.
> I was thinking to call the function GetAllSubmapData() in PhoseGraph3D and iterate through all the submaps and collect the points.

Absolutely feasible, more involved. This is what cartographer_ros does to publish (projected) submaps to rviz. It can give you results at run-time.

> I also tried to have a look into the PlyWritingPointsProcessor class where the method Proces receive in input std::unique_ptr which contains all the 3D Points. But it is difficult to follow and understand where to Get the PointsBatch object.

If offline processing is sufficient, try this first. This is the supported method to generate a point cloud from a pose graph.
You just need to write a lua config file and run the assets_writer on your pbstream.
There is some documentation in https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html

Hope that helps.

andrecmilano, 2018-07-03:
**Update**:
I took a look at the repository of @nitramek mentioned in this issue: #[720](https://github.com/googlecartographer/cartographer_ros/issues/720#issuecomment-399936062) (in certographer_ros).

I have new questions regarding the trajectories organization. I will open a new issue.

Thank you for your support!

closed 2018-07-03 by andrecmilano

DeepakKarishetti, 2019-02-21:
> I uploaded my bags and config files [here](https://drive.google.com/drive/folders/1_rly5XcME6no_VdBMFVNHXNOfvkzIJq0?usp=sharing)
> 
> If you can have a quick look at the .lua file would be really nice. I have the feeling that I am missing something.
> 
> Thank you.

Thank you for your information in there, I am a fresh man to work on the slam recently, could you tell me, whether the cartographer can build the map in real-time and update directly, not just record data and build the map...

Thank you for your patience