cartographer-project/cartographer #1137: Does optimize_every_n_nodes do the same as sampling_ratio?

AlexReimann, 2018-05-02:
From my understanding the sampling ratio basically triggers the optimization every `1 / sampling_ratio` time it gets called.
This seems analog to `optimize_every_n_nodes`?

So in the end one ends up optimizing every `optimize_every_n_nodes * (1 / sampling_ratio)` nodes?

Is there any other difference to the parameters?

gaschler, 2018-05-02:
> From my understanding the sampling ratio basically triggers the optimization every 1 / sampling_ratio time it gets called.

No, it just drops input data. So any other setting than 1.0 is hardly useful.

> Is there any other difference to the parameters?

Yes.

*Use `optimize_every_n_nodes` to adjust the optimization frequency.*
Of course, it also depends on the node frequency, which so `num_accumulated_range_data` makes it less frequent.

AlexReimann, 2018-05-02:
Okay. The default value for the [`sampling_ratio` currently is `0.3`](https://github.com/googlecartographer/cartographer/blob/e080df5ea36c3737b84635bed45b58051dd83efc/configuration_files/pose_graph.lua#L18).

What's the merit of having a lower optimization frequency (with `optimize_every_n_nodes` > 1)?
Doesn't it just change the computation effort from continuous "low" usage to bursts (which are usually worse)?

ojura, 2018-05-02:
You two had a misunderstanding. @gaschler referred to sampling ratios in cartographer_ros - these just drop the input data. @AlexReimann that sampling ratio determines which ratio of scans/nodes the constraint builder tries to use for loop closing i.e. performs scan matching against finished submaps, which might result in a loop closure constraint.

gaschler, 2018-05-02:
Sorry for the confusion. @ojura is right.

> What's the merit of having a lower optimization frequency (with optimize_every_n_nodes > 1)?

Optimization takes time and blocks all background threads that usually search for constraints.

> Doesn't it just change the computation effort from continuous "low" usage to bursts (which are usually worse)?

Optimization works on all nodes (that are not frozen), not only new ones.
I didn't measure it, but when you have many new constraints, optimization doesn't really require more iterations, so it is not a much longer burst.

closed 2018-05-07 by AlexReimann