cartographer-project/cartographer #1138: Reuse absl::Mutex

gaschler, 2018-05-02:
I evaluated to replace our Mutex implementation to use `absl::Mutex` instead of `std::mutex` in 
https://github.com/googlecartographer/cartographer/compare/master...gaschler:abseil-mutex?expand=1 

Pro:
- Slightly better performance. I measured 1% shorter wall time on 2D offline_node runs (without rviz), which is probably the case with most impact.
- Don't implement `Await` ourselves.

Con:
- Abseil is not packaged for most platforms.
- CMake is not supported (yet). I tried to write a hacky CMake script with `externalproject_add`, but this becomes quite complicated and is definitely not supported by some platforms (Indigo uses old CMake) but rather requires CMake 3.2.
- Additional dependency.
- Saw some additional TSAN issues (so I'm not sure if the patch is correct), would require more investigation.

wohe, 2018-05-04:
Another important advantage is that the `absl::Mutex` is approximately fair in the long term, the `std::mutex` is not.

cschuet, 2018-05-04:
@wohe Is the main appeal of fairness expected performance gains? Or what makes fairness desirable?

SirVer, 2018-05-04:
@cschuet Without fairness, there is the possibility that the background thread starves local SLAM or wise versa. We actually see this issue internal at Lyft. It is a desired property for performance and can have a substantial impact in our case.

cschuet, 2018-05-05:
@sirver ok thanks for sharing your experience. So the benefits of mutex fairness in performance seem to be quite Pipeline and use case specific. Do you have a hunch what aspects of your setup make it sensitive to fair mutexes? Because as André wrote we haven’t been able to identify performance gains in the cases we looked at.

wohe, 2018-05-08:
@cschuet To be more precise, the local SLAM is seeing lock contention and you can also observe this in the public data sets. It is not use case specific: if you run SLAM long enough, the pauses that local SLAM observes grow. For example, if you run the public 3D data sets you can see it in RViz, visibly pausing for short periods of time. This is especially bad for the pure localization use case where it introduces additional latency.

This can be resolved by doing 3 things, and all of them are needed to see any benefit:
1. Make sure the locking done in the background thread cannot starve the foreground thread. Only this direction is important. It seems using a mutex which is more fair is the easiest option, but there are alternatives to achieve this.
2. Move global SLAM work currently done in the foreground into the background.
3. Use finer grained locking in the background thread.

wohe, 2018-05-08:
Regarding the inability to see performance gains: You need all three before you see performance gains. If you do, one experiment I did showed approximately 20% gain in wall clock time in a public 3D SLAM data set.

cschuet, 2018-05-08:
@wohe Thanks for the detailed explanation.

> 2. Move global SLAM work currently done in the foreground into the background.
Do you mean by this the behavior where the work items are executed on the FG thread if the work queue is empty?

wohe, 2018-05-08:
@cschuet Correct. Unfortunately in my experiment just doing 2 actually reduces performance substantially. So the best ordering for implementing this change would probably be 1, then 3, and finally 2.

danielsievers, 2018-07-10:
Also see https://github.com/googlecartographer/cartographer/issues/1250 regarding the issue that wohe is referring to in his comment above.

gaschler, 2018-07-26:
Good news everyone, #1301 added Abseil as a dependency that is built (with a hacky cmake script), so `absl::Mutex` can in principle be used.

closed 2018-08-06 by pifon2a