cartographer-project/cartographer #1145: Running Cartographer on Jetson TX2 board

ekow-mensah, 2018-05-08:
Hi I am trying to use the Jetson TX2 board to implement online SLAM. According to the docs, it says that that 16GB of RAM is the minimum requirement, however, the Jetson TX2 board only has 8GB of RAM. Will it be possible to still use the Jetson TX2 board to efficiently run cartographer??

gaschler, 2018-05-08:
> 16GB of RAM is the minimum requirement

It is not that simple.
Memory consumption depends almost only on the map resolution and size.
2D SLAM uses much less than 3D.
I'd say just tune the resolution and filter parameters for speed as in https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency and memory also goes down.

> Jetson TX2 board

You didn't specify its CPU architecture and performance.
Cartographer currently requires a 64bit x86 architecture and works best with many cores and vector math.
Without multiple cores, global SLAM (loop closure) will be slow.

ekow-mensah, 2018-05-08:
Its the Nvidia Jetson TX2 board it has: 
 - dual-core NVIDIA Denver2 + quad-core ARM Cortex-A57
 - 256-core Pascal GPU
 - 8GB LPDDR4, 128-bit interface
 - 32GB eMMC
 - 4kp60 H.264/H.265 encoder & decoder
 - Dual ISPs (Image Signal Processors)
 - 1.4 gigapixel/sec MIPI CSI camera ingest

in terms of its architecture It has 64 bit CPUs as well

gaschler, 2018-05-08:
> dual-core NVIDIA Denver2 + quad-core ARM Cortex-A57

Arm 64bit is theoretically fine but may be hard to compile for, and we currently don't have any documentation for that.

ekow-mensah, 2018-05-08:
oh okay thanks Gaschler I will try it and let you know how it goes.

closed 2018-05-09 by ekow-mensah

ZhiyuLiu1993, 2018-05-24:
@ekow-mensah Hi, did you succeed on 64 bit ARM?My CPU performance is worse，only 512M RAM,。I want to try it with another platform instead of ROS.

ekow-mensah, 2018-05-24:
No I haven't tried it yet unfortunately as I have not had the time. Also, 512M RAM is too small. According to the minimum requirements you need 16GB of RAM. But yes feel free to try another platform instead of ROS to see if your results get any better.

surfertas, 2018-07-14:
@erwincoumans Did you encounter any performance degradation as time progressed. Seems like this user was having such issues with a TX1, albeit the specs are quiet different between TX2 and TX1. https://www.youtube.com/watch?v=F94sJL9vgbU

Cheers.

anakin0620, 2019-03-25:
My experiment on my TX2 is fine with the 2D backpack demo.

guorun198959, 2019-10-14:
@anakin0620 hello,I tried install the cartographer on my TX2,but failed.I think the reason is  I cannot install the dependency clang,Did u install clang on ur tx2? and I tried install it by apt.It failed too. can u give me some advice? thank u so much!!

anakin0620, 2019-12-03:
@guorun198959  Hi, what Ubuntu version you are using? If it's 18.04, you should install clang6. My TX2  has 16.04 on it, I didn't have any problem with the installation of Cartographer, just follow the tutorial you can install it. If you have any questions just ask me.