cartographer-project/cartographer #1150: Difference between 2D and 3D? (question)

ekow-mensah, 2018-05-09:
Hi, 
 Please forgive me I am a real amateur and I just really want to understand whether I could use Cartographer 3D for my application. Please what are the differences between 2D and 3D? I tried running the bag examples and I could not see any differences in terms of the map that was constructed. I'm planning to use a drone or a mobile robot to avoid obstacles using the Map so I will need depth information. Do you think this will be possible with Cartographer 3D?

kdaun, 2018-05-11:
The 2D pipeline estimates a trajectory of 3DoF (x,y, yaw) poseses by matching 2D scans (or 2D scan slices) in 2D submaps. In contrast, the  3D pipeline estimates a trajectory of 6DoF (x,y,z, roll, pitch, yaw)  poses by matching 3D scans in 3D submaps.
Therefore, for 3D maps created with the 2D pipeline the scans are projected in 3D space based on 3DoF poses.
Using Cartographer for SLAM on a mobile robots is common application. For discussions about using Cartographer with a drone please take a look at https://github.com/googlecartographer/cartographer/issues/1134 and
https://github.com/googlecartographer/cartographer/issues/459.

ekow-mensah, 2018-05-11:
Hi kdaun thank you for your answer. So then is it possible to render a 3D occupancy map?

kdaun, 2018-05-11:
Currently, we do not publish 3D maps. See  https://github.com/googlecartographer/cartographer_ros/issues/720 and https://github.com/googlecartographer/cartographer_ros/issues/330.

ekow-mensah, 2018-05-11:
@kdaun Thank you very much I will close this issue now.

closed 2018-05-11 by ekow-mensah