cartographer-project/cartographer #1153: How to choose the vaule of weight when using the GPS data to fix frame ?

daijicheng, 2018-05-11:
It seems the accuracy and covariance of GPS data are not used when optimizing. So what is the relationship between GPS data accuracy and frequency and weight? And I think if change the weight according to GPS data, we can get a better result. Is my idea feasible?

rqren2008, 2018-07-04:
@daijicheng
Do  you have test the cartographer in 3D with GPS?

Fanny-one, 2019-07-03:
Hi @daijicheng ,

> No, I don't test the cartographer in 3D with GPS yet, but I plan to do this.

So now hava you tested cartographer in 3D  with GPS?
I'm planning to do this. I have got my gps topic by  using nmea_navsat_driver and assigned  "use_nav_sat = 1", What else should do?
Hoping for your reply!
Thank you very much!

daijicheng, 2019-07-03:
@Fanny-one 
What you need to do is publishing the nav_sat message with the topic name "fix". 
I have not finished test cartographer with GPS. If you have any progress, I hope you can show me the result.

Fanny-one, 2019-07-03:
> What you need to do is publishing the nav_sat message with the topic name "fix".
> I have not finished test cartographer with GPS. If you have any progress, I hope you can show me the result.
thank your reply!
ok!
My  colleagues are testing cartographer without GPS . Now I want to test it with GPS but don't know how to do that. Maybe we can communicate with each other through qq which is 827938633 or wechat.