cartographer-project/cartographer #1158: Relocalization in 3D

arin07, 2018-05-16:
Hi,
  We are using cartographer for SLAM in an autonomous vehicle in outdoor environment. The sensors used are:
            Lidar: Quanergy M8 - 8 beam lidar
            IMU
We are not using the ROS framework. We are getting good local SLAM output but we have not been able to relocalize in the previous pbstream file generated during the previous run. We have set the configuration parameters according to the cartographer ROS documentation. But the output we get looks like the local SLAM output and not with respect to the previous run i.e non transformed output starting at 0,0,0. Some pointers on using the cartographer relocalization would be highly helpful.

gaschler, 2018-05-16:
What logs/stdout are you seeing?
How many submaps are created, how many constraints are found?
Have you transformed your lidar data correctly into the IMU's reference frame?
What is your plan to visually inspect how the algorithm is working? (You could adapt cartographer_ros_node to only read and publish to rviz.)

arin07, 2018-05-17:
So what I do is first I do an offline SLAM using the pcds from some previous run. I convert it to the requisite format and run cartographer on the whole set and generate the .pbstream file. Thereafter I load the .pbstream file  using the map_builder->Load_state. Thereafter I run the system on  a subset of the original data. 
The lidar and the IMU are mounted on the same axis so we didn't think we needed to transform them. The Local SLAM output is good and we are already using them in our control algorithms.
Visual Inspection of the output , we save the hits and misses and display them later, not in operation.
As ROS has a lot of overhead, hence we are not using ROS in any of our systems.
What we needed was to be able to relocalize, which I understand as the current output being with respect to the previous map frame(the frame of the loaded .pbstream file). The output we get seems to be the Local SLAM output itself.
 I will attach the logs for when we run a subset of data after loading the .pbstream file.  
[carto_relocal.log](https://github.com/googlecartographer/cartographer/files/2011699/carto_relocal.log)

gaschler, 2018-05-17:
> So what I do is first I do an offline SLAM using the pcds from some previous run. I convert it to the requisite format and run cartographer on the whole set and generate the .pbstream file. 

This sounds error prone. Why should you convert to pcd? This sound like you lose the whole pose graph structure, which is necessary to localize or resume SLAMing.

> I0517 10:43:57.791105 23777 constraint_builder_3d.cc:311] 0 computations resulted in 0 additional constraints.

No constraints are found.

> we are not using ROS in any of our systems.

Without any type of visual inspection Cartographer is extremely hard to tune.

> I0517 10:43:58.847160 23777 pose_graph_3d.cc:399] Remaining work items in queue: 2807

Your log ends early, already after the second submap.

arin07, 2018-05-17:
For offline processing we convert the live lidar data to pcds. There are two systems, one live which gets the data directly from the lidar and other where we store the lidar data separately in form of pcds for use by other algorithms. I use the stored pcds to test the cartographer functionalities offline.
There were only 200 pcds hence it didn't go beyond the second submap.

gaschler, 2018-05-17:
> There were only 200 pcds hence it didn't go beyond the second submap.

Is one pcd one laser scan revolution? That would be fine.
With a few seconds worth of laser scan inputs not finding new constraints would be inconclusive.

arin07, 2018-05-17:
Yes, one pcd is one rotation. Ideally how much time would it take to relocalize? And  from where do we extract that output with respect to the loaded .pbstream file.Would the local SLAM output (returned in the local_slam_callback) be transformed with respect to the frame of the pbstream file? Or do we extract it from somewhere else?

gaschler, 2018-05-17:
> Ideally how much time would it take to relocalize? 

Until the first global constraint to the old trajectory is found.
The only way to be sure is reading `PoseGraphInterface::constraints` and finding a constraint across your current trajectory and that of the given map.

> Would the local SLAM output (returned in the local_slam_callback) be transformed with respect to the frame of the pbstream file? 

No, local slam callback provides a `tracking_to_local` transform, so it is on a continuous trajectory.
Look at cartographer_ros `Node::PublishTrajectoryStates`, which get data from `MapBuilderBridge::GetTrajectoryStates` and computes `local_to_map * tracking_to_local`.
The current global pose (in the pbstream coordinate system) is `map_builder_->pose_graph()->GetLocalToGlobalTransform(trajectory_id) * tracking_to_local`.

gaschler, 2018-07-27:
Closing as answered.

closed 2018-07-27 by gaschler