cartographer-project/cartographer #1160: how to use fix_z_in_3d option

slamer-run, 2018-05-17:
I'd like to ask for utilizing "fix_z_in_3d" option at below file. cartographer/cartographer/mapping/internal/optimization/optimization_problem_3d.cc

As far as, I read cartographer 3d, switch residual as following.

- fix_z_in_3d = false
optimizing residual of acceleration and rotation from imu

-  fix_z_in_3d = true
optimizing residual of spa from local slam and wheel odometry 

My question is as following.
1. Why don't you use local slam result every configuration?
2. Why is the fix_z_in_3d false as default? (The fix_z_in_3d is not set at "pose_graph.lua", though...)  In order to estimate pitch and roll angle, I know acceleration sensor is very useful. However, without local slam and odometry, it is not enough precise to estimate translation, even there is loop closure constraint.

FlorentRevest, 2018-07-17:
This question has already been raised here: https://github.com/googlecartographer/cartographer_ros/issues/290 with comments from the developers.

gaschler, 2018-07-27:
Closing duplicate.

closed 2018-07-27 by gaschler