cartographer-project/cartographer #1162: Allow deasactivation of final optimization step for easy tuning of local slam

doisyg, 2018-05-22:
Hello,
In order to see the result of the tuning of the local slam, I deactivated the global slam as recommended in the documentation with the following parameter.
`POSE_GRAPH.optimize_every_n_nodes = 0`
However, at the end of the maping there is a final optimization which is realigning all the submaps. It is working superbly hence it doesn't allow to see the result of the local slam tuning.
I would like to desactivate this optimization. I did not fine a parameter controlling it. Is there any?
There is the parameter `POSE_GRAPH.max_num_final_iterations ` (200 by default), but it cannot be set to 0.
I am using the offline node to precess a bag file.

doisyg, 2018-06-18:
Coming back on this.
Or alternatively, any efficient method to test the results of local slam tuning?

UserUsingGit, 2019-07-30:
Any news so far on that issue? I am having the same problem (but I am using the cartographer_ros package). I looked up the source code (offline_node.cc) and saw the following lines in the end:

// Here is the loop where bag msgs are published. And here is the crucial part I guess:
...
    if (is_last_message_in_bag) {
      node.FinishTrajectory(trajectory_id);
    }
  }
  // Ensure the clock is republished after the bag has been finished, during the
  // final optimization, serialization, and optional indefinite spinning at the
  // end.
  clock_republish_timer.start();
  node.RunFinalOptimization();
...
// After this some logs are printed and so on...

There is no option to run (or not run) the final optimization. So there seems to be no choice in the offline node at this point. As I want to compare several datasets / configurations based on accuracy, it would be great to have other options for automizing several runs in a row and using only local SLAM by choice.
Does anyone know if there is another way of making sure global SLAM is disabled in the offline node (e.g. using the lua file or launch file)?

UserUsingGit, 2019-07-31:
I just commented out node.RunFinalOptimization() for first and it works as quick and dirty solution. If any better (selectable) solution is found I would prefer that of course.

lucamozza, 2022-04-05:
Commenting this old post because I think this could be very helpful for sistematic Local SLAM evaluation and tuning
I'll probably modify 
https://github.com/cartographer-project/cartographer/blob/105c034577220268cd28a304a185adbec46b729f/cartographer/mapping/internal/3d/pose_graph_3d.cc#L811-L829
and
https://github.com/cartographer-project/cartographer/blob/105c034577220268cd28a304a185adbec46b729f/cartographer/mapping/internal/2d/pose_graph_2d.cc#L841-L859

to something like this
``` cpp
void PoseGraphnD::RunFinalOptimization() {
  if (options_.max_num_final_iterations() > 0) {
    AddWorkItem([this]() LOCKS_EXCLUDED(mutex_) {
      // etc
    });
    AddWorkItem([this]() LOCKS_EXCLUDED(mutex_) {
      // etc
    });
  }
  WaitForAllComputations();
}
```
If this does have the intended effect, I could make a pull request

Cheers,
Luca

marcoluis97, 2023-04-09:
Greetings @lucamozza! Must one rebuild the workspace after changing this file?

lucamozza, 2023-04-09:
> Greetings @lucamozza! Must one rebuild the workspace after changing this file?

Absolutely, you need to build again cartographer. 
cheers, Luca