cartographer-project/cartographer #1173: hello,Relatively poor registration accuracy at rest

xiangkui123, 2018-05-30:
hello I found several problems,Lead to poor trajectory accuracy.
1.Trajectory registration accuracy is poor at rest,Accuracy fluctuations of about 3-5cm in the same place.
2.Silver metal has a great impact on velodyne ranging
3.The overall track positioning accuracy is poor, about 10cm

xiangkui123, 2018-05-30:
1.Velodyne noise points and outliers in this province,The main boundary point is relatively poor.
[image]

2.Silver Metal Effect on Velodyne Ranging
[image]

The actual construction of elevator doors affects the construction.

3.Trajectory registration accuracy is poor at rest,Accuracy fluctuations of about 3-5cm in the same place. 
[image]

xiangkui123, 2018-05-30:
The front-end registration itself is greatly influenced by Velodyne noise and outliers. Is there any improvement? The measured positioning accuracy in 3D is about 10cm, and the z-axis value is easy to float or fluctuate. We started to do a good initialization.

xiangkui123, 2018-05-30:
[image]

xiangkui123, 2018-05-30:
Trajectory accuracy is measured by repeated positioning, registration accuracy observes the trajectory fluctuations through stationary equipment.Data analysis through CloudCompare.

xiangkui123, 2018-05-30:
There is another problem I found that there is a serious stratification of the stairs up and down!
1.From the top view, the floors fluctuate so much
[image]

[image]

[image]

[image]

[image]

2.Up and down multi-level stairs,The side is a bit fluttering
[image]
[image]

How to solve the problem of floor flutter?

xiangkui123, 2018-05-30:
Sorry lost a silver metal effect on registration at rest,On the right is the recessed place where there are two elevator doors.
[image]

cschuet, 2018-05-30:
Thanks for your comments and the illustrations.

It is hard to make concrete suggestions without knowing more about your specific setup. If you feel comfortable with this, please fork the cartographer_ros repository and add your setup specific configuration files there and upload a representative sample rosbag.

It would be interesting to understand parameters such as grid resolution and how you do the accumulation of your velodyne data (I assume you accumulate a single revolution?). When you feel the input data from your laser is too noisy (because of silver metal) you can try to accumulate more than one revolution to achieve more averaging.

gaschler, 2018-05-30:
Thanks for sharing your results, this is a very cool setup.

Without seeing any config or data (consider at least sharing cartographer_rosbag_validate output), I can only give some ideas

* The trajectories are not ideal for mapping. Walk more circular routes, not just back and forth. Walk up and down two different stairways. Global SLAM will then be better.
* Take the imu/laser transform that cartographer log outputs in the end to calibrate your urdf. This would fix a systematic tilt angle (moving downwards/upwards while walking through long corridor)

> The front-end registration itself is greatly influenced by Velodyne noise and outliers. Is there any improvement?

Tune the filters, especially the maximum ray length for insertions.

> 2.Silver metal has a great impact on velodyne ranging

To help here, I guess we would need to see example data.

xiangkui123, 2018-06-07:
This is a set of data I collected，I need to count the number of trees and the perimeter, but I find that the trunk is very thick and I'm already thin.

[image]

xiangkui123, 2018-06-07:
[image]
[image]

xiangkui123, 2018-06-07:
[image]

closed 2018-06-12 by xiangkui123