cartographer-project/cartographer #1194: Why the kalman is cancelled in new version of carto?

mantangL, 2018-06-14:
I want to know why the kalman pose tracker in old version of carto is cancelled? it's slow?it's inaccurate? or another ?

gaschler, 2018-06-14:
(Duplicate of https://github.com/googlecartographer/cartographer_ros/issues/624#issuecomment-351666795)

"Cartographer did use an unscented Kalman filter in the past, it was removed in
googlecartographer/cartographer#378 and replaced by the current PoseExtrapolator approach. Both solutions are computationally efficient, but the current gives better results in practice. So, we are very hesitant to add a Kalman filter now."

>  it's slow?

No, neither is the new `PoseExtrapolator`.

> it's inaccurate?

Yes, but the simpler `PoseExtrapolator` was better with less parameters and tuning.

mantangL, 2018-06-14:
Thanks for your help. The new PoseExtrapolator is simpler, but is it rough to compute the pose using linear_velocity_from_poses_ or linear_velocity_from_odometry_ * time_delta ? The old version use kalman filter to predict it.

mantangL, 2018-06-14:
And in unscented_kalman_filter.h in old version, the function void Observe(
      std::function<Eigen::Matrix<FloatType, K, 1>(const StateType&)> h,
      const GaussianDistribution<FloatType, K>& delta)
the last line:
    belief_ = GaussianDistribution<FloatType, N>(
        add_delta_(mu, kalman_gain * -z_hat), new_sigma);
is it right?
Should we change it to these:
    belief_ = GaussianDistribution<FloatType, N>(
        add_delta_(mu, kalman_gain * (z - z_hat)), new_sigma);
But I can't find the z,the true observed value of state.

gaschler, 2018-07-27:
Closing as answered.

closed 2018-07-27 by gaschler