cartographer-project/cartographer #1197: Compiling cartographer & cartographer_ros with a customer install directory for protobuf3

jochemverboom, 2018-06-15:
I am trying to build/install cartographer and cartographer_ros with a custom location for protobuf3 while not adding the protobuf3 to the install folder. I do not want to add protobuf3 to my current catkin workspace directory as I have different packages depended on the system installation of protobuf2 (If i add protobuf3 to the install directory, but I need to use a custom location).

I have forked cartographer (see: [](https://github.com/jochemverboom/cartographer))

gaschler, 2018-06-15:
Protobuf has a compiler and a library, all have to match in both cartographer and cartographer_ros builds.

I think you can select which protobuf to use in the cmake flags
```
 Protobuf_INCLUDE_DIR             /usr/local/include                                                              
 Protobuf_LIBRARY_DEBUG           /usr/local/lib/libprotobuf.so                                                   
 Protobuf_LIBRARY_RELEASE         /usr/local/lib/libprotobuf.so                                                   
 Protobuf_LITE_LIBRARY_DEBUG      /usr/local/lib/libprotobuf-lite.so                                              
 Protobuf_LITE_LIBRARY_RELEASE    /usr/local/lib/libprotobuf-lite.so                                              
 Protobuf_PROTOC_EXECUTABLE       /usr/local/bin/protoc                                                           
 Protobuf_PROTOC_LIBRARY_DEBUG    /usr/local/lib/libprotoc.so                                                     
 Protobuf_PROTOC_LIBRARY_RELEAS   /usr/local/lib/libprotoc.so   
```

Of course, you would install protobuf3 in a non-system folder, not in /usr/local.

Perhaps your solution is almost working, just look at the (hidden) cmake flags for both cartographer and cartographer_ros.

Yet another alternative solution is the install script here: https://github.com/googlecartographer/cartographer_ros/issues/625#issuecomment-358318701

gaschler, 2018-07-27:
Closing as answered

closed 2018-07-27 by gaschler