cartographer-project/cartographer #1199: How to calibrate lidar and imu?

jojo666666, 2018-06-18:
Hi everyone, I have a vlp16 and an imu, but I don't know how to ###configure the .urdf file, which is the calibration problem for velodyne and imu. I want to know if you are using calibration tools or your own code. If you can give me some advice I would greatly appreciate it.

gaschler, 2018-06-18:
1. If IMU and lidar are axis-aligned (i.e., there are only 0/90/180 deg angles), I would draw the coordinate frame diagram and obtain the rotation from that. If not, you probably need to model the setup with CAD software.
2. You can run local SLAM only, open rviz, and look for a systematic drift. Correct until the drift is small.
3. When the IMU/lidar rotation is correct +/- a few degrees, global optimization will do the rest for you, see
https://github.com/googlecartographer/cartographer_ros/issues/902#issuecomment-397971589

jojo666666, 2018-06-20:
Based on your previous tips, I have successfully run the 3d cartographer.
After finishing POSE_GRAPH.optimization_problem.log_solver_summary = true, where is the result？

FlorentRevest, 2018-07-17:
If you enable this option, the corrected pose will simply be logged on the standard output of cartographer.
You'll see quaternion corrections being printed out as they are found by Ceres everytime a global optimization is ran.
https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/optimization/optimization_problem_3d.cc#L575

gaschler, 2018-07-27:
Closing as answered

closed 2018-07-27 by gaschler

xielingsen, 2018-11-28:
@FlorentRevest  There is no calibration output in 2d case with log_solver_summary enabled. I wonder how to obtain the calibration result for 2d ?