cartographer-project/cartographer #1212: Non-sorted data assertion in ordered queue

damienrg, 2018-06-25:
When I finish a trajectory and start a new one I could have:
`F0625 10:08:27.729404 30040 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime()`

In my case this is due to the following scenario:
 - start of cartographer with a trajectory
 - imu data at 1millisecond
 - odometry data at 1millisecond (common start time in ordered_multi_queue is then 1 millisecond)
 - odometry data at 3 milliseconds
 - finish trajectory
 - new trajectory
 - imu data a 2 milliseconds

This can be reproduced by the following code:
```c++
TEST(OrderedMultiQueueTestOther, MarkQueueAsFinishedMultiple) {
  std::vector<std::unique_ptr<Data>> values;
  OrderedMultiQueue queue;

  const int trajectory1 = 1;
  const QueueKey imuTrajectory1{trajectory1, "imu"};
  const QueueKey odometryTrajectory1{trajectory1, "odometry"};

  const int trajectory2 = 2;
  const QueueKey imuTrajectory2{trajectory2, "imu"};

  const auto AddQueues = [&queue,&values](std::initializer_list<QueueKey> queueKeys) {
    for (const QueueKey& queueKey : queueKeys) {
      queue.AddQueue(queueKey, [&values](std::unique_ptr<Data> data) {
        if (!values.empty()) {
          EXPECT_GE(data->GetTime(), values.back()->GetTime());
        }
        values.push_back(std::move(data));
      });
    }
  };

  AddQueues({imuTrajectory1, odometryTrajectory1});

  queue.Add(imuTrajectory1, MakeImu(1));
  EXPECT_TRUE(values.empty());
  queue.Add(odometryTrajectory1, MakeOdometry(1));
  EXPECT_EQ(1, values.size());
  queue.MarkQueueAsFinished(imuTrajectory1);
  EXPECT_EQ(2, values.size());
  queue.Add(odometryTrajectory1, MakeOdometry(3));
  EXPECT_EQ(3, values.size());
  queue.MarkQueueAsFinished(odometryTrajectory1);
  EXPECT_EQ(3, values.size());

  AddQueues({imuTrajectory2});
  queue.Add(imuTrajectory2, MakeImu(2)); // Check failed: last_dispatched_time_ <= next_data->GetTime() (3 vs. 2) Non-sorted data added to queue: '(2, imu)'
}
```
I want to know if you consider it as a bug as if I do not finished the trajectory there is no assertion or if I should modify my code to handle this case.

gaschler, 2018-06-25:
I'm not entirely sure.
I think you are only supposed to call MarkQueueAsFinished for all sensors of a trajectory after all sensors have finished.
Does it still fail when you move the MarkQueueAsFinished to the end of the block where you "finish trajectory1"?

damienrg, 2018-06-25:
If you mean:
```c++
// same code as before

queue.Add(imuTrajectory1, MakeImu(1));
EXPECT_TRUE(values.empty());
queue.Add(odometryTrajectory1, MakeOdometry(1));
EXPECT_EQ(1, values.size());
queue.Add(odometryTrajectory1, MakeOdometry(3));
EXPECT_EQ(1, values.size());
queue.MarkQueueAsFinished(imuTrajectory1);
queue.MarkQueueAsFinished(odometryTrajectory1);
EXPECT_EQ(3, values.size());

// same code as before
```
then I got the same error.

gaschler, 2018-06-25:
I think the minimal case will be like this:

1. Create trajectory 1
2. Send data with timestamp 100 to trajectory 1
3. Create trajectory 2
4. Send data with timestamp 50 to trajectory 2

`OrderedMultiQueue` or `Collator` or `CollatedTrajectoryBuilder` really cannot handle this kind of case.
They guarantee to forward data strictly time-sequential, but at step (2.) they didn't know and could not wait for trajectory 2.

What do you think the expected behavior should be?

Have you seen this issue in the wild?
`TrajectoryCollator` won't have this issue, so it could be a work around to enable it, but it won't sort across trajectories.

damienrg, 2018-06-26:
> I think the minimal case will be like this:
> 
>     Create trajectory 1
>     Send data with timestamp 100 to trajectory 1
>     Create trajectory 2
>     Send data with timestamp 50 to trajectory 2

In this case you will got a failure when you insert data with timestamp 50 in case data of trajectory 1 was dispatched. In my case it is:
 - create trajectory 1
 - send imu data with timestamp 100 to trajectory 1
 - send odometry data with timestamp 100 to trajectory 1
 - send odometry data with timestamp 200 to trajectory 1
 - finish trajectory 1
 - create trajectory 2
 - send imu data with timestamp 199 to trajectory 2 (due to a message received a bit later)

If I create trajectory 2 before  finishing trajectory 1, it does not fail as `OrderedMultiQueue::Dispatch()` will call `OrderedMultiQueue::CannotMakeProgres()`.

Just note that in my case I have:
  - imu data at: 100 and 199
  - odometry data at: 100 and 200
So imu dates have dates which always increase, same for odometry.

> OrderedMultiQueue or Collator or CollatedTrajectoryBuilder really cannot handle this kind of case.
> They guarantee to forward data strictly time-sequential, but at step (2.) they didn't know and could not wait for trajectory 2.

Yes but you could discard data for e.g.

> What do you think the expected behavior should be?

It depends on OrderedMultiQueue semantics. If you think that this it is the expected behaviour maybe just add documentation about this case.

cschuet, 2018-06-27:
> If I create trajectory 2 before finishing trajectory 1, it does not fail as OrderedMultiQueue::Dispatch() will call OrderedMultiQueue::CannotMakeProgres().

Are you saying

* create trajectory 1
* send imu data with timestamp 100 to trajectory 1
* send odometry data with timestamp 100 to trajectory 1
* send odometry data with timestamp 200 to trajectory 1
* create trajectory 2
* finish trajectory 1
* send imu data with timestamp 199 to trajectory 2 (due to a message received a bit later)

does not fail or 

* create trajectory 1
* send imu data with timestamp 100 to trajectory 1
* send odometry data with timestamp 100 to trajectory 1
* send odometry data with timestamp 200 to trajectory 1
* create trajectory 2
* send imu data with timestamp 199 to trajectory 2 (due to a message received a bit later)
* finish trajectory 1

does not fail?

damienrg, 2018-06-28:
I meant the first case but both do not fail as trajectory2 is created before finishing trajectory 1.

damienrg, 2018-07-06:
Well the creation of trajectory 2 before trajectory 1 does not work as it breaks invariant at a higher level:
```c++
TEST_F(MapBuilderTest, FinishTrajectoryWithOtherTrajectory) {
  const auto addFakeRangeTo = [this](int trajectory_id, double elapsed_time) {
    TrajectoryBuilderInterface* trajectory_builder =
        map_builder_->GetTrajectoryBuilder(trajectory_id);
    cartographer::sensor::TimedPointCloudData data{
        cartographer::common::FromUniversal(123) +
            cartographer::common::FromSeconds(elapsed_time),
        Eigen::Vector3f::Zero(),
        {Eigen::Vector4f::Zero()}};
    trajectory_builder->AddSensorData(kRangeSensorId.id, data);
  };

  const auto addFakeImuTo = [this](int trajectory_id, double elapsed_time) {
    TrajectoryBuilderInterface* trajectory_builder =
        map_builder_->GetTrajectoryBuilder(trajectory_id);
    trajectory_builder->AddSensorData(
        kIMUSensorId.id,
        sensor::ImuData{cartographer::common::FromUniversal(123) +
                            cartographer::common::FromSeconds(elapsed_time),
                        Eigen::Vector3d(0., 0., 9.8), Eigen::Vector3d::Zero()});
  };

  trajectory_builder_options_.mutable_trajectory_builder_2d_options()
    ->set_use_imu_data(true);
  BuildMapBuilder();

  int trajectory_id_1 = map_builder_->AddTrajectoryBuilder(
      {kRangeSensorId, kIMUSensorId}, trajectory_builder_options_,
      nullptr /* GetLocalSlamResultCallbackForSubscriptions */);
  addFakeRangeTo(trajectory_id_1, 0.); // data not dispatched as there is no imu

  int trajectory_id_2 = map_builder_->AddTrajectoryBuilder(
      {kRangeSensorId}, trajectory_builder_options_,
      nullptr /* GetLocalSlamResultCallbackForSubscriptions */);

  // data not dispatched by internal queue as there is no data for 2nd trajectory
  addFakeImuTo(trajectory_id_1, 0.);

  // mark internal queue as finished
  map_builder_->FinishTrajectory(trajectory_id_1);

  // dispatch of imu (as imu key is sorted before range)
  // which triggers the following check:
  // F0706 17:05:18.893314  24578 pose_graph_2d.cc:720] trajectory_id 0 has finished but modification is requested, skipping
  addFakeRangeTo(trajectory_id_2, 1.);
  map_builder_->FinishTrajectory(trajectory_id_2);
}
```
Is it possible to change fatal error to warning as there is: `// TODO(gaschler): Replace all FATAL to WARNING after some testing.`?

damienrg, 2018-07-25:
After replacing `LOG(FATAL)` by `LOG(WARNING)` I sometimes got assertions:
 - `F0725 08:39:37.499804  5049 pose_graph_2d.cc:318] Check failed: data_.submap_data.at(submap_id).state == SubmapState::kActive`
 - `F0725 10:39:42.344398 18830 pose_graph_2d.cc:185] Check failed: data_.trajectories_state.at(trajectory_id).state != TrajectoryState::DELETED`

Unfortunately, I am not able to reproduce it in a test case for now but the problem is due to the fact that 
`PoseGraph2D::AddNode()` does not check if we can add a work item that modify the trajectory. A possible fix could be:
```c++
NodeId PoseGraph2D::AddNode(
    std::shared_ptr<const TrajectoryNode::Data> constant_data,
    const int trajectory_id,
    const std::vector<std::shared_ptr<const Submap2D>>& insertion_submaps) {
  // [...] skipped for simplification
  AddWorkItem([=]() EXCLUDES(mutex_) {
    if (!CanAddWorkItemModifying(trajectory_id)) { // <--- check added here
      return WorkItem::Result::kDoNotRunOptimization;
    }
    return ComputeConstraintsForNode(node_id, insertion_submaps,
                                     newly_finished_submap);
  });
  return node_id;
}
```
If you agree with the fix I could send PR.

damienrg, 2018-07-25:
Well the fix is not complete as I now got:
```F0725 11:10:37.249950 23600 pose_graph_2d.cc:186] Check failed: data_.trajectories_state.at(trajectory_id).state != TrajectoryState::FINISHED                                                                   
*** Check failure stack trace: ***                                                                                                                                                                              
    @     0x7fb65afaf0cd  google::LogMessage::Fail()                                                                                                                                                            
    @     0x7fb65afb0f33  google::LogMessage::SendToLog()                                                                                                                                                       
    @     0x7fb65afaec28  google::LogMessage::Flush()                                                                                                                                                           
    @     0x7fb65afb1999  google::LogMessageFatal::~LogMessageFatal()                                                                                                                                           
    @     0x7fb65c3b8d3c  cartographer::mapping::PoseGraph2D::AddTrajectoryIfNeeded()                                                                                                                           
    @     0x7fb65c3c0cf9  cartographer::mapping::PoseGraph2D::AppendNode()                                                                                                                                      
    @     0x7fb65c3c154f  cartographer::mapping::PoseGraph2D::AddNode()
    @     0x7fb65c40b3fa  cartographer::mapping::(anonymous namespace)::GlobalTrajectoryBuilder<>::AddSensorData()
    @     0x7fb65c34549d  cartographer::sensor::Dispatchable<>::AddToTrajectoryBuilder()
    @     0x7fb65c3f5ee4  cartographer::mapping::CollatedTrajectoryBuilder::HandleCollatedSensorData()
    [...] removed for simplicity
```
Could you explain the rational behind the following code to help me trying to fix this issue:
```c++
AddTrajectoryIfNeeded(submap_id.trajectory_id);
if (!CanAddWorkItemModifying(submap_id.trajectory_id)) return;
```
Especially why the 2 call are not inverted?

gaschler, 2018-07-25:
In my eyes, AddNode should always execute, and we should not do this fix without fully understanding the issue. There is a risk that the fix only covers the real issue.
Can you please explain again in which case the failed check is triggered?
Perhaps we can understand this corner case first and then write a test that reproduces it reliably?

gaschler, 2018-07-25:
> AddTrajectoryIfNeeded(submap_id.trajectory_id);
> if (!CanAddWorkItemModifying(submap_id.trajectory_id)) return;
> Especially why the 2 call are not inverted?

Reading these lines again, I think `if (!CanAddWorkItemModifying(submap_id.trajectory_id)) return;` can even be omitted because `AddTrajectoryIfNeeded` has already passed all the checks.
But it does not hurt.

I think it is rather unrelated to your issue, and is just the point when the checks fail.

(Just a reminder, callers may never add sensor data to a trajectory after finishing it.)

damienrg, 2018-07-25:
Inverting the 2 calls also mean modifying `CanAddWorkItemModifying` or creating another function to return true when there is no trajectory.

damienrg, 2018-07-25:
> Can you please explain again in which case the failed check is triggered?

In my initial report I was doing:
```
  load_map()
  current_trajectory_id = create_trajectory()
  while (true)
    add_data(current_trajectory_id, imu, odo, lidar)
    finish_trajectory_id(current_trajectory_id)
    current_trajectory_id = create_trajectory()
```

Since https://github.com/googlecartographer/cartographer/issues/1212#issuecomment-403065078 I have replaced `FATAL` by `WARNING` and I do:
```
  load_map()
  current_trajectory_id = create_trajectory()
  while (true)
    add_data(current_trajectory_id, imu, odo, lidar)
    old_trajectory_id = current_trajectory_id
    current_trajectory_id = create_trajectory()
    finish_trajectory(old_trajectory_id)
```
This to test that finishing/creating a trajectory is fully supported by cartographer.

damienrg, 2018-07-25:
Note that as data dispatching is delayed by `OrderedMultiQueue`, even though the user call `FinishTrajectory` after adding all data, internally `FinishTrajectory` can be called before data processing is finished, like in the following case:
```c++
// ...
int trajectory_id_1 =  map_builder.AddTrajectoryBuilder();
auto trajectory_builder_1 = map_builder.GetTrajectoryBuilder(trajectory_id_1);
trajectory_builder_1.AddSensorData(lidar_data);
trajectory_builder_1.AddSensorData(imu_data);
trajectory_builder_1.AddSensorData(odo_data);
// ...
int trajectory_id_2 =  map_builder.AddTrajectoryBuilder();
auto trajectory_builder_2 = map_builder.GetTrajectoryBuilder(trajectory_id_2);

// void MapBuilder::FinishTrajectory(const int trajectory_id) {
//  sensor_collator_->FinishTrajectory(trajectory_id);
//  pose_graph_->FinishTrajectory(trajectory_id);
// }
// map_builder.FinishTrajectory() calls queue_.MarkQueueAsFinished() which
// calls Dispatch() to consume remaining data. But as we have another trajectory
// Dispatch() return without processing remaining data as we do not have data
// for trajectory 2.
// map_builder.FinishTrajectory() calls pose_graph_->FinishTrajectory()
// but all data from trajectory 1 was not dispatched.
map_builder.FinishTrajectory(trajectory_id_1);

// may call Dispatch() and then data of trajectory 1 is dispatched
trajectory_builder_2.AddSensorData(lidar_data);
trajectory_builder_2.AddSensorData(imu_data);
trajectory_builder_2.AddSensorData(odo_data);
// ...
```
So we are back to a previous question do you want to support this case in `OrderedMultiQueue`?
If not you should mention that `Collator` should not be used with multiple trajectories on the same robot.

gaschler, 2018-07-27:
> So we are back to a previous question do you want to support this case in OrderedMultiQueue?

What is the expected behavior? Drop remaining items of a finishing trajectory that are waiting for newer timestamps of newly added sensors?
Without a clear expected behavior this issue is not actionable.

> If not you should mention that Collator should not be used with multiple trajectories on the same robot.

The use case that I heard of is finishing a trajectory to "pause SLAM" and adding a new one later, often with the initial pose set to the last known pose. This is supported.

If you don't need strong determinism, `TrajectoryCollator` should work in this case.

damienrg, 2018-08-20:
> What is the expected behavior? Drop remaining items of a finishing trajectory that are waiting for newer timestamps of newly added sensors?

Yes, you could drop remaining items instead of dispatching them.

> Without a clear expected behavior this issue is not actionable.

I understand, I was not probably clear enough. My point is just that I was surprised by the fact that I got an assert when:
   1. I sent sensor data chronologically ordered for each sensor
   2. I finished and started a new trajectory with sensor data also chronologically ordered for each
       sensor (like in the cases discussed)

I wanted to know if you consider it a bug or not. Depending on your answer I would have fixing it or
fixing my code.

> If you don't need strong determinism, TrajectoryCollator should work in this case.

Yes, I now use it.

damienrg, 2018-09-25:
Close as `OrderedMultiQueue` will not be changed to drop remaining items.

closed 2018-09-25 by damienrg