cartographer-project/cartographer #1219: Deletable trajectories + landmarks seem broken

ojura, 2018-06-29:
Hi (@cschuet?), using landmarks seems to clash with the new `CanAddWorkItemModifying` stuff:

```
cartographer\cartographer\mapping\internal\3d\pose_graph_3d.cc:751] trajectory_id:0 has not been added but modification is requested.
```

With a stack trace like this:

```
    @   00007FF6DE1CC233  private: bool __cdecl cartographer::mapping::PoseGraph3D::CanAddWorkItemModifying(int) __ptr64
    @   00007FF6DE1CA7A5  public: virtual void __cdecl cartographer::mapping::PoseGraph3D::AddLandmarkData(int,struct cartographer::sensor::LandmarkData const & __ptr64) __ptr64
    @   00007FF6DE0811FA  public: void __cdecl cartographer_ros::SensorBridge::HandleLandmarkMessage....
```

ojura, 2018-06-29:
I have also noticed that I get a bunch of these for the IMU as well (even if I disable landmarks).

gaschler, 2018-06-29:
So you are getting these as warnings, because nobody has called `AddTrajectoryIfNeeded` yet.
If it goes away after the first node is added, this is harmless.
If it happens later, it would be good to find out the reason.

ojura, 2018-06-29:
Okay, the real bug is that my landmarks queue is blocking everything for some reason. Not sure why that has started happening since I pulled master, will investigate. Thanks.

closed 2018-06-29 by ojura

ojura, 2018-06-30:
Just to follow up, #1184 was causing problems for me. My sensor data was marked as `SensorId::SensorType::FIXED_FRAME_POSE` (because that's what I'm using landmarks for)  instead of `SensorId::SensorType::LANDMARK`, so the multi ordered sensor queue thought it needed to collate this as well.