cartographer-project/cartographer #1220: Add option to skip collator for landmark observations and fixed_frame

cschuet, 2018-07-02:
[Recently](https://github.com/googlecartographer/cartographer/pull/1184) the behavior of landmark observations was changed and they now skip the collator.

This removes determinism in the sense that previously what makes it into an optimization problem was determined by the sensor timestamps alone and now it depends on when you add the landmark observations to the problem not when they occurred.

I think it might be worth adding an option for this with a default to collate landmark observations.

Exactly the same applies to fixed_frame_poses. Same option?

cschuet, 2018-07-02:
@ojura @gaschler FYI

pifon2a, 2018-07-02:
I will add the option. Just as a short background, why it was done: marker/landmark detection can be time consuming, so that the landmark observations can come with a delay. This can be cured by distinguishing the time of observation and the time of landmark detection. Also, landmark data does not come all the time, so we would have to send an empty vector of the landmark observations to unblock the collator.

closed 2018-07-02 by pifon2a