cartographer-project/cartographer #1229: Using Cartographer without ROS

SubMishMar, 2018-07-02:
Is there any tutorial or API available where once can use the cartographer standalone library without using the ROS interface?

ojura, 2018-07-03:
My approach to this is to use cartographer_ros without ros :) I yanked out into a separate library some ROS headers and a small subset of roscpp (rosbag and tf buffers) which is enough to build cartographer_ros. It even works on windows. I can share it if there's any interest.

SubMishMar, 2018-07-03:
That sounds cool! Please share @ojura !! Thanks :)

ojura, 2018-07-03:
https://github.com/larics/cartographer_superbuild/tree/ros_stub

You should be able to build the offline node by calling

```
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make cartographer_offline_node -j8
```
Afterwards you can use the offline node to process ros bags. Do not source /opt/ros/setup.bash, it should be self-contained.

SubMishMar, 2018-07-03:
@ojura Thanks for the reply and I am going to look into this today. But could you please tell me what is the basic difference between cartographer_ros and the repository you have provided here?

SubMishMar, 2018-07-03:
Here are some more queries:
1. Do I still need ROS on my system to run this?
2. Do I need to provide data as ros topics?

SubMishMar, 2018-07-03:
My cmake fails:

```-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "FindGMock.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "GMock", but
  CMake did not find one.

  Could not find a package configuration file provided by "GMock" with any of
  the following names:

    GMockConfig.cmake
    gmock-config.cmake

  Add the installation prefix of "GMock" to CMAKE_PREFIX_PATH or set
  "GMock_DIR" to a directory containing one of the above files.  If "GMock"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/perceptin/github-projects/cartographer_superbuild/build/CMakeFiles/CMakeOutput.log".

ojura, 2018-07-04:
1) You do not need to have ROS installed.
2) You can use the cartographer offline node as a template. In your code you can call instantiate a cartographer node and call Handle<Sensor> methods

The offline node itself should be functional, i.e. it can process ROS bags and output .pbstreams.

Building on a clean Ubuntu 18.04 install:

ojura, 2018-07-04:
1) You do not need to have ROS installed.
2) You can use the cartographer offline node as a template. In your code you can call instantiate a cartographer node and call its `Handle<Sensor>` methods. You pass the data as ROS sensor messages, the headers for them are included in `ros_stub`.

   The offline node itself should be functional, i.e. it can process ROS bags and output .pbstreams. It cannot publish or subscribe to messages, the network functionality in ROS all been replaced with noop functions.

Building on a clean Ubuntu 18.04 install:
```
sudo add-apt-repository universe
sudo apt update
sudo apt install git cmake bzip2 libbz2-dev libceres-dev libprotobuf-dev protobuf-compiler libpcl-dev liburdfdom-headers-dev liburdfdom-model liblz4-dev libtinyxml-dev liburdfdom-dev
git clone https://github.com/larics/cartographer_superbuild --branch ros_stub
./cartographer_superbuild/cartographer/scripts/install_debs_cmake.sh 
cd cartographer_superbuild
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make cartographer_offline_node -j8
```

`cartographer` and `cartographer_ros` folders contain unchanged code from current upstream. However, updating `cartographer_ros` might need new `cartographer_ros_msgs message` headers in `ros_stub`, have a look at the readme for more details.

SubMishMar, 2018-07-05:
Thanks @ojura

aziesemer, 2018-07-18:
I was able to compile it on my MAC installing similar packages using brew. 
However, using Clang it gives me some warnings during compilation and fails in some of the test cases. I tried with GCC and it didn't give me any warning but failed to link.

I'm now trying to make it run with Clang, I created a small program to test it but I don't know yet how to display the map on the screen.

ojura, 2018-07-19:
Check out the occupancy grid node in cartographer_ros and how it constructs the map. Normally it publishes it as a message - you can try adapting it to display it on the screen instread.

ZhiyuLiu1993, 2018-11-20:
@ojura Hello！I don't know how to get the trajectory without ROS，I cross-compiled the  cartographer on the arm，Can you give some advice?

ojura, 2018-11-20:
In node.cc, see calling `map_builder_bridge_.GetTrajectoryNodeList())`.

Or if you have a .pbstream from calling SaveState, see https://gist.github.com/ojura/85507cd6c8be419357305b30d9bb949d

blewando, 2019-10-04:
@ojura I managed to successfully compile your cartographer_offline_node. As far as I understood, it should be possible to run it with a ROS bag. What do I have to type in exactly into the command line in order to run it? When I simply type ./cartographer_offline_node I get the following error:

`F1004 23:00:12.574383 25529 offline_node.cc:80] Check failed:  FLAGS_configuration_directory.empty() -configuration_directory is missing.
F1004 23:00:12.000000 25529 offline_node.cc:80] Check failed: !FLAGS_configuration_directory.empty() -configuration_directory is missing.
`
Adding a config-file from cartographer_ros/launch/ as argument does not help. The error is the same.

Thank you!

klintan, 2019-10-07:
> My cmake fails even though I have google-mock in my system.
> 
> ```
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Error at CMakeLists.txt:10 (find_package):
>   By not providing "FindGMock.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "GMock", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "GMock" with any of
>   the following names:
> 
>     GMockConfig.cmake
>     gmock-config.cmake
> 
>   Add the installation prefix of "GMock" to CMAKE_PREFIX_PATH or set
>   "GMock_DIR" to a directory containing one of the above files.  If "GMock"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/perceptin/github-projects/cartographer_superbuild/build/CMakeFiles/CMakeOutput.log".
> ```

I had this issue as well I solved it by:

1. Install googletest library using instructions on their github. (cmake, make, make install)
2. In the CMakelists.txt file change line 250 from
`target_link_libraries(${TEST_LIB} PUBLIC ${GMOCK_LIBRARY})` --> to  `target_link_libraries(${TEST_LIB} PUBLIC gmock gtest)`

This variables `GMOCK_LIBRARY` is empty, and I can't find where it would be set, however `GMOCK_LIBRARIES` are set, but couldn't get that to work.

You might need to add 
```
find_packages(gmock REQUIRED)
find_packages(gtest REQUIRED)
```
But it shouldn't be necessary, since gmock contains gtest and the `find_packages(gmock)` is already in the functions file referenced in the cmakelists file.

closed 2020-07-09 by wohe