cartographer-project/cartographer #1242: Wrong accumulated_range_data_.origin in local_trajectory_builder.cc??

JoyTam, 2018-07-06:
It's about the code! I don't know if this is wrong but it's confusing me a lot? Much appreciated if anyone can help me out!

Well the question is about a piece of code in function AddRangeDate in local_trajectory_builder_2d.cc . It tries to convert RangeData from tracking frame to local frame and put them into the "accumulated_range_data_" , 
`   const Eigen::Vector3f origin_in_local =
        range_data_poses[i] *
        synchronized_data.origins.at(synchronized_data.ranges[i].origin_index);
    const Eigen::Vector3f hit_in_local = range_data_poses[i] * hit.head<3>();
    const Eigen::Vector3f delta = hit_in_local - origin_in_local;
if (range >= options_.min_range()) {
      if (range <= options_.max_range()) {
        accumulated_range_data_.returns.push_back(hit_in_local);
      } else {
        accumulated_range_data_.misses.push_back(
            origin_in_local +
            options_.missing_data_ray_length() / range * delta);
      }
    }`
well now "hit in local " and "misses in local" are pushed back into returns and misses of RandeData separately.
However, the origin of the RangeDate is set as follow:
` accumulated_range_data_.origin = range_data_poses.back().translation();`
Shouldn't its origin be set as "origin in local " as well?
Here it takes the origin of tracking frame as the origin of RangeData origin. But the PointClouds are not necessarily collected from tracking frame. For example, I set the "track frame" as "base_link", and PointClouds are collected from camera frame. There can be an offset bewteen these two and may not just in z direction, which would make no difference in case of 2d then......

I don't know if i make myself understood here. Anyway, look forward to some reply here.
Thanks!

gaschler, 2018-07-06:
Is this the same as https://github.com/googlecartographer/cartographer/issues/947?

gaschler, 2018-07-09:
@JoyTam Thanks for reporting.
I get the point that the origin is already incorrect for one lidar.
However, when we fix #947 for multiple lidars, this bug will also be fixed.

JoyTam, 2018-07-10:
@gaschler  Do you mean that the next release will fix this? Because i did't find the fix in the newest commit yet.

gaschler, 2018-07-10:
There is no estimated time. 
As a work around, you can configure your urdf and lua with `options.tracking_frame` that the tracking frame coincides with origin of the lidar (unless have an imu, in which case the tracking frame must be at the same position as the imu).

JoyTam, 2018-07-21:
@gaschler  Thank you for your patience!