cartographer-project/cartographer #1260: Tuning / handling for wide open spaces

AlexReimann, 2018-07-11:
I run into some backend problems when it comes to wide open spaces.

I'm running 3D offline mode without just imu and one velodyne. The imu is not very reliable which is why the trajectory builder `occupied_space` weights are rather high:

```lua
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_0 = 30.0
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_1 = 50.0
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.translation_weight = 5.0
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.rotation_weight = 10.0 
```

It seems like the ceres optimization tends to come to the conclusion that the robot did not move at all in absence of lidar data even though the imu reports movement.

Is there any way to fix this by adjusting the configuration or is this a general problem which can not be fixed through parameter tuning? (`use_nonmonotonic_steps` ?)

If the translation weight is increased a lot (e.g. to 300) the result is ok. But for me it seems like the matcher should not just stick to it's position when the  point matching doesn't work well

AlexReimann, 2018-07-11:
Local slam doesn't work because the ceres optimization says we're are not moving because there is no / not enough point data to do proper local slam (which means it should run purely on the imu instead, but it does not because the weights for rotation and translation are low).

I'll try to get a small bag showing the issue.

gaschler, 2018-11-30:
Closing as inactive

closed 2018-11-30 by gaschler