cartographer-project/cartographer #1262: error  catkin_make_isolated --install --use-ninja

Jin-S-Kim, 2018-07-11:
==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/jinsoo/catkin_ws/install_isolated/env.sh'
==> cmake /home/jinsoo/catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/jinsoo/catkin_ws/install_isolated -G Ninja in '/home/jinsoo/catkin_ws/build_isolated/cartographer/install'
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /home/jinsoo/catkin_ws/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP]
CMake Error at /usr/local/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Protobuf: Found unsuitable version "2.6.1", but required is
  at least "3.0.0" (found /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.12/Modules/FindProtobuf.cmake:543 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:36 (find_package)

-- Configuring incomplete, errors occurred!
<== Failed to process package 'cartographer': 
  Command '['/home/jinsoo/catkin_ws/install_isolated/env.sh', 'cmake', '/home/jinsoo/catkin_ws/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/jinsoo/catkin_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/jinsoo/catkin_ws/build_isolated/cartographer && /home/jinsoo/catkin_ws/install_isolated/env.sh cmake /home/jinsoo/catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/jinsoo/catkin_ws/install_isolated -G Ninja

Command failed, exiting.

How can I fix it?

FlorentRevest, 2018-07-11:
You need to install protobuf 3. Compiling it is documented at https://google-cartographer.readthedocs.io/en/latest/index.html#getting-started-without-ros

gaschler, 2018-07-27:
Closing as answered

closed 2018-07-27 by gaschler