cartographer-project/cartographer #1289: Issue on Mac

aziesemer, 2018-07-17:
It seems that cartographer fails in some tests cases with Clang (version 5.0.0) on Mac OS 10.13.6 (I also tried on 10.10 and got similar errors):

The following tests FAILED:
	 21 - cartographer.mapping.2d.submap_2d_test (ILLEGAL)
	 28 - cartographer.mapping.id_test (SEGFAULT)
	 31 - cartographer.mapping.internal.2d.overlapping_submaps_trimmer_2d_test (ILLEGAL)
	 32 - cartographer.mapping.internal.2d.pose_graph_2d_test (ILLEGAL)
	 50 - cartographer.mapping.internal.constraints.constraint_builder_2d_test (ILLEGAL)
	 60 - cartographer.mapping.map_builder_test (ILLEGAL)

It also gives me a few warnings when I'm compiling it. See file attached.

[log.txt](https://github.com/googlecartographer/cartographer/files/2201755/log.txt)

pifon2a, 2018-08-06:
Interesting. We do not officially support mac. Do you get the same issues with gcc?

aziesemer, 2018-08-08:
GCC (I installed using brew and added "SET(CMAKE_CXX_COMPILER /usr/local/Cellar/gcc/8.1.0/bin/g++-8)" to CMakeLists.txt) gives me no warning but fails to link.

ojura, 2018-08-08:
Can you paste the linking error?

aziesemer, 2018-08-08:
Clang gives me some warnings when compiling and fails on these tests:
The following tests FAILED:
	 21 - cartographer.mapping.2d.submap_2d_test (ILLEGAL)
	 23 - cartographer.mapping.2d.tsdf_range_data_inserter_2d_test (Child aborted)
	 28 - cartographer.mapping.id_test (Child aborted)
	 31 - cartographer.mapping.internal.2d.overlapping_submaps_trimmer_2d_test (ILLEGAL)
	 32 - cartographer.mapping.internal.2d.pose_graph_2d_test (ILLEGAL)
	 50 - cartographer.mapping.internal.constraints.constraint_builder_2d_test (ILLEGAL)
	 60 - cartographer.mapping.map_builder_test (ILLEGAL)

GCC gives me no warning when building but fails when linking. The log is attached

I'm using Mac OS 10.13.6 but same happened with 10.10
[cartographer_gcc_log.txt](https://github.com/googlecartographer/cartographer/files/2271760/cartographer_gcc_log.txt)

jihoonl, 2018-12-04:
I am also failing to compile cartographer in Mac OS High Sierra(10.13.6) as a pure cmake project due to `Undefined symbols for architecture x86_64:` while linking `cartographer_pbstream`, `cartographer_autogenerate_ground_truth`, `cartographer_compute_relations_metrics`.

Anyone knows how to fix this? I have tried `clang++`, `g++`, `g++-8`. But, I got no luck so far.

Here is a part of error log.
```
Scanning dependencies of target cartographer_pbstream
Scanning dependencies of target cartographer_compute_relations_metrics
Scanning dependencies of target cartographer_test_library
Scanning dependencies of target cartographer_autogenerate_ground_truth
[ 54%] Building CXX object CMakeFiles/cartographer_pbstream.dir/cartographer/io/pbstream_main.cc.o
[ 54%] Building CXX object CMakeFiles/cartographer_compute_relations_metrics.dir/cartographer/ground_truth/compute_relations_metrics_main.cc.o
[ 54%] Building CXX object CMakeFiles/cartographer_autogenerate_ground_truth.dir/cartographer/ground_truth/autogenerate_ground_truth_main.cc.o
[ 55%] Building CXX object CMakeFiles/cartographer_test_library.dir/cartographer/io/fake_file_writer_test.cc.o
[ 55%] Building CXX object CMakeFiles/cartographer_test_library.dir/cartographer/io/fake_file_writer.cc.o
[ 55%] Building CXX object CMakeFiles/cartographer_test_library.dir/cartographer/io/testing/test_helpers.cc.o
[ 55%] Building CXX object CMakeFiles/cartographer_test_library.dir/cartographer/mapping/internal/testing/test_helpers.cc.o
[ 55%] Linking CXX executable cartographer_pbstream
Undefined symbols for architecture x86_64:
  "google::SetUsageMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      _main in pbstream_main.cc.o
      cartographer::io::pbstream_info(int, char**) in libcartographer.a(pbstream_info.cc.o)
      cartographer::io::pbstream_migrate(int, char**) in libcartographer.a(pbstream_migrate.cc.o)
  "google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()", referenced from:
      std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* google::MakeCheckOpString<int, int>(int const&, int const&, char const*) in libcartographer.a(proto_stream_deserializer.cc.o)
      std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* google::MakeCheckOpString<int, int>(int const&, int const&, char const*) in libcartographer.a(compressed_point_cloud.cc.o)
      std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* google::MakeCheckOpString<unsigned long, int>(unsigned long const&, int const&, char const*) in libcartographer.a(compressed_point_cloud.cc.o)
      std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* google::MakeCheckOpString<float, int>(float const&, int const&, char const*) in libcartographer.a(compressed_point_cloud.cc.o)
  "google::protobuf::MessageLite::ParseFromString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      cartographer::io::ProtoStreamReader::ReadProto(google::protobuf::Message*)     in libcartographer.a(proto_stream.cc.o)
  "google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))", referenced from:
      protobuf_cartographer_2fmapping_2fproto_2fserialization_2eproto::AddDescriptorsImpl()       in libcartographer.a(serialization.pb.cc.o)
      protobuf_cartographer_2fmapping_2fproto_2fsubmap_2eproto::AddDescriptorsImpl()       in libcartographer.a(submap.pb.cc.o)
      protobuf_cartographer_2fmapping_2fproto_2ftrajectory_5fbuilder_5foptions_2eproto::AddDescriptorsImpl()       in libcartographer.a(trajectory_builder_options.pb.cc.o)
      protobuf_cartographer_2fmapping_2fproto_2fpose_5fgraph_2eproto::AddDescriptorsImpl()       in libcartographer.a(pose_graph.pb.cc.o)
      protobuf_cartographer_2fsensor_2fproto_2fsensor_2eproto::AddDescriptorsImpl()       in libcartographer.a(sensor.pb.cc.o)
      protobuf_cartographer_2fmapping_2fproto_2f3d_2fhybrid_5fgrid_2eproto::AddDescriptorsImpl()       in libcartographer.a(hybrid_grid.pb.cc.o)
      protobuf_cartographer_2fmapping_2fproto_2ftrajectory_2eproto::AddDescriptorsImpl()       in libcartographer.a(trajectory.pb.cc.o)
      ...
  "google::protobuf::io::CodedOutputStream::WriteStringWithSizeToArray(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned char*)", referenced from:
      cartographer::mapping::proto::SensorId::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const in libcartographer.a(trajectory_builder_options.pb.cc.o)
      cartographer::mapping::proto::TrajectoryBuilderOptionsWithSensorIds::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const in libcartographer.a(trajectory_builder_options.pb.cc.o)
      cartographer::mapping::proto::PoseGraph_LandmarkPose::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const in libcartographer.a(pose_graph.pb.cc.o)
      cartographer::sensor::proto::LandmarkData_LandmarkObservation::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const in libcartographer.a(sensor.pb.cc.o)
  "google::protobuf::internal::WireFormatLite::WriteBytesMaybeAliased(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::io::CodedOutputStream*)", referenced f [... cut ...]
      vtable for cartographer::mapping::proto::OdometryData in libcartographer.a(serialization.pb.cc.o)
      vtable for cartographer::mapping::proto::FixedFramePoseData in libcartographer.a(serialization.pb.cc.o)
      vtable for cartographer::mapping::proto::LandmarkData in libcartographer.a(serialization.pb.cc.o)
      vtable for cartographer::mapping::proto::TrajectoryData in libcartographer.a(serialization.pb.cc.o)
      ...
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
```

klintan, 2019-10-07:
>   "google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()", referenced from:

Are you sure you have protobuf installed? 

to check run 
`brew info protobuf`

if not run

```brew install protobuf```