cartographer-project/cartographer #1299: Map layering

Rmargin, 2018-07-19:
hi
Map stratification is easy to occur when the backpack is moving at high speed.What caused it?
I don't have access to GPS.
[image]
[image]
[image]

Rmargin, 2018-07-20:
Our test environment is mainly highway, and the speed of movement is about 15km/h. When the speed is faster, the map layer is more obvious.
In the low-speed test process map layering, you can properly circle or wait in the same place, the stratification phenomenon will close, but the high-speed situation does not improve the effect

Rmargin, 2018-07-25:
[image]
[image]
[image]
[image]

Rmargin, 2018-07-25:
[image]
[image]
[image]
[image]
[image]

Rmargin, 2018-07-25:
hi
The trajectory is not closed properly,The speed of the backpack is about 30km/h.The journey is about 2-3km,How to solve this problem?tks.haha

Rmargin, 2018-07-25:
@gaschler

FlorentRevest, 2018-07-25:
If you want people to be helpful:

1. Follow [the issue template](https://github.com/googlecartographer/cartographer_ros/issues/new?labels=question) by uploading your ``.bag`` file and ``cartographer_ros`` fork.
2. Edit your comment instead of posting multiple comments in a row
3. Ask a **clear** question. "how to solve this" is not a clear question. Tons of screenshots is not a question either.
4. Don't highlight developers, they already get notifications. If they want to answer, they will, whenever their time will permit.

Rmargin, 2018-07-26:
ok! haha

closed 2018-07-26 by Rmargin