cartographer-project/cartographer #1328: Pure Localization Option

marcelpatzsch, 2018-07-24:
When `TRAJECTORY_BUILDER.pure_localization = true` is set in the lua file, as stated [here](https://github.com/googlecartographer/cartographer_ros/blob/master/docs/source/tuning.rst), the cartographer_node dies at startup. The error provided is the following:
`[FATAL] [1532423911.518104549]: F0724 11:18:31.000000  5704 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'pure_localization' was used the wrong number of times.`

Has this feature been disabled?

I'm using the cartographer a08037dc47d544d697261940727b934fefd8dfb9 (Jul 13)
and cartographer_ros [3cb7ef](https://github.com/googlecartographer/cartographer_ros/commit/3cb7efd62c38125081b86844c5b2fee3147d0667) (Jun 8).
The whole config.lua is available [here](https://gist.github.com/marcelpatzsch/f8b79f5ffd184737614e6e88d56d3f86).

gaschler, 2018-07-24:
The format of the lua config (and the protobuf options) has changed to support more generic trimmers.

Do not set `pure_localization = true`, this was the old format.

Instead, set only
```
TRAJECTORY_BUILDER.pure_localization_trimmer = {
  max_submaps_to_keep = 3,
}
```

Examples are `backpack_2/3d_localization.lua` in cartographer_ros/configuration_files.

marcelpatzsch, 2018-07-24:
Ok thank you!

closed 2018-07-24 by marcelpatzsch