cartographer-project/cartographer #1347: FindAbseil is platform-specific

ojura, 2018-07-27:
@pifon2a I've tried out building the new master on windows (MSVC14). The good news is that the `externalproject_add` stuff seems to work on windows! In fact, everything builds okay, and it only fails on linking, because the library names you've put into `FindAbseil.cmake` are platform specific. On windows, the libraries end up being called like this:

```
[13/117] Linking CXX static library absl\debugging\absl_stack_consumption.lib
[14/117] Linking CXX static library absl\container\absl_container.lib
[16/117] Linking CXX static library absl\memory\absl_memory.lib
[20/117] Linking CXX static library absl\algorithm\absl_algorithm.lib
[21/117] Linking CXX static library absl\base\absl_dynamic_annotations.lib
[27/117] Linking CXX static library absl\container\test_instance_tracker_lib.lib
[46/117] Linking CXX static library absl\debugging\absl_examine_stack.lib
[47/117] Linking CXX static library absl\base\absl_spinlock_wait.lib
[49/117] Linking CXX static library absl\debugging\absl_stacktrace.lib
[58/117] Linking CXX static library absl\base\absl_throw_delegate.lib
[64/117] Linking CXX static library absl\numeric\absl_int128.lib
[68/117] Linking CXX static library absl\numeric\absl_numeric.lib
[71/117] Linking CXX static library absl\base\absl_malloc_internal.lib
[85/117] Linking CXX static library absl\base\absl_base.lib
[86/117] Linking CXX static library absl\utility\absl_utility.lib
[87/117] Linking CXX static library absl\meta\absl_meta.lib
[88/117] Linking CXX static library absl\debugging\absl_leak_check.lib
[89/117] Linking CXX static library absl\types\absl_span.lib
[90/117] Linking CXX static library absl\debugging\absl_symbolize.lib
[92/117] Linking CXX static library absl\debugging\absl_debugging.lib
[95/117] Linking CXX static library absl\strings\absl_strings.lib
[97/117] Linking CXX static library absl\strings\str_format_extension_internal.lib
[102/117] Linking CXX static library absl\types\absl_bad_any_cast.lib
[103/117] Linking CXX static library absl\strings\str_format_internal.lib
[104/117] Linking CXX static library absl\types\absl_variant.lib
[105/117] Linking CXX static library absl\types\absl_any.lib
[106/117] Linking CXX static library absl\types\absl_bad_optional_access.lib
[107/117] Linking CXX static library absl\strings\absl_str_format.lib
[109/117] Linking CXX static library absl\types\absl_optional.lib
[115/117] Linking CXX static library absl\time\absl_time.lib
[116/117] Linking CXX static library absl\synchronization\absl_synchronization.lib
[117/117] Linking CXX static library absl\debugging\absl_failure_signal_handler.lib
```

So no `lib` prefix, and `.lib` instead of `.a`.

ojura, 2018-07-27:
Related:
https://stackoverflow.com/a/6355851

pifon2a, 2018-07-27:
I'll fix the library names. Now I have a bigger problem: https://github.com/googlecartographer/cartographer_ros/pull/955 works only on 3 platforms out of 4 for some reason.

ojura, 2018-07-27:
#1348 should fix this.

pifon2a, 2018-07-27:
please, use [CMAKE_STATIC_LIBRARY_SUFFIX](https://cmake.org/cmake/help/v3.0/variable/CMAKE_STATIC_LIBRARY_SUFFIX.html)

and

 [CMAKE_STATIC_LIBRARY_PREFIX](https://cmake.org/cmake/help/v3.0/variable/CMAKE_STATIC_LIBRARY_PREFIX.html)

PR #1348 (2018-07-27): FindAbseil.cmake: cross-platform library names
Fixes #1347
commit 27a7a3db (2018-07-27): FindAbseil.cmake: cross-platform library names (#1348)

Fixes #1347
closed 2018-07-27 by wally-the-cartographer