cartographer-project/cartographer #1372: real_time_correlative_scan_matcher_3d_test failed

EdwardLiuyc, 2018-08-07:
I change the resolution of hybrid_grid_ real_time_correlative_scan_matcher_3d_test.cc to 0.05f or 0.2f, the test ends up failed. The test succeed only in resolution 0.1f, why?
`hybrid_grid_(0.05f),`
```
[       OK ] RealTimeCorrelativeScanMatcher3DTest.AlongXYZ (265 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX
I0807 15:38:30.244786 12696 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.899924
../cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d_test.cc:71: Failure
Value of: pose
Expected: is nearly { t: [-1.000000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
  Actual: { t: [-1.000000, 0.000000, 0.000000], q: [0.999988, 0.003732, -0.003249, -0.000023] }
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX (264 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundY
I0807 15:38:30.509793 12696 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.899848
[       OK ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundY (265 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundYZ
I0807 15:38:30.778345 12696 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.899696
[       OK ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundYZ (269 ms)
[----------] 7 tests from RealTimeCorrelativeScanMatcher3DTest (1994 ms total)

[----------] Global test environment tear-down
[==========] 7 tests from 1 test case ran. (1994 ms total)
[  PASSED  ] 6 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX
```
`hybrid_grid_(0.2f),`
```
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.AlongXYZ
I0807 15:42:46.291740 12844 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.89928
../cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d_test.cc:71: Failure
Value of: pose
Expected: is nearly { t: [-1.000000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
  Actual: { t: [-0.900000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.AlongXYZ (0 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX
I0807 15:42:46.292315 12844 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.9
../cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d_test.cc:71: Failure
Value of: pose
Expected: is nearly { t: [-1.000000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
  Actual: { t: [-1.000000, 0.000000, 0.000000], q: [0.999976, 0.006981, 0.000000, 0.000000] }
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX (1 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundY
I0807 15:42:46.292858 12844 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.9
../cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d_test.cc:71: Failure
Value of: pose
Expected: is nearly { t: [-1.000000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
  Actual: { t: [-1.000000, 0.000000, 0.000000], q: [0.999976, 0.000000, 0.006981, 0.000000] }
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundY (0 ms)
[ RUN      ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundYZ
I0807 15:42:46.293411 12844 real_time_correlative_scan_matcher_3d_test.cc:70] Score: 0.9
../cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d_test.cc:71: Failure
Value of: pose
Expected: is nearly { t: [-1.000000, 0.000000, 0.000000], q: [1.000000, 0.000000, 0.000000, 0.000000] }
  Actual: { t: [-1.000000, 0.000000, 0.000000], q: [0.999951, 0.000000, 0.006981, 0.006981] }
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundYZ (1 ms)
[----------] 7 tests from RealTimeCorrelativeScanMatcher3DTest (4 ms total)

[----------] Global test environment tear-down
[==========] 7 tests from 1 test case ran. (4 ms total)
[  PASSED  ] 3 tests.
[  FAILED  ] 4 tests, listed below:
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.AlongXYZ
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundX
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundY
[  FAILED  ] RealTimeCorrelativeScanMatcher3DTest.RotationAroundYZ
```

EdwardLiuyc, 2018-09-14:
@SirVer @gaschler @wohe @cschuet 
any idea?

gaschler, 2018-09-14:
In my eyes this sounds like a false alarm. I guess the accuracies of the checks in the test are too strict.

EdwardLiuyc, 2018-09-15:
> In my eyes this sounds like a false alarm. I guess the accuracies of the checks in the test are too strict.

Thanks a lot, I get it but some time I cought this fault but I have no idea about this : 
```
[ RUN      ] RealTimeCorrelativeScanMatcherTest.AlongZ
I0915 11:30:45.266042  4900 real_time_correlative_scan_matcher_test.cc:71] Score: 0.899797
unknown file: Failure
C++ exception with description "basic_string::_M_create" thrown in the test body.
[  FAILED  ] RealTimeCorrelativeScanMatcherTest.AlongZ (3 ms)
[ RUN      ] RealTimeCorrelativeScanMatcherTest.AlongXYZ
I0915 11:30:45.268503  4900 real_time_correlative_scan_matcher_test.cc:71] Score: 0.897157
unknown file: Failure
C++ exception with description "basic_string::_M_create" thrown in the test body.
[  FAILED  ] RealTimeCorrelativeScanMatcherTest.AlongXYZ (2 ms)
[ RUN      ] RealTimeCorrelativeScanMatcherTest.RotationAroundX
I0915 11:30:45.270977  4900 real_time_correlative_scan_matcher_test.cc:71] Score: 0.899657
unknown file: Failure
C++ exception with description "basic_string::_M_create" thrown in the test body.
[  FAILED  ] RealTimeCorrelativeScanMatcherTest.RotationAroundX (3 ms)
```

gaschler, 2018-09-17:
@EdwardLiuyc Can you reproduce in a debugger and provide a stack trace for this uncaught exception? This would be really helpful.

gaschler, 2018-11-30:
Closing as inactive

closed 2018-11-30 by gaschler