cartographer-project/cartographer #1391: Failed to install/build Google Cartographer

Semih67, 2018-08-15:
I'm trying to install google cartographer but I got some error when i tried it.

So if I type in catkin_make_isolated I get these errors:

Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for gflags, will perform search for installed gflags components.
-- Found Google Flags header in: /usr/include, in namespace: google
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if availab

Scanning dependencies of target cartographer_rosbag_validate
[ 77%] Building CXX object cartographer_ros/CMakeFiles/cartographer_rosbag_validate.dir/rosbag_validate_main.cc.o
/opt/ros/kinetic/lib/liburdf.so: undefined reference to `rosconsole_bridge::RegisterOutputHandlerProxy::~RegisterOutputHandlerProxy()'
collect2: error: ld returned 1 exit status
cartographer_ros/CMakeFiles/cartographer_offline_node.dir/build.make:215: recipe for target '/home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_offline_node' failed
make[2]: *** [/home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_offline_node] Error 1
CMakeFiles/Makefile2:3532: recipe for target 'cartographer_ros/CMakeFiles/cartographer_offline_node.dir/all' failed
make[1]: *** [cartographer_ros/CMakeFiles/cartographer_offline_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 80%] Linking CXX executable /home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_assets_writer
/opt/ros/kinetic/lib/liburdf.so: undefined reference to `rosconsole_bridge::RegisterOutputHandlerProxy::~RegisterOutputHandlerProxy()'
collect2: error: ld returned 1 exit status
cartographer_ros/CMakeFiles/cartographer_assets_writer.dir/build.make:241: recipe for target '/home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_assets_writer' failed
make[2]: *** [/home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_assets_writer] Error 1
CMakeFiles/Makefile2:3643: recipe for target 'cartographer_ros/CMakeFiles/cartographer_assets_writer.dir/all' failed
make[1]: *** [cartographer_ros/CMakeFiles/cartographer_assets_writer.dir/all] Error 2
[ 82%] Linking CXX executable /home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_node
[ 82%] Built target cartographer_node
[ 85%] Linking CXX executable /home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_pbstream_map_publisher
[ 85%] Built target cartographer_pbstream_map_publisher
[ 87%] Linking CXX executable /home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_occupancy_grid_node
[ 87%] Built target cartographer_occupancy_grid_node
[ 90%] Linking CXX executable /home/nvidia/catkin_ws/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_rosbag_validate
[ 90%] Built target cartographer_rosbag_validate
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'cartographer_ros': 
  Command '['/home/nvidia/catkin_ws/devel_isolated/cartographer/env.sh', 'make', '-j6', '-l6']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/nvidia/catkin_ws/build_isolated/cartographer_ros && /home/nvidia/catkin_ws/devel_isolated/cartographer/env.sh make -j6 -l6

Command failed, exiting.
catkin_make_isolated  2668.78s user 146.34s system 407% cpu 11:30.08 total

I already asked this on the ROS-forum and I got the answer: "Best guess, your versions of liburdf and rosconsole_bridge have an ABI mismatch. Maybe try an upgrade or rebuild of both packages?" I can't test it because I'm not working now so I want to ask everywhere and want to get as many answers as possible. 
Do you know whats going on here?

mojovski, 2018-09-10:
Compiling robot_state_publisher and rosconsole_bridge from source resolves this:
```python
FROM osrf/ros:kinetic-desktop-full

RUN apt-get update 

RUN apt-get install -y curl \
	build-essential \
    git  \
    ros-kinetic-catkin

WORKDIR /root
ENV CATKIN_WS /root/catkin_ws/

RUN mkdir -p $CATKIN_WS/src
WORKDIR $CATKIN_WS
RUN wstool init src

RUN echo "source /opt/ros/kinetic/setup.bash" >> /root/.bashrc
RUN bash -c "source /root/.bashrc"
RUN /bin/bash -c '. /opt/ros/kinetic/setup.bash; cd $CATKIN_WS; catkin_make'

WORKDIR $CATKIN_WS/src

RUN git clone https://github.com/ros/robot_state_publisher.git
RUN cd robot_state_publisher && git checkout kinetic-devel

WORKDIR $CATKIN_WS

RUN apt-get install -y ros-kinetic-orocos-kdl 
RUN /bin/bash -c '. /opt/ros/kinetic/setup.bash; catkin_make'

```

gaschler, 2018-11-30:
closing as inactive

closed 2018-11-30 by gaschler