cartographer-project/cartographer #1393: Submaps seem to have no constraints between them.

thomasyoung, 2018-08-17:
# System
Ubuntu 16.04, Bazel 0.16.1,

# Phenomena
I am using cartographer without ROS. I have simulated Lidar giving scan data and simulated odometry data.
From `LocalSlamResultCallback` in `MapBuilder::AddTrajectoryBuilder()` I am seeing reasonable trajectory.
From `MapBuilder::pose_graph()->GetAllSubmapData()` I got `MapById<SubmapId, SubmapData>` and I am seeing reasonable pieces in `SubmapData::submap->grid()` which is `const ::cartographer::mapping::ProbabilityGrid*`.
I try to use `Submap::pose` to stitch these pieces together.
And I see these Submaps seem to have correct orientation but in-consistent translation.

# Visualization

I use pixel size of `0.05` and apply ``Submap::pose` to pixels before putting them together.

# Feeding Data
Initialization:
```
  std::set<::cartographer::mapping::TrajectoryBuilderInterface::SensorId> expected_sensor_ids{
      {::cartographer::mapping::TrajectoryBuilderInterface::SensorId::SensorType::RANGE,
       kLidarName},
      {::cartographer::mapping::TrajectoryBuilderInterface::SensorId::SensorType::ODOMETRY,
       kOdometryName}};
  trajectory_id_ = map_builder_->AddTrajectoryBuilder(
      expected_sensor_ids, trajectory_builder_options,
      [this](int trajectory_id, ::cartographer::common::Time time,
             ::cartographer::transform::Rigid3d local_pose,
             ::cartographer::sensor::RangeData range_data_in_local,
             std::unique_ptr<
                 const ::cartographer::mapping::TrajectoryBuilderInterface::InsertionResult>) {
        OnLocalSlamResult(trajectory_id, time, local_pose, range_data_in_local);
      });
  trajectory_builder_ = map_builder_->GetTrajectoryBuilder(trajectory_id_);
```
Provide Data:
```
  // world_T_lidar_rigid3 is the Rigid3 that convert from Lidar frame to world frame
  trajectory_builder_->AddSensorData(
      kOdometryName, ::cartographer::sensor::OdometryData{scan_time, *world_T_lidar_rigid3});
  trajectory_builder_->AddSensorData(
      kLidarName, ::cartographer::sensor::TimedPointCloudData{
                      scan_time, Vector3f::Zero(),
                      ::cartographer::sensor::TransformTimedPointCloud(
                          pointcloud.points, ::cartographer::transform::Rigid3f::Identity())});
```

# Config

```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
}

MAP_BUILDER.use_trajectory_builder_2d = true
MAP_BUILDER.use_trajectory_builder_3d = false
MAP_BUILDER.pose_graph.optimize_every_n_nodes = 2
MAP_BUILDER.pose_graph.global_constraint_search_after_n_seconds = 3
MAP_BUILDER.pose_graph.constraint_builder.max_constraint_distance = 100
TRAJECTORY_BUILDER_2D.use_imu_data = false

TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability = 0.6
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability = 0.4
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 50
```

kdaun, 2018-08-20:
Thank you for the detailed overview.
Can you please post the whole log as text?

> And I see these Submaps seem to have correct orientation but in-consistent translation.

Can you elaborate on:
- Are the translations off by a few centimeters and roughly correct?
- Are they completely wrong in terms of direction and scale? 

> I0816 18:05:41.186154 27835 constraint_builder_2d.cc:282] 1 computations resulted in 0 additional constraints.

This indicates there was 1 candidate for loop closure evaluated which did not meet
`POSE_GRAPH.constraint_builder.min_score` and therefore was not added as a constraint.
- Reduce the value of `POSE_GRAPH.constraint_builder.min_score` to accept more (and potentially worse candidates as constraints).
- Increasing `POSE_GRAPH.constraint_builder.sampling_ratio` to evaluate more candidates.

Check the [Tuning Guide](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html) for more information.

kdaun, 2018-08-21:
> TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability = 0.9
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability = 0.1

These values don't look reasonable to me. With these update probabilities each update step is almost completely overwriting the previous updates in the submap (read [here](https://ai.google/research/pubs/pub45466) for more background).
`hit_probability = 0.55` and `miss_probability = 0.49` are usually reasonable starting values.

> POSE_GRAPH.optimize_every_n_nodes = 10
POSE_GRAPH.optimize_every_n_nodes = 2

This parameter is set twice in your configuration. Try setting it to the same value as `TRAJECTORY_BUILDER_2D.submaps.num_range_data`.

What exactly are the black/white maps showing?

The blue/cyan map shows a lot of single rays at the boundary of the map. If laser rangefinders observe free space, sometimes they can give you such ghost points. Especially in combination with the low value for the miss probability that can lead to artifacts similar to the artifacts shown in your map. Try adding min/max range parameters, e.g.
`TRAJECTORY_BUILDER_2D.min_range = 1.5`
`TRAJECTORY_BUILDER_2D.max_range = 20`.

thomasyoung, 2018-08-21:
@kdaun Thanks for suggestions! Now I am using following parameters:
```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
}

MAP_BUILDER.use_trajectory_builder_2d = true
MAP_BUILDER.use_trajectory_builder_3d = false

TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 4e2
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 2e2
TRAJECTORY_BUILDER_2D.max_range = 20
TRAJECTORY_BUILDER_2D.min_range = 1.5
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 100
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability = 0.55
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability = 0.49
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

POSE_GRAPH.constraint_builder.max_constraint_distance = 100
POSE_GRAPH.constraint_builder.min_score=0.2
POSE_GRAPH.constraint_builder.sampling_ratio = 0.7
POSE_GRAPH.global_constraint_search_after_n_seconds = 3
POSE_GRAPH.optimize_every_n_nodes = 100
```
Besides I have `map_builder_options.set_collate_by_trajectory(true);`
The black/white maps are created using following code (roughly):
```
  const auto submap_list = map_builder_->pose_graph()->GetAllSubmapData();
  int count = 0;
  for (const auto& map_data : submap_list) {
    std::string label = absl::StrCat("submap_", count);
    count++;
    // All submaps are Submap2D with ProbabilityGrid inside.
    const std::shared_ptr<const ::cartographer::mapping::Submap> submap = map_data.data.submap;
    const ::cartographer::mapping::Submap2D* submap_2d =
        static_cast<const ::cartographer::mapping::Submap2D*>(submap.get());
    CHECK(submap_2d != nullptr, "Submap is not Submap2D");
    CHECK(submap_2d->grid()->GetGridType() == ::cartographer::mapping::GridType::PROBABILITY_GRID,
           "grid2d is not ProbabilityGrid");

      const ::cartographer::mapping::ProbabilityGrid* prob_grid =
          static_cast<const ::cartographer::mapping::ProbabilityGrid*>(submap_2d->grid());
      Eigen::Array2i offset;
      ::cartographer::mapping::CellLimits cell_limits;
      prob_grid->ComputeCroppedLimits(&offset, &cell_limits);
      submap_.resize(cell_limits.num_y_cells, cell_limits.num_x_cells); // rows-by-cols 4 channels image of uint8_t
      auto submap_write_iter = submap_.raw_begin(); // Go over each pixel row by row
      for (const Eigen::Array2i& xy_index :
           ::cartographer::mapping::XYIndexRangeIterator(cell_limits)) {
        uint8_t val = 0;
        if (prob_grid->IsKnown(xy_index + offset)) {
          // per comment in cartographer:
          // Converts a probability to a log odds integer. 0 means unknown, [kMinLogOdds,
          // kMaxLogOdds] is mapped to [1, 255].
          val = ::cartographer::mapping::ProbabilityToLogOddsInteger(
              prob_grid->GetProbability(xy_index + offset));
        }
        *submap_write_iter = val;
        submap_write_iter++;
        *submap_write_iter = val;
        submap_write_iter++;
        *submap_write_iter = val;
        submap_write_iter++;
        *submap_write_iter = val;
        submap_write_iter++;
      }

      auto const map_pose3d = map_data.data.pose;
      Rigid2d offset_pose2d = Rigid2d::Translation(kPixelSize * offset.matrix().cast<double>());

      Draw(submap_, Rigid3dToRidge2dOnXY(map_pose3d)* offset_pose2d, kPixelSize); // Draws to canvas with pose
  }
```
Following are the result I got (Visualized using my code above and `PaintSubmapSlices()` separately:

[image]
[image]

Here is the neighborhood that I am simulating flat Lidar scan from:
[image]

We can see that the 2 submaps are off by couple of meters.

kdaun, 2018-08-22:
Thank you for the update.
Can you elaborate a little bit more on your system?
How do you simulate the LiDAR returns and the odometry? With which frequency are they providing observations?
The lots of long lines in your map look somewhat unusual to me.

For tuning your system I would suggest trying to tune local SLAM first, to avoid potential issues from the constraint detection. And then if local SLAM looks good, start tuning global SLAM. As your environment is small local SLAM should already give sufficient results. Therefore:

1. Validate your input data is correct. Visualize simulated LiDAR + Odometry results
2. Local SLAM only. Turn off the constraint search + correlative_scan_matching and create only a single submap 
`TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = false`
`TRAJECTORY_BUILDER_2D.submaps.num_range_data = 1e8`
`POSE_GRAPH.constraint_builder.sampling_ratio = 1e8`
`POSE_GRAPH.optimize_every_n_nodes = 1e8`
And see if you can get a consistent map by tuning:
`TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability`
`TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability`
`TRAJECTORY_BUILDER_2D.num_accumulated_range_data`
`TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight`
`TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight`
3. If Local SLAM works for you, you can start tuning the global SLAM as mentionend in the [Tuning Guide](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#example-tuning-local-slam).

thomasyoung, 2018-08-30:
@kdaun Thanks for helping!

Believe the local SLAM is working fine as each submap looks reasonable.
Turns out that local_pose for every submap has 0 rotation. And following code gives translation of submaps that make them match together:

```
// Converts to 2d matrix as image
  for (int row = 0; row < rows; row++) {
    for (int col = 0; col < cols; col++) {
      // Compute corresponding location in probability grid
      const Eigen::Array2i carto_cell =
          offset_cells + Eigen::Array2i{cols - 1 - col, rows - 1 - row};
      image(row, col) = ::cartographer::mapping::ProbabilityToLogOddsInteger(
                        prob_grid->GetProbability(carto_cell));
}
}
```
      const auto map_limits = prob_grid->limits();
      const double resolution = map_limits.resolution();

      const double row_translation =
          map_limits.max().x()/resolution - (offset_cells.y() + cell_limits.num_y_cells);
      const double col_translation =
          map_limits.max().y()/resolution - (offset_cells.x() + cell_limits.num_x_cells);
 ```
But I could not find any relation between these translation and the one in `submap->local_pose`.

closed 2018-09-02 by thomasyoung