cartographer-project/cartographer #1395: catkin_make_isolated --install --use-ninja, compile error

nsslab, 2018-08-19:
I use this command to compile cartographer package 

 catkin_make_isolated --install --use-ninja

but, I keep getting this error below 

Is there any solution for this issue?

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/nsslab/Howon/ros/catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
[1/1] Re-running CMake...
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /home/nsslab/Howon/ros/catkin_ws/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install
ninja: no work to do.
==> ninja -j8 -l8 in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
[47/338] Building documentation.
Running Sphinx v1.3.6
making output directory...
loading pickled environment... not yet created
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 5 source files that are out of date
updating environment: 5 added, 0 changed, 0 removed
reading sources... [ 20%] configuration
reading sources... [ 40%] cost_functions
reading sources... [ 60%] evaluation
reading sources... [ 80%] index
reading sources... [100%] terminology

looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [ 20%] configuration
writing output... [ 40%] cost_functions
writing output... [ 60%] evaluation
writing output... [ 80%] index
writing output... [100%] terminology

generating indices... genindex
writing additional pages... search
copying images... [ 50%] high_level_system_overview.png
copying images... [100%] autogenerate_groundtruth.png

copying static files... done
copying extra files... done
dumping search index in English (code: en) ... done
dumping object inventory... done
build succeeded.
[195/338] Linking CXX executable cartographer_migrate_serialization_format
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   CMakeFiles/cartographer_migrate_serialization_format.dir/cartographer/io/migrate_serialization_format_main.cc.o  -o cartographer_migrate_serialization_format  -rdynamic libcartographer.a /home/nsslab/Howon/ros/catkin_ws/install_isolated/lib/libceres.a -lglog -lgflags -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog -lgflags -lcairo -lprotobuf -lpthread && :
libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::Submap::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
serialization.pb.cc:(.text+0x123): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::Node::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
serialization.pb.cc:(.text+0x19b): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::SerializedData::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
serialization.pb.cc:(.text+0x233): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::SerializationHeader::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const':
[... 1099 lines cut ...]
libcartographer.a(probability_grid.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2f2d_2fprobability_5fgrid_2eproto::TableStruct::InitDefaultsImpl()':
probability_grid.pb.cc:(.text+0x648): undefined reference to `google::protobuf::internal::InitProtobufDefaults()'
libcartographer.a(probability_grid.pb.cc.o): In function `cartographer::mapping::proto::ProbabilityGrid::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)':
probability_grid.pb.cc:(.text+0x7a4): undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'
probability_grid.pb.cc:(.text+0x7c6): undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'
libcartographer.a(probability_grid.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2f2d_2fprobability_5fgrid_2eproto::TableStruct::InitDefaultsImpl()':
probability_grid.pb.cc:(.text+0x660): undefined reference to `google::protobuf::internal::OnShutdownDestroyMessage(void const*)'
libcartographer.a(probability_grid.pb.cc.o): In function `void google::protobuf::Arena::Own<cartographer::mapping::proto::ProbabilityGrid>(cartographer::mapping::proto::ProbabilityGrid*)':
probability_grid.pb.cc:(.text._ZN6google8protobuf5Arena3OwnIN12cartographer7mapping5proto15ProbabilityGridEEEvPT_[_ZN6google8protobuf5Arena3OwnIN12cartographer7mapping5proto15ProbabilityGridEEEvPT_]+0xd): undefined reference to `google::protobuf::internal::ArenaImpl::AddCleanup(void*, void (*)(void*))'
libcartographer.a(probability_grid.pb.cc.o):(.data.rel.ro._ZTVN12cartographer7mapping5proto15ProbabilityGridE[_ZTVN12cartographer7mapping5proto15ProbabilityGridE]+0xb8): undefined reference to `google::protobuf::Message::SpaceUsedLong() const'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::CellLimits::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
cell_limits.pb.cc:(.text+0x11e): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::CellLimits::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const':
cell_limits.pb.cc:(.text+0x1c3): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::CellLimits::ByteSizeLong() const':
cell_limits.pb.cc:(.text+0x23b): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2f2d_2fcell_5flimits_2eproto::(anonymous namespace)::protobuf_RegisterTypes(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
cell_limits.pb.cc:(.text+0x2ba): undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2f2d_2fcell_5flimits_2eproto::(anonymous namespace)::protobuf_AssignDescriptors()':
cell_limits.pb.cc:(.text+0x52d): undefined reference to `google::protobuf::internal::AssignDescriptors(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::MessageFactory*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2f2d_2fcell_5flimits_2eproto::TableStruct::InitDefaultsImpl()':
cell_limits.pb.cc:(.text+0x738): undefined reference to `google::protobuf::internal::InitProtobufDefaults()'
libcartographer.a(cell_limits.pb.cc.o): In function `cartographer::mapping::proto::CellLimits::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)':
cell_limits.pb.cc:(.text+0x9a7): undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'
cell_limits.pb.cc:(.text+0x9f3): undefined reference to `google::protobuf::io::Co

kdaun, 2018-08-20:
It looks like some of your proto dependencies are messed up.
Please try:
1. Update cartographer and cartographer_ros and do a clean (delete build_isolated, devel_isolated and install_isolated) rebuild.
2. Run `protoc --version` and copy the output.

nsslab, 2018-08-21:
1. I have followed what u said , and it still has some problems below.

==> Processing plain cmake package: 'cartographer'
==> Creating build directory: 'build_isolated/cartographer/install'
==> Building with env: '/home/nsslab/Howon/ros/catkin_ws/install_isolated/env.sh'
==> cmake /home/nsslab/Howon/ros/catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/nsslab/Howon/ros/catkin_ws/install_isolated -G Ninja in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
Files /home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install/AllFiles.cmake and - differ
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found GMock: gmock_main;-lpthread  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE - Success
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.13.0 installed in: /home/nsslab/Howon/ros/catkin_ws/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
-- Found Lua: /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.2.4") 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "3.0.0") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'cairo>=1.12.16'
-- Found Sphinx: /usr/bin/sphinx-build  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install
==> ninja -j8 -l8 in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
[47/338] Building documentation.
Running Sphinx v1.3.6
making output directory...
loading pickled environment... not yet created
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 5 source files that are out of date
updating environment: 5 added, 0 changed, 0 removed
reading sources... [ 20%] configuration
reading sources... [ 40%] cost_functions
reading sources... [ 60%] evaluation
reading sources... [ 80%] index
reading sources... [100%] terminology

looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
[... 1064 lines cut ...]
but after I followed these command below

rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y

It newly installed  "libprotoc-dev".

The following NEW packages will be installed:
  libprotoc-dev
0 upgraded, 1 newly installed, 0 to remove and 603 not upgraded.
Need to get 720 kB of archives.
After this operation, 5,556 kB of additional disk space will be used.
Get:1 http://ppa.launchpad.net/maarten-fonville/protobuf/ubuntu xenial/main amd64 libprotoc-dev amd64 3.1.0-0ubuntu1~maarten0 [720 kB]
Fetched 720 kB in 2s (299 kB/s)        
Selecting previously unselected package libprotoc-dev:amd64.
(Reading database ... 927996 files and directories currently installed.)
Preparing to unpack .../libprotoc-dev_3.1.0-0ubuntu1~maarten0_amd64.deb ...
Unpacking libprotoc-dev:amd64 (3.1.0-0ubuntu1~maarten0) ...
Setting up libprotoc-dev:amd64 (3.1.0-0ubuntu1~maarten0) ...
#All required rosdeps installed successfully

and it changed the version of protoc from "3.4.0" to "3.1.0."

and I tried to build cartographer package with command with " catkin_make_isolated --install --use-ninja "
still, it has a problem..

kdaun, 2018-08-21:
Your error log states:

> #error incompatible with your Protocol Buffer headers. Please regenerate this file with a newer version of protoc.

indicating there are some issues with your protoc versions.

Which operating system and which ROS version are you using?

nsslab, 2018-08-22:
I am using Ubuntu 16.04(xenial) , ROS version is kinetic.

kdaun, 2018-08-31:
If you are building without grpc and prometheus a version >=3.0 should be sufficient.
You can also try installing the package directly via:
`sudo apt install ros-ROS_DISTRIBUTION-cartographer-ros`
Replace ROS_DISTRIBUTION by your ros distribution, e.g. `lunar`.

sweetquiet, 2018-09-11:
Source space: /home/youibot/catkin_ws1/src
Build space: /home/youibot/catkin_ws1/build_isolated
Devel space: /home/youibot/catkin_ws1/devel_isolated
Install space: /home/youibot/catkin_ws1/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'cartographer_ros_msgs'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
ninja: no work to do.
==> ninja -j4 -l4 in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
[13/13] cd /home/youibot/catkin_ws1/bu...ographer_ros_msgs/msg/SensorTopics.msg
==> ninja install in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
[14/14] Install the project...
-- Install configuration: ""
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/_setup_util.py
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/env.sh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.bash
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.sh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.zsh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/.rosinstall
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/LandmarkEntry.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/LandmarkList.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/StatusCode.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/StatusResponse.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapList.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapEntry.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapTexture.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SensorTopics.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/TrajectoryOptions.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/SubmapQuery.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/FinishTrajectory.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/StartTrajectory.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/WriteState.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/cmake/cartographer_ros_msgs-msg-paths.cmake
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteState.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StatusResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/LandmarkEntry.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapEntry.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQueryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectoryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectory.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQuery.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapTexture.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SensorTopics.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectory.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectoryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQueryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectoryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteStateRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/LandmarkList.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StatusCode.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteStateResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectoryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapList.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/TrajectoryOptions.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/manifest.l
[... cut ...]
gmock_main.cc:(.text._Z13RUN_ALL_TESTSv[_Z13RUN_ALL_TESTSv]+0xd)：对‘testing::UnitTest::Run()’未定义的引用
collect2: error: ld returned 1 exit status
[4/142] Building CXX object CMakeFiles...er/transform/rigid_transform_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/youibot/catkin_ws1/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/youibot/catkin_ws1/build_isolated/cartographer && /home/youibot/catkin_ws1/install_isolated/env.sh ninja -j4 -l4

sradmard, 2018-11-22:
Regarding another compiling error, it looks like that compiling cartographer returns some undefined ceres errors if we are using the latest version of ceres, 1.14.0. However, by switching back to the 1.13.0 version of ceres as documented in the "scripts/install_ceres.sh", the error disappears. It would be great if somebody can confirm that.
Thank you.

closed 2018-11-30 by gaschler

jinfagang, 2018-12-15:
@nsslab @kdaun @sweetquiet  Turns out to be protobuf version messed up. It must using protobuf 3.4.1 I have many version installed (3.6 and 3.4.1), but it default using 3.6, once that true all files generated by protoc will have a version identifier indicates that is 3.6 generated file, which can not recogonised by cartographer, so it runs to undefined error.

anyway, uninstall your higher protobuf version, or install from `sudo apt install ros-melodic-cartographer-ros`

BTW, protobuf is a really annoying library, just avoid to using that...........

zchao9456, 2018-12-27:
> I use this command to compile cartographer package
> 
> catkin_make_isolated --install --use-ninja
> 
> but, I keep getting this error below
> 
> Is there any solution for this issue?
> 
> ==> Processing plain cmake package: 'cartographer'
> ==> Building with env: '/home/nsslab/Howon/ros/catkin_ws/install_isolated/env.sh'
> build.ninja exists, skipping explicit cmake invocation...
> ==> ninja build.ninja in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
> [1/1] Re-running CMake...
> -- Build type: Release
> -- Boost version: 1.58.0
> -- Found the following Boost libraries:
> -- iostreams
> -- regex
> -- Found installed version of Eigen: /usr/lib/cmake/eigen3
> -- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
> -- Found required Ceres dependency: glog
> -- Found required Ceres dependency: gflags
> -- Found Ceres version: 1.13.0 installed in: /home/nsslab/Howon/ros/catkin_ws/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install
> ninja: no work to do.
> ==> ninja -j8 -l8 in '/home/nsslab/Howon/ros/catkin_ws/build_isolated/cartographer/install'
> [47/338] Building documentation.
> Running Sphinx v1.3.6
> making output directory...
> loading pickled environment... not yet created
> building [mo]: targets for 0 po files that are out of date
> building [html]: targets for 5 source files that are out of date
> updating environment: 5 added, 0 changed, 0 removed
> reading sources... [ 20%] configuration
> reading sources... [ 40%] cost_functions
> reading sources... [ 60%] evaluation
> reading sources... [ 80%] index
> reading sources... [100%] terminology
> 
> looking for now-outdated files... none found
> pickling environment... done
> checking consistency... done
> preparing documents... done
> writing output... [ 20%] configuration
> writing output... [ 40%] cost_functions
> writing output... [ 60%] evaluation
> writing output... [ 80%] index
> writing output... [100%] terminology
> 
> generating indices... genindex
> writing additional pages... search
> copying images... [ 50%] high_level_system_overview.png
> copying images... [100%] autogenerate_groundtruth.png
> 
> copying static files... done
> copying extra files... done
> dumping search index in English (code: en) ... done
> dumping object inventory... done
> build succeeded.
> [195/338] Linking CXX executable cartographer_migrate_serialization_format
> FAILED: : && /usr/bin/c++ -O3 -DNDEBUG CMakeFiles/cartographer_migrate_serialization_format.dir/cartographer/io/migrate_serialization_format_main.cc.o -o cartographer_migrate_serialization_format -rdynamic libcartographer.a /home/nsslab/Howon/ros/catkin_ws/install_isolated/lib/libceres.a -lglog -lgflags -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog -lgflags -lcairo -lprotobuf -lpthread && :
> libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::Submap::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const': serialization.pb.cc:(.text+0x123): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
> libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::Node::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const': serialization.pb.cc:(.text+0x19b): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
> libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::SerializedData::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const': serialization.pb.cc:(.text+0x233): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
> libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::SerializationHeader::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const': serialization.pb.cc:(.text+0x3b3): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
> libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::SerializationHeader::ByteSizeLong() const': serialization.pb.cc:(.text+0x403): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
> libcartographer.a(serialization.pb.cc.o):serialization.pb.cc:(.text+0x483): more undefined references to `google::protobuf::internal::proto3_preserve_unknown_' follow libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2fserialization_2eproto::(anonymous namespace)::protobuf_RegisterTypes(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)': serialization.pb.cc:(.text+0x8fa): undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)' libcartographer.a(serialization.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2fserialization_2eproto::(anonymous namespace)::protobuf_AssignDescriptors()' [... cut ...] mapping::proto::CellLimits>(cartographer::mapping::proto::CellLimits*)': cell_limits.pb.cc:(.text._ZN6google8protobuf5Arena3OwnIN12cartographer7mapping5proto10CellLimitsEEEvPT_[_ZN6google8protobuf5Arena3OwnIN12cartographer7mapping5proto10CellLimitsEEEvPT_]+0xd): undefined reference to `google::protobuf::internal::ArenaImpl::AddCleanup(void*, void (_)(void_))' libcartographer.a(cell_limits.pb.cc.o):(.data.rel.ro._ZTVN12cartographer7mapping5proto10CellLimitsE[_ZTVN12cartographer7mapping5proto10Cel

xijunke, 2019-04-12:
# Comment by XJ：
Comment the following lines for CMakeLists.txt from "/home/q/catkin_ws/src/cartographer"
###################################################################################
# foreach(ABS_FIL ${ALL_TESTS})
#   file(RELATIVE_PATH REL_FIL ${PROJECT_SOURCE_DIR} ${ABS_FIL})
#   get_filename_component(DIR ${REL_FIL} DIRECTORY)
#   get_filename_component(FIL_WE ${REL_FIL} NAME_WE)
#  # Replace slashes as required for CMP0037.
#   string(REPLACE "/" "." TEST_TARGET_NAME "${DIR}/${FIL_WE}")
#   google_test("${TEST_TARGET_NAME}" ${ABS_FIL})
#   if(${BUILD_GRPC})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC grpc++)
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC async_grpc)
#   endif()
#   if(${BUILD_PROMETHEUS})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC prometheus-cpp)
#   endif()
#   target_link_libraries("${TEST_TARGET_NAME}" PUBLIC ${TEST_LIB})
# endforeach()
###################################################################################

racla, 2019-05-21:
@xijunke not work for me.

shurikrulik, 2019-11-07:
@xijunke this method works !