cartographer-project/cartographer #1396: fast_correlative_scan_matcher_2d.cc:68] Check failed: non_ascending_maxima_.size() == 0

narutojxl, 2018-08-20:
Hello, i use cartographer 1.0, cartographer_ros 1.0, and ceres 1.14.0,  laser is  sick Tim571,  when i running cartographer_ros  demo_backpack_2d.launch,  it always throw  the same error when the robot bringup and teleop for a moment , here is  terminal log and cfg file.  

- BTW,  i launch  demo_backpack_2d.launch and robot is stationary, when i see in the RVIZ, i saw the TF frome "map"  to "odom" published  by cartographer is  oscillated , according to my configure lua file. I not use odom topic and imu , only use sick laser scan . 
- Any suggestions will be much appreciated!!!

`hans@robo00:~$ roslaunch cartographer_ros   demo_backpack_2d_jxl.launch 
... logging to /home/xlh/.ros/log/3a8399b8-a466-11e8-ad96-00045f9a45a3/roslaunch-robo00-8919.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/xlh/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

WARNING: Package name "sonar2PointCloud" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "sonar2PointCloud" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "sonar2PointCloud" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "sonar2PointCloud" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
started roslaunch server http://192.168.0.3:51039/

SUMMARY
========

PARAMETERS
 * /diagnostic_aggregator/analyzers/input_ports/contains: ['Digital Input',...
 * /diagnostic_aggregator/analyzers/input_ports/path: Input Ports
 * /diagnostic_aggregator/analyzers/input_ports/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/input_ports/timeout: 5.0
 * /diagnostic_aggregator/analyzers/input_ports/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/kobuki/contains: ['Watchdog', 'Mot...
 * /diagnostic_aggregator/analyzers/kobuki/path: Kobuki
 * /diagnostic_aggregator/analyzers/kobuki/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/kobuki/timeout: 5.0
 * /diagnostic_aggregator/analyzers/kobuki/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/power/contains: ['Battery']
 * /diagnostic_aggregator/analyzers/power/path: Power System
 * /diagnostic_aggregator/analyzers/power/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/power/timeout: 5.0
 * /diagnostic_aggregator/analyzers/power/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/sensors/contains: ['Cliff Sensor', ...
 * /diagnostic_aggregator/analyzers/sensors/path: Sensors
 * /diagnostic_aggregator/analyzers/sensors/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /mobile_base/base_frame: base_footprint
 * /mobile_base/battery_capacity: 16.5
 * /mobile_base/battery_dangerous: 13.2
 * /mobile_base/battery_low: 14.0
 * /mobile_base/bias: 0.482
 * /mobile_base/cmd_vel_timeout: 0.6
 * /mobile_base/device_ip: 192.168.0.4
 * /mobile_base/device_port: 2000
 * /mobile_base/odom_frame: odom
 * /mobile_base/publish_tf: True
 * /mobile_base/tick_to_rad: 0.00698132
 * /mobile_base/use_imu_heading: True
 * /mobile_base/wheel_l_radius: 0.065
 * /mobile_base/wheel_left_joint_name: wheel_left_joint
 * /mobile_base/wheel_r_radius: 0.065
 * /mobile_base/wheel_right_joint_name: wheel_right_joint
 * /mobile_base/wheel_rotate_speed_scale: 1.0
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /sick_tim551_2050001/frame_id: laser
 * /sick_tim551_2050001/hostname: 192.168.0.2
 * /sonar2PointCloud/cutting: False
 * /sonar2PointCloud/interval_dist_between_sonars: 0.16
 * /sonar2PointCloud/laser_frame: laser
 * /sonar2PointCloud/max_range: 3.0
 * /sonar2PointCloud/measurement_angles: 30
[... 141 lines cut ...]
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 10,

  num_point_clouds = 0,

  lookup_transform_timeout_sec = 0.2,  ---0.2
  submap_publish_period_sec = 0.3,  -- 0.3

  pose_publish_period_sec = 50e-3,   --5e-3    ---: map--->odom TF

  trajectory_publish_period_sec = 100e-3,  --30e-3
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 15  --10
TRAJECTORY_BUILDER_2D.use_imu_data = false

return options`

kdaun, 2018-08-20:
Can you please record a bag file and provide the output of `rosbag_validate` and `rosbag info` of your bag file.

kdaun, 2018-08-22:
Thank you for providing the output of `rosbag info` in such a convenient way.

Please also run rosbag_validate, e.g.:
`rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ${HOME}/example_dir/example.bag`
And post the text output here.

kdaun, 2018-08-23:
Thank you for providing the output of rosbag_validate.
Let's look at the output.

>E0823 09:44:06.304395 13436 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames 

It looks, like the bag file is recorded with Cartographer running. Can you please record a bag without Cartographer running and run rosbag_validate again?

>W0823 09:44:06.304738 13436 rosbag_validate_main.cc:103] frame_id gyro_link time 1534761994760554313: IMU linear acceleration is 0 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0 0 0

>E0823 09:44:06.304785 13436 rosbag_validate_main.cc:315] Sensor with frame_id "gyro_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.

>E0823 09:44:06.506513 13436 rosbag_validate_main.cc:365] Average IMU linear acceleration is 0 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.

These messages indicate that your IMU does not expose the acceleration and the sensor time stamps jump back in time. As long as we are not using the IMU this should not cause any issues.

>E0823 09:44:06.327728 13436 rosbag_validate_main.cc:160] Sensor with frame_id "laser" is not sequential in time.Previous range message ends at time 636703588086866674, current one at time 636703588086851791

These messages indicate that range observations are not sequential in time. Cartographer requires sequential range observations. Please ensure the time stamping is correct.

In order to further evaluate the issue:
- Keep the system as simple as possible. Disable all nodes you don't need for Cartographer.
- Record a bag without Cartographer running and run rosbag_validate again.
- Ensure the time stamping of the scan messages is correct.

PS: Please answer to this issues instead of editing the initial issue, then the issues is more comprehensible for other people.

kdaun, 2018-08-31:
The output for this rosbag looks ok with respect to the `/scan` messages.
Do you observe the `Check failed: non_ascending_maxima_.size() == 0` error when you play the bag file and run Cartographer?

narutojxl, 2018-09-04:
No, when i played  the bag file and running cartographer on my computer,  not on robot system,  i didn't see the error  `Check failed: non_ascending_maxima_.size() == 0` in the carto running terminal.

Sbergdahl, 2018-11-08:
Is there any solution to this yet? Were encountering the same error. It seems to be tirggered especially by sudden rotations although It will trigger sooner or later anyway unless the robot is standing still.