cartographer-project/cartographer #1397: drifting in height

kongcheng08, 2018-08-21:
I run code with a vlp16 LiDAR without imu , I found that  map and position is drifting in Z-direction. How to fix it.

kdaun, 2018-08-21:
Are you running the 2D or 3D pipeline?
What kind of system or you using (ground vehicle, human carrying the LiDAR, ...)?
What kind of environment are you mapping?
Are you using cartographer_ros?
If you are using cartographer_ros please record a rosbag file and:
1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.

gaschler, 2018-11-30:
Closing as inactive

closed 2018-11-30 by gaschler