cartographer-project/cartographer #1399: Question: Configuration Error while trying to use Multiple Lidars with Cartographer

jollysg, 2018-08-27:
Hello,

I've been trying to configure Cartographer with 2 Lidar scans. For this, as mentioned in the configuration page (https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) I am setting the 'num_laser_scans' parameter in the lua file as 2. I have modified this lua file from the backpacker_2d.lua example. Since our robot doesn't have an IMU, I have disabled that. Apart from that all other configurations in map_builder.lua and trajectory_builder.lua are untouched. 

When I run the launch file however, I get the following error - 

---

[FATAL] [1535388655.530800981]: F0827 12:50:55.000000  1275 lua_parameter_dictionary.cc:409] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'num_laser_scans' was used the wrong number of times.
*** Check failure stack trace: ***
    @     0x7f2ec67425cd  google::LogMessage::Fail()
    @     0x7f2ec6744433  google::LogMessage::SendToLog()
    @     0x7f2ec674215b  google::LogMessage::Flush()
    @     0x7f2ec6744e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x519c22  cartographer::common::LuaParameterDictionary::CheckAllKeysWereUsedExactlyOnceAndReset()
    @           0x519e05  cartographer::common::LuaParameterDictionary::~LuaParameterDictionary()
    @           0x4ad93a  (unknown)
    @           0x4abb24  main
    @     0x7f2ec33bc830  __libc_start_main
    @           0x4ad509  _start
    @              (nil)  (unknown)
[cartographer_node-1] process has died [pid 1275, exit code -6, cmd /opt/ros/kinetic/lib/cartographer_ros/cartographer_node -configuration_directory /home/jasprit/Documents/workspace/catkin_ws/src/scr_map_updating/config -configuration_basename cartographer_config_dual_lidar.lua scan_1:=/front/scan scan_2:=/rear/scan __name:=cartographer_node __log:=/home/jasprit/.ros/log/d33d164e-aa05-11e8-83d1-847beb1d9404/cartographer_node-1.log].
log file: /home/jasprit/.ros/log/d33d164e-aa05-11e8-83d1-847beb1d9404/cartographer_node-1*.log
---

Am I missing some other configuration over here? My understanding was all I needed to do was to set num_laser_scans parameter and use scan_1, scan_2 topics for the scans in the launch file. 

Another thing that I notices is - When I take use_laser_scan parameter out of the lua file (I did that to check in case it conflicts with the num_laser_scans parameter), I get another error saying parameter "use_laser_scan" not set. This parameter isn't documented in the cartographer documentations, and I have come across a few other configuration files online where people used it without use_laser_scan parameters for 2 lidars (https://projekter.aau.dk/projekter/files/281489815/Report_17gr934.pdf). Is the documentation outdated ?

On a side note, I am able to use the cartographer with single Lidar scan with the above configurations (the only difference is I don't use num_laser_scans parameter in the lua file in that case, and ensure that the use_laser_scan parameter is set to true).  

 The configuration file, launch file and rosbag I am using can be found over here (https://drive.google.com/open?id=18aAW3-EBG8sKgkKvcY-5tJsOOeZzHSv3). However, this is more of a configuration issue, and can be reproduced without using a bag file. The node crashes at launch with the above error even before the bag files are run.

Any insight would be appreciated.

Thanks,
Jasprit

kdaun, 2018-08-31:
The error message indicates there is an error your configuration file with the parameter `num_laser_scans`.
See the [MIR 100 configuration file](https://github.com/googlecartographer/cartographer_mir/blob/master/cartographer_mir/configuration_files/mir-100-mapping.lua) for an example of a configuration with two 2D lidars performing 2D SLAM.

gaschler, 2018-11-30:
Closing as answered

closed 2018-11-30 by gaschler